wolfSSL/wolfssl #4382: Big Delay of 'Server Hello'

ElLamparto, 2021-09-08:
Hello,

wolfSSL acts as a server. When the number of client connections is > 30, there is a delay (in the attached example 13s) in sending Server Hello (being a reply to Client Hello). What can cause that? Please see the png and then the Wireshark attached.

Thanks...

wolfssl-4.3.0 (Windows)

[image]
[capture 31 08 21.zip](https://github.com/wolfSSL/wolfssl/files/7128889/capture.31.08.21.zip)

dgarske, 2021-09-08:
Hi @ElLamparto ,

Can you tell us more about the server platform running wolfSSL? Is this an embedded target? What is your network stack? How is your RNG setup? Are you using session resumption? You mentioned Windows? Perhaps enabling our logging would show if it is a delay in getting the `client_hello` or time it takes in generating the `server_hello` or if we have sent the server_hello and the network stack is not flushing it? To enable logging define `DEBUG_WOLFSSL` and call `wolfSSL_Logging_ON();`.

Thanks,
David Garske, wolfSSL

ElLamparto, 2021-09-15:
Hi @David,

Just to answer some of your questions, I include the TCP socket code. Maybe you will see something that I do not see.
If not, I will enable the wolfSSL logging and send you the traces. Thanks a lot!

`//Server socket creation:

#include "Winsock2.h"

				char szLog [255];
				sprintf		(szLog, "INFO: Creating TCP server socket for local IP [%s] and port [%d]", m_szLocalIpAddress, m_iLocalPort);
				LogMessage	(szLog, "CSSLServerEngine::bSSLInit", 2);

				m_srv_sockfd = socket (AF_INET_V, SOCK_STREAM, IPPROTO_TCP);

				bOk = (m_srv_sockfd != INVALID_SOCKET);

				if (bOk)
				{
					SOCKADDR_IN_T addr;
					memset (&addr, 0, sizeof (addr));

					addr.sin_family			= AF_INET_V;
					addr.sin_port			= htons		(m_iLocalPort);
					addr.sin_addr.s_addr	= inet_addr	(m_szLocalIpAddress);

					bOk = (bind (m_srv_sockfd, (const struct sockaddr*)&addr, sizeof(addr)) == 0);

					if (bOk)
					{
						// set the socket I/O mode
						// if iMode = 0, blocking is enabled
						// if iMode != 0, non-blocking mode is enabled

						u_long iMode = 1;	// Non-blocking mode
						ioctlsocket (m_srv_sockfd, FIONBIO, &iMode);

						bOk = (listen (m_srv_sockfd, SOMAXCONN) == 0);

						if (!bOk)
						{	LogMessage ("ERROR: listen() failed", "CSSLServerEngine::bSSLInit", 0);
						}
					}
					else
					{	LogMessage ("ERROR: bind() failed", "CSSLServerEngine::bSSLInit", 0);
					}
				}
				else
				{	LogMessage ("ERROR: Socket creation failed", "CSSLServerEngine::bSSLInit", 0);
				}

`

`
// Accept client connections:
		char szLog [255];

		SOCKET_T	cli_sockfd;	// Socket temporaire contenant la nouv connexion
		sockaddr_in	cli_addr;	// Adresse associée
		socklen_t	cli_addr_len = sizeof (struct sockaddr_in);

		fd_set TabListen;
		struct timeval tps;

		FD_ZERO (&TabListen);
		FD_SET  (m_srv_sockfd, &TabListen);

		// select() will wait 50ms max.
		tps.tv_sec  = 0;	// [s]
		tps.tv_usec = 0;	// [µs]

		if (select (0, &TabListen, 0, NULL, &tps) != 0)
		{
			// Etablissement de la connexion
			cli_sockfd = accept (m_srv_sockfd, (struct sockaddr*)&cli_addr, &cli_addr_len);

			bNewCnx = (cli_sockfd != SOCKET_ERROR);

			if (!bNewCnx)
			{
				const int iError = WSAGetLastError();

				if (iError != WSAEWOULDBLOCK)
				{
					sprintf (szLog, "WARNING: New connection FAILED with winsock error code [%d]", WSAGetLastError());
					LogMessage (szLog, "CSSLServerEngine::bCheckForConnections", 0);
				}
			}
		}

		if (bNewCnx)
		{
			// set the socket I/O mode
			// if iMode = 0, blocking is enabled
			// if iMode != 0, non-blocking mode is enabled

			u_long iMode = 1;	// Non-blocking mode
			ioctlsocket (cli_sockfd, FIONBIO, &iMode);

			struct linger var_linger;
			var_linger.l_onoff = 0;
			var_linger.l_linger = 0;
			setsockopt (m_srv_sockfd, SOL_SOCKET, SO_LINGER, (const char *)& var_linger, sizeof (var_linger));

			iClientFd	= cli_sockfd;
			iClientPort	= ntohs (cli_addr.sin_port);
			strcpy (szClientIP, inet_ntoa (cli_addr.sin_addr));

			WOLFSSL* ssl = wolfSSL_new (m_ctx);
			wolfSSL_set_fd (ssl, cli_sockfd);

			// Sauvegarder la connexion dans le map
			typedef std::map <int, WOLFSSL*>::value_type vItem;
			mapClientCnx.insert (vItem (iClientFd, ssl));

			sprintf (szLog, "INFO: Accepted new connection: TCP server fd [%d], local IP [%s], local port [%d], client fd [%d], client IP [%s], client port [%d]",
						m_srv_sockfd, m_szLocalIpAddress, m_iLocalPort, iClientFd, szClientIP, iClientPort);
			LogMessage (szLog, "CSSLServerEngine::bCheckForConnections", 0);
		}

`

`
// Close client connection:

		WOLFSSL* ssl = GetSsl (iClientFd);

		char szLog [128];

		if (ssl == NULL)
		{
			sprintf (szLog, "INTERNAL ERROR: Not found client fd [%d]", iClientFd);
			LogMessage (szLog, "CSSLServerEngine::bCloseClientCnx", 0);
		}
		else
		{
			wolfSSL_shutdown (ssl);
			wolfSSL_free	 (ssl);
			ssl = NULL;

			// Supprimer du map
			mapClientCnx.erase (mapClientCnx.find (iClientFd));

			closesocket (iClientFd);

			sprintf (szLog, "INFO: Closed client connection fd [%d]", iClientFd);
			LogMessage (szLog, "CSSLServerEngine::bCloseClientCnx", 1);

			bOk = true;
		}

`

dgarske, 2021-09-16:
Hi @ElLamparto ,

What is your reasoning for setting `SO_LINGER`? Typically that will allow a closed socket to remain open until close() when it sends FIN.

I don't see your calling code for wolfSSL_connect here?

Also if you are using a non-blocking socket your calls to wolfSSL_connect, wolfSSL_read and wolfSSL_write will need some retry logic to handle want read/write. See out non-blocking TLS examples here:
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-nonblocking.c

Thanks,
David Garske, wolfSSL

ElLamparto, 2021-12-05:
Hello,

Actually it was my fault. I've blocked the socket on reading.
Thanks.

closed 2021-12-05 by ElLamparto