wolfSSL/wolfssl #4397: [question] Subject Alternative Names appear in a different order

prince-chrismc, 2021-09-14:
:wave: While working on adding support to my library https://github.com/prince-chrismc/jwt-cpp/pull/12 , I had one unit [test fail](https://github.com/prince-chrismc/jwt-cpp/runs/3593512681) with wolfSSL... This causes the PEM to be different so a naive string compare does not work.

This is by no means a bug, but if it's easy to switch it would be great to have even closer compatibility

<details>
<summary>With OpenSSL and LibreSSL</summary>

```
-----BEGIN CERTIFICATE-----
MIIF8DCCBVmgAwIBAgIKYFOB9QABAACIvTANBgkqhkiG9w0BAQUFADBGMQswCQYD
VQQGEwJVUzETMBEGA1UEChMKR29vZ2xlIEluYzEiMCAGA1UEAxMZR29vZ2xlIElu
dGVybmV0IEF1dGhvcml0eTAeFw0xMzA1MjIxNTQ5MDRaFw0xMzEwMzEyMzU5NTla
MGYxCzAJBgNVBAYTAlVTMRMwEQYDVQQIEwpDYWxpZm9ybmlhMRYwFAYDVQQHEw1N
b3VudGFpbiBWaWV3MRMwEQYDVQQKEwpHb29nbGUgSW5jMRUwEwYDVQQDFAwqLmdv
b2dsZS5jb20wWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAARmSpIUbCqhUBq1UwnR
Ai7/TNSk6W8JmasR+I0r/NLDYv5yApbAz8HXXN8hDdurMRP6Jy1Q0UIKmyls8HPH
exoCo4IECjCCBAYwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMAsGA1Ud
DwQEAwIHgDAdBgNVHQ4EFgQUU3jT0NVNRgU5ZinRHGrlyoGEnoYwHwYDVR0jBBgw
FoAUv8Aw6/VDET5nup6R+/xq2uNrEiQwWwYDVR0fBFQwUjBQoE6gTIZKaHR0cDov
L3d3dy5nc3RhdGljLmNvbS9Hb29nbGVJbnRlcm5ldEF1dGhvcml0eS9Hb29nbGVJ
bnRlcm5ldEF1dGhvcml0eS5jcmwwZgYIKwYBBQUHAQEEWjBYMFYGCCsGAQUFBzAC
hkpodHRwOi8vd3d3LmdzdGF0aWMuY29tL0dvb2dsZUludGVybmV0QXV0aG9yaXR5
L0dvb2dsZUludGVybmV0QXV0aG9yaXR5LmNydDAMBgNVHRMBAf8EAjAAMIICwwYD
VR0RBIICujCCAraCDCouZ29vZ2xlLmNvbYINKi5hbmRyb2lkLmNvbYIWKi5hcHBl
bmdpbmUuZ29vZ2xlLmNvbYISKi5jbG91ZC5nb29nbGUuY29tghYqLmdvb2dsZS1h
bmFseXRpY3MuY29tggsqLmdvb2dsZS5jYYILKi5nb29nbGUuY2yCDiouZ29vZ2xl
LmNvLmlugg4qLmdvb2dsZS5jby5qcIIOKi5nb29nbGUuY28udWuCDyouZ29vZ2xl
LmNvbS5hcoIPKi5nb29nbGUuY29tLmF1gg8qLmdvb2dsZS5jb20uYnKCDyouZ29v
Z2xlLmNvbS5jb4IPKi5nb29nbGUuY29tLm14gg8qLmdvb2dsZS5jb20udHKCDyou
Z29vZ2xlLmNvbS52boILKi5nb29nbGUuZGWCCyouZ29vZ2xlLmVzggsqLmdvb2ds
ZS5mcoILKi5nb29nbGUuaHWCCyouZ29vZ2xlLml0ggsqLmdvb2dsZS5ubIILKi5n
b29nbGUucGyCCyouZ29vZ2xlLnB0gg8qLmdvb2dsZWFwaXMuY26CFCouZ29vZ2xl
Y29tbWVyY2UuY29tgg0qLmdzdGF0aWMuY29tggwqLnVyY2hpbi5jb22CECoudXJs
Lmdvb2dsZS5jb22CFioueW91dHViZS1ub2Nvb2tpZS5jb22CDSoueW91dHViZS5j
b22CFioueW91dHViZWVkdWNhdGlvbi5jb22CCyoueXRpbWcuY29tggthbmRyb2lk
LmNvbYIEZy5jb4IGZ29vLmdsghRnb29nbGUtYW5hbHl0aWNzLmNvbYIKZ29vZ2xl
LmNvbYISZ29vZ2xlY29tbWVyY2UuY29tggp1cmNoaW4uY29tggh5b3V0dS5iZYIL
eW91dHViZS5jb22CFHlvdXR1YmVlZHVjYXRpb24uY29tMA0GCSqGSIb3DQEBBQUA
A4GBAAMn0K3j3yhC+X+uyh6eABa2Eq7xiY5/mUB886Ir19vxluSMNKD6n/iY8vHj
trn0BhuW8/vmJyudFkIcEDUYE4ivQMlsfIL7SOGw6OevVLmm02aiRHWj5T20Ds+S
OpueYUG3NBcHP/5IzhUYIQJbGzlQaUaZBMaQeC8ZslMNLWI2
-----END CERTIFICATE-----
```

</details>

[image]

<details>
<summary>With wolfSSL</summary>

```
-----BEGIN CERTIFICATE-----
MIIFATCCBGqgAwIBAgIKYFOB9QABAACIvTANBgkqhkiG9w0BAQUFADBGMQswCQYD
VQQGEwJVUzETMBEGA1UEChMKR29vZ2xlIEluYzEiMCAGA1UEAxMZR29vZ2xlIElu
dGVybmV0IEF1dGhvcml0eTAeFw0xMzA1MjIxNTQ5MDRaFw0xMzEwMzEyMzU5NTla
MGYxCzAJBgNVBAYTAlVTMRMwEQYDVQQIEwpDYWxpZm9ybmlhMRYwFAYDVQQHEw1N
b3VudGFpbiBWaWV3MRMwEQYDVQQKEwpHb29nbGUgSW5jMRUwEwYDVQQDDAwqLmdv
b2dsZS5jb20wWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAARmSpIUbCqhUBq1UwnR
Ai7/TNSk6W8JmasR+I0r/NLDYv5yApbAz8HXXN8hDdurMRP6Jy1Q0UIKmyls8HPH
exoCo4IDGzCCAxcwggLDBgNVHREEggK6MIICtoIUeW91dHViZWVkdWNhdGlvbi5j
b22CC3lvdXR1YmUuY29tggh5b3V0dS5iZYIKdXJjaGluLmNvbYISZ29vZ2xlY29t
bWVyY2UuY29tggpnb29nbGUuY29tghRnb29nbGUtYW5hbHl0aWNzLmNvbYIGZ29v
LmdsggRnLmNvggthbmRyb2lkLmNvbYILKi55dGltZy5jb22CFioueW91dHViZWVk
dWNhdGlvbi5jb22CDSoueW91dHViZS5jb22CFioueW91dHViZS1ub2Nvb2tpZS5j
b22CECoudXJsLmdvb2dsZS5jb22CDCoudXJjaGluLmNvbYINKi5nc3RhdGljLmNv
bYIUKi5nb29nbGVjb21tZXJjZS5jb22CDyouZ29vZ2xlYXBpcy5jboILKi5nb29n
bGUucHSCCyouZ29vZ2xlLnBsggsqLmdvb2dsZS5ubIILKi5nb29nbGUuaXSCCyou
Z29vZ2xlLmh1ggsqLmdvb2dsZS5mcoILKi5nb29nbGUuZXOCCyouZ29vZ2xlLmRl
gg8qLmdvb2dsZS5jb20udm6CDyouZ29vZ2xlLmNvbS50coIPKi5nb29nbGUuY29t
Lm14gg8qLmdvb2dsZS5jb20uY2+CDyouZ29vZ2xlLmNvbS5icoIPKi5nb29nbGUu
Y29tLmF1gg8qLmdvb2dsZS5jb20uYXKCDiouZ29vZ2xlLmNvLnVrgg4qLmdvb2ds
ZS5jby5qcIIOKi5nb29nbGUuY28uaW6CCyouZ29vZ2xlLmNsggsqLmdvb2dsZS5j
YYIWKi5nb29nbGUtYW5hbHl0aWNzLmNvbYISKi5jbG91ZC5nb29nbGUuY29tghYq
LmFwcGVuZ2luZS5nb29nbGUuY29tgg0qLmFuZHJvaWQuY29tggwqLmdvb2dsZS5j
b20wHQYDVR0OBBYEFFN409DVTUYFOWYp0Rxq5cqBhJ6GMB8GA1UdIwQYMBaAFL/A
MOv1QxE+Z7qekfv8atrjaxIkMA4GA1UdDwEB/wQEAwIHgDANBgkqhkiG9w0BAQUF
AAOBgQADJ9Ct498oQvl/rsoengAWthKu8YmOf5lAfPOiK9fb8ZbkjDSg+p/4mPLx
47a59AYblvP75icrnRZCHBA1GBOIr0DJbHyC+0jhsOjnr1S5ptNmokR1o+U9tA7P
kjqbnmFBtzQXBz/+SM4VGCECWxs5UGlGmQTGkHgvGbJTDS1iNg==
-----END CERTIFICATE-----
```

</details>

[image]

embhorn, 2021-09-14:
Hi @prince-chrismc 

Thanks for the suggestion. I'll run it past the team. Only drawback I can foresee would be breaking existing functionality.

dgarske, 2021-09-14:
Hi @prince-chrismc , which openssl API('s) are you using to generate this alt name list? I agree with you, we should return the alt names in the same order as openssl to be more compatible.
Thanks, David Garske, wolfSSL

prince-chrismc, 2021-09-14:
The screen shots are from https://www.sslshopper.com/certificate-decoder.html

- [different certs in pem](https://github.com/prince-chrismc/jwt-cpp/blob/70066bb5fc5cbf896784a26d6ecf6ddedd68f810/tests/HelperTest.cpp)
- [implementation](https://github.com/Thalhammer/jwt-cpp/blob/aaf3c714e343dc381a5dfca921075b27f89d6a2c/include/jwt-cpp/jwt.h#L445-L462)

In short we are using
- d2i_X509
- PEM_write_bio_X509
- BIO_get_mem_data

embhorn, 2021-09-15:
Hi @prince-chrismc 

I've worked up a PR #4406 to address this concern. Could you please check that it works with your test?

prince-chrismc, 2021-09-16:
That definitely helped and the SAN now match but sadly my test still fails. https://github.com/prince-chrismc/jwt-cpp/runs/3616280679?check_suite_focus=true#step:8:66

[openssl.summary.txt](https://github.com/wolfSSL/wolfssl/files/7174114/openssl.summary.txt)
[wolfssl.summary.txt](https://github.com/wolfSSL/wolfssl/files/7174113/wolfssl.summary.txt)

Doing a compare there's a few fields missing and different ordering

[diff.txt](https://github.com/wolfSSL/wolfssl/files/7174131/diff.txt)

embhorn, 2021-09-21:
The remaining discrepancies appear to be extensions that are either not enabled or not currently supported by wolfSSL:

```
            X509v3 Extended Key Usage:
                TLS Web Server Authentication, TLS Web Client Authentication
.
.
.
            X509v3 CRL Distribution Points:

                Full Name:
                  URI:http://www.gstatic.com/GoogleInternetAuthority/GoogleInternetAuthority.crl

            Authority Information Access:
                CA Issuers - URI:http://www.gstatic.com/GoogleInternetAuthority/GoogleInternetAuthority.crt

            X509v3 Basic Constraints: critical
                CA:FALSE
```

Is wolfSSL configured with `enable-crl`?

I can open a feature request for the remaining extensions is you'd like?

prince-chrismc, 2021-09-22:
My setup scripts call `--enable-all`, I assume that turns every feature?

No. Thats not need, it only impacts test code. Otherwise the functionality is intact

embhorn, 2021-09-22:
Yes, `--enable-all` does turn on support for CRL and TLS extensions.

Understood, I'll go ahead and close this ticket out. Thanks again for bringing this to our attention.

closed 2021-09-22 by embhorn