wolfSSL/wolfssl #4399: "libwolfssl.so: undefined reference to `wc_Sha256Transform'" with --enable-devcrypto=yes

hauke, 2021-09-14:
When we set `--enable-devcrypto=yes` in OpenWrt libraries linking against libwollssl 4.8.1 fail, this worked with wolfssl 4.7.0.  With  `--enable-devcrypto=no` it works fine.

Linking fails like this:
```
/home/hauke/openwrt/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-10.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/10.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /home/hauke/openwrt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libwolfssl.so: undefined reference to `wc_Sha256Transform'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I see the following difference between the working an the broken wolfssl configuration
```
--- wolfssl-working.txt 2021-09-14 12:22:22.201246550 +0200
+++ wolfssl-broken.txt  2021-09-14 12:22:08.858077571 +0200
@@ -173,7 +173,7 @@
 --disable-curve25519
 --disable-afalg
 --enable-openvpn
---enable-devcrypto=no
+--enable-devcrypto=yes
 --enable-ocsp
 --enable-ocspstapling
 --enable-ocspstapling2
@@ -189,6 +189,7 @@
 
 
 
+---
 Configuration summary for wolfssl version 4.8.1
 
    * Installation prefix:        /usr
@@ -370,7 +371,7 @@
    * Xilinx Hardware Acc.:       no
    * Inline Code:                yes
    * Linux AF_ALG:               no
-   * Linux devcrypto:            no
+   * Linux devcrypto:            yes
    * Crypto callbacks:           no
    * i.MX6 CAAM:                 no
```

full working configuration:
```
Configuration summary for wolfssl version 4.8.1

   * Installation prefix:        /usr
   * System type:                openwrt-linux-gnu
   * Host CPU:                   arm
   * C Compiler:                 arm-openwrt-linux-muslgnueabi-gcc
   * C Flags:                    -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/hauke/openwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -DFP_MAX_BITS=8192 -fomit-frame-pointer -flto    -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:               arm-openwrt-linux-muslgnueabi-g++
   * C++ Flags:                  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/hauke/openwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -DFP_MAX_BITS=8192 -fomit-frame-pointer -flto 
   * CPP Flags:                  -I/home/hauke/openwrt/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-10.3.0_musl_eabi/usr/include -I/home/hauke/openwrt/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-10.3.0_musl_eabi/include/fortify -I/home/hauke/openwrt/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-10.3.0_musl_eabi/include 
   * CCAS Flags:                 -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/hauke/openwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -DFP_MAX_BITS=8192 -fomit-frame-pointer -flto   
   * LIB Flags:                   -pie -z relro -z now 
   * Debug enabled:              no
   * Coverage enabled:           
   * Warnings as failure:        no
   * make -j:                    no
   * VCS checkout:               no

   Features 
   * FIPS:                       no
   * Single threaded:            yes
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          yes
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
[... 134 lines cut ...]
   * Crypt tests:                no
   * Stack sizes in tests:       no
   * Heap stats in tests:        no
   * User Crypto:                no
   * Fast RSA:                   no
   * Single Precision:           no
   * SP math implementation:     no
   * Async Crypto:               no
   * PKCS#11:                    no
   * PKCS#12:                    yes
   * Cavium Nitrox:              no
   * Cavium Octeon (Sync):       no
   * Intel Quick Assist:         no
   * ARM ASM:                    no
   * AES Key Wrap:               yes
   * Write duplicate:            no
   * Xilinx Hardware Acc.:       no
   * Inline Code:                yes
   * Linux AF_ALG:               no
   * Linux devcrypto:            yes
   * Crypto callbacks:           no
   * i.MX6 CAAM:                 no
```

This was reported to us by @rmandrad, see https://github.com/openwrt/openwrt/commit/7d92bb0509615550b98e2dc71091073c8258d564

embhorn, 2021-09-14:
@SparkiDev Could you please review this report?

embhorn, 2021-09-16:
Hi @peterwillcn 

We'll post an update after this has been reviewed. For priority customer support, please send a message to support@wolfssl.com

JacobBarthelmeh, 2021-09-17:
Hi @peterwillcn ,

As an update on this I have reproduced the issue reported using "./configure --enable-devcrypto --enable-opensslextra" when compiling wolfSSL version 4.8.1. A fix is in process, will put a link to the PR when it gets posted.

Regards,
Jacob from wolfSSL.

JacobBarthelmeh, 2021-09-17:
Looks like it was some missed macro guards around a new feature added in to the compatibility layer. We have a pull request that is under review here https://github.com/wolfSSL/wolfssl/pull/4414. Thanks for the report.

peterwillcn, 2021-09-23:
Thank everybody

closed 2021-09-24 by JacobBarthelmeh