wolfSSL/wolfssl #4410: verifyCallback not called

Xeenych, 2021-09-17:
I'm using WolfSSL client on embedded board.
The board doesn't have RTC, so I need to override certificate DATE check errors
```
wolfSSL_CTX_set_verify(ctx, WOLFSSL_VERIFY_DEFAULT, verifyCallback);

if ((rc = wolfSSL_CTX_trust_peer_buffer(ctx, certBuf, certBufSz, WOLFSSL_FILETYPE_PEM)) != SSL_SUCCESS) {
    print_dbg("Error loading cert, %d\n", rc);
    goto cleanup;
}
print_dbg("Certificate loaded\n");
```

The debug output is:
```
[APP]ctx created
5334  : wolfSSL Entering wolfSSL_CTX_set_verify
5338  : wolfSSL Entering wolfSSL_CTX_trust_peer_buffer
5342  : Processing CA PEM file
5346  : wolfSSL Entering PemToDer
5352  : Adding a Trusted Peer Cert
5352  : wolfSSL Entering GetExplicitVersion
5356  : wolfSSL Entering GetSerialNumber
5358  : Got Cert Header
5360  : wolfSSL Entering GetAlgoId
5364  : wolfSSL Entering GetObjectId()
5368  : Got Algo ID
5370  : Getting Cert Name
5372  : Getting Cert Name
5374  : Date BEFORE check failed
5376  : Getting Cert Name
5378  : Getting Cert Name
5382  : Got Subject Name
5384  : wolfSSL Entering GetAlgoId
5386  : wolfSSL Entering GetObjectId()
5390  : Got Key
5392  : Parsed Past Key
5394  : wolfSSL Entering DecodeCertExtensions
5398  : wolfSSL Entering GetObjectId()
5400  : wolfSSL Entering DecodeSubjKeyId
5404  : wolfSSL Entering GetObjectId()
5408  : wolfSSL Entering DecodeAuthKeyId
5412  : wolfSSL Entering GetObjectId()
5414  : wolfSSL Entering DecodeBasicCaConstraint
5418  : wolfSSL Entering GetAlgoId
5422  : wolfSSL Entering GetObjectId()
5426  : Error adding trusted peer
5428  : wolfSSL error occurred, error = -150
5432  : CA Parse failed, with progress in file.
5436  : Search for other certs in file
[APP]Error loading cert, -150
```

WolfSSL v 4.8.1

verifyCallback is never called!!

Xeenych, 2021-09-17:
Perhaps in ssl.c line 5704 should be changed from
```

ret = AddTrustedPeer(ctx->cm, &der, !ctx->verifyNone);
```
to
```
ret = AddTrustedPeer(ctx->cm, &der, verify);
```
and certificate can be loaded with
ProcessBuffer(ctx, certBuf, certBufSz, WOLFSSL_FILETYPE_PEM, TRUSTED_PEER_TYPE, NULL, NULL, 0, VERIFY_SKIP_DATE)

dgarske, 2021-09-17:
Hi @Xeenych ,

Curious why are you trying to use the trusted peer type? 

Typically we load trusted certificates (root, intermediate or leaf certs) using `wolfSSL_CTX_load_verify_buffer_ex` and provide the `WOLFSSL_LOAD_FLAG_DATE_ERR_OKAY` flag to override a date error. Another option is to 100% disable certificate date checking by using `NO_ASN_TIME`, however using this comes with risks, since no certificates will have their validity dates checked.

Example:

```
ret = wolfSSL_CTX_load_verify_buffer_ex(ctx, ca_expired_cert, sizeof(ca_expired_cert), WOLFSSL_FILETYPE_ASN1, 0, WOLFSSL_LOAD_FLAG_DATE_ERR_OKAY);
```

The verify callback is only issued for certificates provided during a TLS handshake from the peer. You can at build time define these to issue additional callback cases:

```
#define WOLFSSL_ALWAYS_VERIFY_CB /* Always call verify callback (configured via wolfSSL_CTX_set_verify API) */
#define WOLFSSL_VERIFY_CB_ALL_CERTS /* Call verify callback for all intermediate certs */
```

Thanks,
David Garske, wolfSSL

Xeenych, 2021-09-17:
What is it so bad to use wolfSSL_CTX_trust_peer_buffer()?

I have a private server to connect to with my self-signed certificate. I have no need to check certificate chains, so I use wolfSSL_CTX_trust_peer_buffer();
Am I right?

When using wolfSSL_CTX_load_verify_buffer_ex the connection handshake takes a lot more time.

ejohnstown, 2021-09-20:
Using a trusted peer certificate skips all checking, including the signature.

Since you are using a self-signed certificate, the signature on the certificate is checked, hence the delay.

Is this an application where you can use a pre-shared key? Both endpoints will have a shared secret. You still get authentication as both ends need to know the shared secret. You can use ECDHE or DHE for key-agreement, and still maintain PFS.

Xeenych, 2021-09-20:
Yes, we are considering to use PSK, but at first we made some performance tests with certificates.

ejohnstown, 2022-11-24:
When you try loading a trusted cert, with `wolfSSL_CTX_trust_peer_buffer()`, it calls `ProcessBuffer()` with the type set to `TRUSTED_PEER_TYPE`. For that type, we do not call the verify callback. Right now, that's as-intended. The date validity is still checked. (I was wrong in my earlier message.)

For self-signed certificates, we load the certificate as a CA with `wolfSSL_CTX_load_verify_*()`. It ends up checking the signature on the certificate with the key in the certificate. I believe that's the slowdown you are seeing. But it will give you the chance to bypass the date check using @dgarske's example.

closed 2022-11-24 by ejohnstown