wolfSSL/wolfssl #4417: Alert handling

devbharat, 2021-09-20:
In internal.c  in function 

```
static int DoAlert(WOLFSSL* ssl, byte* input, word32* inOutIdx, int* type,
                   word32 totalSz)
```

in the following lines

```
    /* make sure can read the message */
    if (dataSz != ALERT_SIZE) {
#ifdef WOLFSSL_EXTRA_ALERTS
        SendAlert(ssl, alert_fatal, unexpected_message);
#endif
        printf("dataSz %d != ALERT_SIZE %d\n", dataSz, ALERT_SIZE);
        return BUFFER_E;
    }
```

I do not understand why the number of bytes in the internal buffer are expected to be the same as the fixed value of ALERT_SIZE. When pulling in data into the internal buffer from an IOcallback, you'd pull in an arbitrary number of bytes upto a specified number of max value, so dataSz here is not fixed, can be anything actually depending on what else is buffered up after the alert bytes, so comparing it to fixed ALERT_SIZE here (2?!) does not make sense to me. Would appreciate an explanation.

dgarske, 2021-09-20:
Hi @devbharat ,

Note: this code was changed in PR https://github.com/wolfSSL/wolfssl/pull/4377 from `if (*inOutIdx + ALERT_SIZE > totalSz)
` -> `dataSz != ALERT_SIZE`.

@SparkiDev , do you have any thoughts on this check?

Thanks,
David Garske, wolfSSL

SparkiDev, 2021-09-20:
Hi @devbharat,

While it may appear that totalSz is going to be any length, the caller only loads the data for that TLS record.
The change was made for security reasons - to ensure only 2 byte alerts were being seen.
Code was added to handle encrypted alerts tool.

devbharat, 2021-09-21:
Thanks for the reply. How/where do you ensure that the caller is loading data only for one record at a time?

i am working with DTLS and memory buffers (with serial device forming the transport layer), and we push all incoming data on the serial device into the memory buffer and let DTLS iocallback pullout as many bytes as it wants from the memory buffer. For DTLS it seems to want to pullout dtls_expected_rx (MTU=512 for us) bytes from the buffer on each iorecv, and while it’s not always 512 bytes, it’s usually more than the data from 1 single record.

SparkiDev, 2021-09-21:
Hi @devbharat

With reference to master, in the function ProcessReplyEx, looking at the code in the 'getData' case of the switch, you can see two calls to GetInputData. These calls are passed ssl->curSize and will only read as much as that.

ssl->curSize is set from the call to GetRecordHeader() where it is passed as an output parameter. That is, GetRecordHeader() gets the size of the record data from the record header and sets into size which is ssl->curSize.

Does this make sense?

devbharat, 2021-09-22:
Hey @SparkiDev 

In the call to `GetInputData()` for DTLS the size is overwritten to `dtls_expected_rx` as shown below

```
static int GetInputData(WOLFSSL *ssl, word32 size)
{
    int in;
    int inSz;
    int maxLength;
    int usedLength;
    int dtlsExtra = 0;

    /* check max input length */
    usedLength = ssl->buffers.inputBuffer.length - ssl->buffers.inputBuffer.idx;
    maxLength  = ssl->buffers.inputBuffer.bufferSize - usedLength;
    inSz       = (int)(size - usedLength);      /* from last partial read */

#ifdef WOLFSSL_DTLS
    if (ssl->options.dtls) {
        if (size < ssl->dtls_expected_rx)
            dtlsExtra = (int)(ssl->dtls_expected_rx - size);
        inSz = ssl->dtls_expected_rx;
    }
#endif

```

SparkiDev, 2021-09-22:
Hi @devbharat

Yes, the GetInputData function can read more than the current record with DTLS.
I'm not familiar with the DTLS differences and there is an issue here.

I've put up a pull request that fixes this: https://github.com/wolfSSL/wolfssl/pull/4421

Let us know if you are happy with the fix!

Thanks

devbharat, 2021-09-23:
Could someone explain why it is 'allowed' to buffer up more bytes than just the current record in the internal buffer for DTLS but not for TLS? I am afraid the PR just fixes this particular check for alert size, but there might be other consequences of buffering up more bytes than the current record.

ejohnstown, 2021-09-23:
DTLS allows one to put as many DTLS records as you want into a datagram. You can use the fragmentation mechanism to fill the datagram. This is used during the handshake by some DTLS implementations to send as few datagrams as possible. When you call recvfrom(), you are going to get the entire datagram, which will have at least one DTLS record in it.

From RFC section 4.1.1:
```
   Multiple DTLS records may be placed in a single datagram.  They are
   simply encoded consecutively.  The DTLS record framing is sufficient
   to determine the boundaries.  Note, however, that the first byte of
   the datagram payload must be the beginning of a record.  Records may
   not span datagrams.
```

wolfSSL will typically send a single a record per transmit or datagram. During the handshake, it'll put the change cipher spec message and the finished handshake message into a single transmit.