wolfSSL/wolfssl #4426: Enforced  cmake option strings breaks downstream projects scripts

rmarquis, 2021-09-24:
Enforcing cmake option strings as done in https://github.com/wolfSSL/wolfssl/commit/cd9f400cf33d17af84472733f9f4f83c7da098f5 has the naughty side effect of breaking CMake scripts of user projects.

Not only the `ON/OFF` syntax doesn't work anymore, but the capitalized `YES/NO` doesn't work either. We have to use the special `yes/no` syntax *just* for the WolfSSL part.

While I understand the desire to unify the syntax used through your whole project, is there any reason why this should affects (potentially dozens or hundred) downstream projects that embed WolfSSL? Why not use "yes/no" within CMake files of WolfSSL while allowing users to use whatever they are already using in their pipeline?

embhorn, 2021-09-24:
Hi @rmarquis 

Thanks for your feedback. We will review and post an update.

haydenroche5, 2021-09-24:
Hi @rmarquis . I agree with your reasoning. Thanks for raising this issue. I'll put up a PR soon to support all the normal CMake true/false literals in our CMake build options interface. Internally, we will reduce all those values that evaluate to `true` to `yes`. Same goes for `false` and `no`. This should prevent the breakage you're describing (e.g. `ON/OFF` and `YES/NO` will work as expected).

rmarquis, 2021-09-24:
Thank you for looking at this issue!

I was quite surprise to see our CI break when I updated to the latest WolfSSL release.

haydenroche5, 2021-09-24:
@rmarquis no problem. Out of curiosity, what is your project? I'm just interested to see how you're using wolfSSL.

rmarquis, 2021-09-24:
See https://www.etlegacy.com/, some oldie FPS that was quite popular back in the day.

It's using SSL for in-game pack download, either OpenSSL or WolfSSL. I'm not the one that did the implementation, so I might not know about details - I'm just the janitor that update the game libs every few months, and get desperate when everything inevitably breaks :)

I first found the issue through our automated Coverity Scan github action, which complained that it couldn't succeed anymore.

haydenroche5, 2021-09-24:
Wow, that takes me back. I used to play that game waaaay back in the day. It's very cool that the project's using wolfSSL.

I'll post a link to the PR here when it's ready. Thanks for your patience.

rmarquis, 2021-09-24:
There's still a sizeable active community of players, if you ever feel nostalgic :)

haydenroche5, 2021-09-24:
Hi @rmarquis. I've added another commit to this existing PR that should fix your issue: https://github.com/wolfSSL/wolfssl/pull/4422 Let me know if it works for you. Thanks!

rmarquis, 2021-09-25:
That should work, yes! We did necessary change as our next release will ship soon with WolfSSL 4.8.1, but I'm glad to see this fixed for future WolfSSL version.
Thank you again for your time!

closed 2021-09-25 by rmarquis