wolfSSL/wolfssl #4443: WolfSSL seems to be unable to use Let's encrypt certificates after DST Root CA X3 Expiration

jaapcrezee, 2021-10-01:
[chain.txt](https://github.com/wolfSSL/wolfssl/files/7267766/chain.txt)

Using wolfssl 66a6153456b04d9bccd079e3b6f69a4f86ca4a58 (master as of writing) using:
`./autogen.sh && ./configure && make`

Now the examples/client fails which is unexpected (for me):

```
[jaap@jaap ~/src/wolfssl/examples/client ]$ ./client -h letsencrypt.org -p 443 -A /tmp/chain.txt 
wolfSSL_connect error -188, ASN no signer error to confirm failure
wolfSSL error: wolfSSL_connect failed
```

jaapcrezee, 2021-10-01:
Curl is doing fine on the same chain.txt

```
[jaap@jaap ~/src/wolfssl/examples/client ]$ curl -v -o /dev/null --cacert /tmp/chain.txt https://letsencrypt.org
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 18.159.128.50:443...
* Connected to letsencrypt.org (18.159.128.50) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /tmp/chain.txt
*  CApath: none
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [3904 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [80 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [52 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [52 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=lencr.org
*  start date: Aug  6 01:03:53 2021 GMT
*  expire date: Nov  4 01:03:51 2021 GMT
*  subjectAltName: host "letsencrypt.org" matched cert's "letsencrypt.org"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multiplexing
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x55a9ca4289a0)
} [5 bytes data]
> GET / HTTP/2
> Host: letsencrypt.org
> user-agent: curl/7.78.0
> accept: */*
> 
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [81 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS == 4294967295)!
} [5 bytes data]
< HTTP/2 200 
< cache-control: public, max-age=0, must-revalidate
< content-security-policy: default-src 'none'; font-src 'self'; style-src 'unsafe-inline' 'self'; script-src 'unsafe-eval' 'unsafe-inline' 'self' data: https://www.google-analytics.com https://www.googleadservices.com https://www.googletagmanager.com https://googleads.g.doubleclick.net https://donorbox.org https://js.stripe.com/v3/ https://sdks.shopifycdn.com ; img-src 'self' data: blob: https://www.google-analytics.com https://www.paypal.com https://www.paypalobjects.com https://ak2s.abmr.net https://ak1s.abmr.net https://www.google.com https://cdn.shopify.com https://v.shopify.com ; frame-src https://donorbox.org https://www.youtube.com https://www.youtube-nocookie.com https://bid.g.doubleclick.net https://js.stripe.com/v3/ https://js.stripe.com/v2/ ; connect-src 'self' https://d4twhgtvn0ff5.cloudfront.net/ https://letsencrypt-merch.myshopify.com https://monorail-edge.shopifysvc.com ;
< content-type: text/html; charset=UTF-8
< date: Fri, 01 Oct 2021 13:48:40 GMT
< etag: "072887f8cb3a90eda7db29bbef054255-ssl"
< permissions-policy: geolocation=(), midi=(), notifications=(), push=(), sync-xhr=(), microphone=(), camera=(), magnetometer=(), gyroscope=(), speaker=(self), vibrate=(), fullscreen=(self), interest-cohort=()
< referrer-policy: no-referrer
< strict-transport-security: max-age=31536000
< x-xss-protection: 1; mode=block
< age: 2182
< server: Netlify
< x-nf-request-id: 01FGY3X17Y5T6V9KT85SCR64MV
< content-length: 32283
< x-content-type-options: nosniff
< x-frame-options: DENY
< 
{ [5 bytes data]
100 32283  100 32283    0     0   434k      0 --:--:-- --:--:-- --:--:--  437k
* Connection #0 to host letsencrypt.org left intact
```

dhewg, 2021-10-03:
The stable versions are affected too and a potentially fix should be backported:
https://forum.openwrt.org/t/21-02-0-cannot-connect-to-any-letsencrypt-site-using-ssl-letsencrypt-global-root-cert-expiration/107925

It looks like the master branch is not affected, but this epic merge mess made me give up finding the commit fixing the issue...

kabuobeid, 2021-10-04:
Hi @jaapcrezee / @dhewg ,

I am able to reproduce this behavior here.  letsencrypt.org is cross-signing its CA (ISRG Root X1) with DST Root CA X3 using subject key identifier `C4:A7:B1:A4:7B:2C:71:FA:DB:E1:4B:90:75:FF:C4:15:60:85:89:10`.  There is no certificate with this SKI which is not expired, as you can see here: https://crt.sh/?ski=c4a7b1a47b2c71fadbe14b9075ffc41560858910  You can disable our date check and load DST Root CA X3, but I would not recommend it.

The subdomain `valid-isrgrootx1.letsencrypt.org` is using ISRG Root X1 as a true CA rather than a cross-signed CA.  Enable SNI by passing `--enable-sni` to ./configure and using the `-S` flag to our client example and you will see the X1 CA verify as you'd expect: `./examples/client/client -h valid-isrgrootx1.letsencrypt.org -S valid-isrgrootx1.letsencrypt.org -p 443 -A isrgrootx1.pem`

curl is using OpenSSL which is less strict with certificate verification by default.  It will still connect even if it runs into most certificate errors by default.  wolfSSL has the opposite behavior of failing on any certificate error by default.  A cross-signed CA is just an intermediate cert from our perspective and not suitable for verification, we need a valid root CA verified to connect.  I would contact letsencrypt.org and confirm why they're using an outdated certificate chain.

Thanks,
Kareem

jaapcrezee, 2021-10-04:
Hello @kabuobeid ,
Thank you for the explainer. I think the problem is - in my case (for other domains than letsencrypt.org) - the letsencrypt acme client I am using is old and thus giving me the old/previous chain with the cross-signed CA cert.
But as I am (mostly) only using wolfSSL in embedded devices I will now setup to my own private CA-PKI to prevent problems like this in the future.

Jaap

dhewg, 2021-10-05:
Thanks @jaapcrezee for the analysis.

But I think this is intentional by let's encrypt:
https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

If I understood you correctly, the cert itself is from a valid CA, it's just the additional cross-signing that makes wolfssl fail its checks.

OpenWrt uses curl with wolfssl by default, and the current situation breaks tons of websites.
Disabling the date check globally is obviously not the smart way forward, but if wolfssl can't be adapted to succeed its cert checks for this situation, can we at least have a flag/api or whatever for this? Maybe a build time option or an option which curl can set at runtime?

dhewg, 2021-10-05:
not literally but something a la ---enable-accepting-valid-ca-event-with-expired-cross-signing ;)

kabuobeid, 2021-10-05:
The issue is it's being cross-signed by an outdated X3 certificate, found here: https://letsencrypt.org/certs/trustid-x3-root.pem.txt
The communication from Let's Encrypt claims there is an updated X3 certificate for Android devices but I haven't been able to locate it.  The linked IdenTrust certificate found here does not validate the chain: https://www.identrust.com/node/1330
Let's Encrypt needs to use a root CA which is not outdated.
We are not currently planning to add support for cross-signed CAs.

pgporada, 2021-10-05:
> The communication from Let's Encrypt claims there is an updated X3 certificate for Android devices but I haven't been able to locate it.

I think this is the document you want https://letsencrypt.org/2020/12/21/extending-android-compatibility.html

osirisinferi, 2021-10-05:
> Let's Encrypt needs to use a root CA which is not outdated.

That is by design.

> We are not currently planning to add support for cross-signed CAs with invalid root certs.

Note that many (all major) root certificate stores have `ISRG Root X1` included. So it's very possible to build a valid chain from R3 to the ISRG Root X1 in the trust stores.

kabuobeid, 2021-10-05:
I've researched this some more and Let's Encrypt should work if you load `ISRG Root X1` and define `WOLFSSL_ALT_CERT_CHAINS`.  You also still need SNI enabled as mentioned above.  So you would build and run as follows:
`CFLAGS="-DWOLFSSL_ALT_CERT_CHAINS" ./configure --enable-sni`
`./examples/client/client -h letsencrypt.org -p 443 -S letsencrypt.org -A /path/to/X1.pem`
`WOLFSSL_ALT_CERT_CHAINS` allows us to replace the cross-signed X1 cert in the chain with the given self-signed X1 cert.

Let me know if this works for you.

dhewg, 2021-10-05:
Oh, thanks, I'll look into it.

Meanwhile I cooked up this patch: 
[dstrootcax3.patch.txt](https://github.com/wolfSSL/wolfssl/files/7289561/dstrootcax3.patch.txt)

Seems to be doing what I think it's doing, works for me (tm) so far...

jaapcrezee, 2021-10-06:
I can confirm `WOLFSSL_ALT_CERT_CHAINS` works (tested it before), but does it really not "degrade" any other security aspects?

dhewg, 2021-10-06:
Same here, WOLFSSL_ALT_CERT_CHAINS fixes curl without any additional patches or options.
But can OpenWrt just use that option? It's unclear what implication this might have

dhewg, 2021-10-06:
FYI, I raised this issue on the LE forums too
https://community.letsencrypt.org/t/wolfssl-cert-issue/162212

I don't know what the way forward is, but I believe we need a solution soon (which may be just WOLFSSL_ALT_CERT_CHAINS).
We've only seen the first few days of LE cert renewals. For WolfSSL only distros this could feel like "half the internet is broken" in a few weeks if this stays as-is.

dhewg, 2021-10-06:
Thanks David!

That sounds like a really useful explanation to add to the documentation ;)
Or well, maybe it's written down somewhere, but the only thing I could find is `Alternate certificate chain support WOLFSSL_ALT_CERT_CHAINS. Enables checking cert against multiple CA's.` which doesn't help much.

So that indeed is the flag that OpenWrt should be using, thanks everyone!

kabuobeid, 2021-10-06:
We do document `WOLFSSL_ALT_CERT_CHAINS` in `src/internal.c`, see below:
```
 * WOLFSSL_ALT_CERT_CHAINS:
 *     Allows CA's to be presented by peer, but not part of a valid chain.
 *     Default wolfSSL behavior is to require validation of all presented peer
 *     certificates. This also allows loading intermediate CA's as trusted
 *     and ignoring no signer failures for CA's up the chain to root.
```

elms, 2021-10-06:
Related PR to make this option more accessible https://github.com/wolfSSL/wolfssl/pull/4455

closed 2021-10-08 by kabuobeid