wolfSSL/wolfssl #4450: AES-GCM should not require authentication tags

pprindeville, 2021-10-05:
Looking at `wc_AesGcmDecrypt()` and `wc_AesGcmEncrypt()`, a valid scenario should be `(authTag == NULL && authTagSz == 0)` but this isn't handled.

My reading of the [AES-GCM specification ](https://nvlpubs.nist.gov/nistpubs/Legacy/SP/nistspecialpublication800-38d.pdf) section 7.1 implies that `A` might be the zero-length bit string:

> In Steps 4 and 5, the AAD and the ciphertext are each appended with the minimum number of ‘0’ bits, possibly none, so that the bit lengths of the resulting strings are multiples of the block size. The concatenation of these strings is appended with the 64-bit representations of the lengths of the AAD and the ciphertext, and the GHASH function is applied to the result to produce a single output block.

and Figure 3.  In this case, "possibly none" is also tacitly "possibly all".

That is, if the AAD (denoted `A`) is zero-length, then `v` bits of padding (where `v` is the block length) are appended.

Similarly for section 7.2:

> In Steps 5 and 6, the AAD and the ciphertext are each appended with the minimum number of ‘0’ bits, possibly none, so that the bit lengths of the resulting strings are multiples of the block size. The concatenation of these strings is appended with 64-bit representations of the lengths of the AAD and the ciphertext, and the GHASH function is applied to the result to produce a single output block.

and Figure 4.

Lastly, see "Appendix C: Requirements and Guidelines for Using Short Tags".

And [RFC 5084](https://datatracker.ietf.org/doc/html/rfc5084) clearly states:

> AES-GCM has four inputs: an AES key, an initialization vector (IV), a plaintext content, and optional additional authenticated data (AAD).  AES-GCM generates two outputs: a ciphertext and message

Indeed, we've confirmed that `pyca` running on Openssl-1.1 on MacOS will happily handle the case of no Authentication Tag, and the [documentation](https://cryptography.io/en/latest/hazmat/primitives/aead/?highlight=AESGCM#cryptography.hazmat.primitives.ciphers.aead.AESGCM) states:

> *associated_data* (_bytes_) – Additional data that should be authenticated with the key, but is not encrypted. Can be `None`.

pprindeville, 2021-10-05:
See PR #4453.

elms, 2021-10-07:
@pprindeville I'm trying to understand the use case little better. I've recreated what I think you are suggesting (with a terrible key and iv for example only)

```python
from cryptography.hazmat.primitives.ciphers.aead import AESGCM
aesgcm = AESGCM(16*b'\x00')
data = b"a secret message"
ct = aesgcm.encrypt(12*b'\x00', data, None)
' '.join([hex(x) for x in ct[:16]])
' '.join([hex(x) for x in ct[16:]])
```

output:
```
'0x62 0xa8 0xa9 0xab 0x3 0xc4 0xc6 0xe6 0xd3 0x45 0xa7 0xca 0x2 0xd3 0x99 0x1d'
'0x10 0xae 0xa3 0x4d 0xe 0x90 0xb0 0x57 0x2f 0xa 0x7a 0x86 0xcc 0xef 0x90 0x60'
```

The equivalent calls using wolfSSL

```c
    Aes    aes;
    byte key[16] = {0};
    byte iv[12] = {0};
    byte in[16] = "a secret message"; 
    byte out[16] = {0};
    byte out2[16] = {0};
    byte authTag[16] = {0};
    
    /* sets key */
    int ret = wc_AesGcmSetKey(&aes, key, sizeof(key));
    if (ret != 0) {
      printf("setkey failed: %d\n", ret);
      exit(1);
    }

    ret = wc_AesGcmEncrypt(&aes, out,
                           in, sizeof(in),
                           iv, sizeof(iv),
                           authTag, sizeof(authTag),
                           NULL, 0);
    if (ret != 0) {
      printf("Encrypt failed: %d\n", ret);
      exit(1);
    }

    for(int ii=0; ii<sizeof(out); ii++) {
      printf("0x%x ", out[ii]);
    }
    printf("\n");
    for(int ii=0; ii<sizeof(authTag); ii++) {
      printf("0x%x ", authTag[ii]);
    }
    printf("\n");
```

with output (identical to pyca output)

```
0x62 0xa8 0xa9 0xab 0x3 0xc4 0xc6 0xe6 0xd3 0x45 0xa7 0xca 0x2 0xd3 0x99 0x1d 
0x10 0xae 0xa3 0x4d 0xe 0x90 0xb0 0x57 0x2f 0xa 0x7a 0x86 0xcc 0xef 0x90 0x60
```

In the case of wolfSSL the `authTag` is placed in a separate buffer, but pyca appends the tag to the output ([link](https://cryptography.io/en/latest/hazmat/primitives/aead/?highlight=AESGCM#cryptography.hazmat.primitives.ciphers.aead.AESGCM.encrypt))

>Returns bytes: The ciphertext bytes with the 16 byte tag appended.

If I am misunderstanding, can you provide an example using pyca (or similar) that is representative of the other end of the channel?

kareem-wolfssl, 2021-11-23:
Closing as per feedback in https://github.com/wolfSSL/wolfssl/pull/4453.  Feel free to re-open if you disagree.

closed 2021-11-23 by kareem-wolfssl