wolfSSL/wolfssl #4526: sss_se05x_key_store_generate_key() support for 521 curve

MalikAlmazrua, 2021-11-02:
in se050_ecc_create_key() function , the function that generates the keys and its parameters are shown below:

status = sss_key_store_generate_key(&host_keystore, &keyPair, keySize*8, NULL); 

the keybitlength parameter is defined to be 8 times the private key size. This is usually correct but not the case for the 521 curve. 

The curve's private key size is 66. 8 times 66 is 528 NOT 521. Therefore, the function returns an error whenever the curve 521 is used.

dgarske, 2021-11-02:
Hi @MalikAlmazrua ,

Thank you for the report. This is a very minor issue and I will push a fix.

Related to ZD 13147 and PR https://github.com/wolfSSL/wolfssl/pull/4523 

Thanks,
David Garske, wolfSSL

dgarske, 2024-03-18:
Hi @MalikAlmazrua ,

It looks like changes were pushed to #4526 to resolve this. Can you please confirm if your issue is solved? If so you are welcome to close this issue.

Thanks,
David Garske, wolfSSL

closed 2024-08-22 by dgarske