wolfSSL/wolfssl #4537: Is it compatible with aws-iot-device-sdk-embedded-C?

PierrePeng, 2021-11-04:
Hi, 
1. I wanna whether it is compatible with [aws-iot-device-sdk-embedded-C](https://github.com/aws/aws-iot-device-sdk-embedded-C)? 
2. Can I use wolfSSL instead of OpenSSL which is required from this [line](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/a54643fe39bc90a964c96716bc8055097d099ab5/platform/CMakeLists.txt#L5)?
3. If Yes, what steps should I do to replace OpenSSL in aws-iot-device-sdk-embedded-C project?

kareem-wolfssl, 2021-11-04:
Hi PierrePeng,

1. We do not currently support this, but it doesn't appear that it would be difficult to add support at a glance.
2. Yes, I would suggest using our OpenSSL compatibility layer if you intend to use us as a drop-in replacement for OpenSSL.  Either define `OPENSSL_EXTRA` for most of our compatibility layer, or `OPENSSL_ALL` for everything, if you aren't concerned about library size.
3. See above.  With our OpenSSL compatibility layer active, everything should just work. 
4. Yes, we appear to be missing this define.  You can manually define it as a temporary workaround, until this is implemented.

Can you tell me a bit more about your project?  Is this a personal or commercial project?  What platform are you targetting?

Thanks,
Kareem

dgarske, 2021-11-04:
Hi PierrePeng,

If you are not comfortable sharing this information publicly feel free to directly email support@wolfssl.com and reference this GitHub issue.

Thanks,
David Garske, wolfSSL

closed 2021-11-23 by kareem-wolfssl