wolfSSL/wolfssl #4546: Can not build python 3.8.5 with wolfssl

pg83, 2021-11-07:
`/home/pg/mix/build/c46d6d08a1769d7bf84dad6b5dd72bf4/tools/clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall    -std=c99 -Wextra -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -Wstrict-prototypes -Werror=implicit-function-declaration  -I./Include/internal  -I. -I./Include    -DPy_BUILD_CORE_BUILTIN  -c ./Modules/_hashopenssl.c -o Modules/_hashopenssl.o
In file included from ./Modules/_hashopenssl.c:23:
In file included from /home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/evp.h:33:
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/wolfcrypt/settings.h:2329:14: warning: "For timing resistance / side-channel attack prevention consider using harden options" [-W#warnings]
            #warning "For timing resistance / side-channel attack prevention consider using harden options"
             ^
./Modules/_hashopenssl.c:30:4: error: "OPENSSL_THREADS is not defined, Python requires thread-safe OpenSSL"
#  error "OPENSSL_THREADS is not defined, Python requires thread-safe OpenSSL"
   ^
./Modules/_hashopenssl.c:35:9: warning: 'EVP_MD_CTX_new' macro redefined [-Wmacro-redefined]
#define EVP_MD_CTX_new EVP_MD_CTX_create
        ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/evp.h:831:9: note: previous definition is here
#define EVP_MD_CTX_new          wolfSSL_EVP_MD_CTX_new
        ^
./Modules/_hashopenssl.c:36:9: warning: 'EVP_MD_CTX_free' macro redefined [-Wmacro-redefined]
#define EVP_MD_CTX_free EVP_MD_CTX_destroy
        ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/evp.h:833:9: note: previous definition is here
#define EVP_MD_CTX_free         wolfSSL_EVP_MD_CTX_free
        ^
./Modules/_hashopenssl.c:77:15: error: implicit declaration of function 'wolfSSL_ERR_peek_last_error' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    errcode = ERR_peek_last_error();
              ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/err.h:31:42: note: expanded from macro 'ERR_peek_last_error'
#define ERR_peek_last_error              wolfSSL_ERR_peek_last_error
                                         ^
./Modules/_hashopenssl.c:77:15: note: did you mean 'wolfSSL_ERR_peek_error'?
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/err.h:31:42: note: expanded from macro 'ERR_peek_last_error'
#define ERR_peek_last_error              wolfSSL_ERR_peek_last_error
                                         ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/ssl.h:1930:27: note: 'wolfSSL_ERR_peek_error' declared here
WOLFSSL_API unsigned long wolfSSL_ERR_peek_error(void);
                          ^
./Modules/_hashopenssl.c:82:5: error: implicit declaration of function 'ERR_clear_error' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    ERR_clear_error();
    ^
./Modules/_hashopenssl.c:84:11: error: implicit declaration of function 'ERR_lib_error_string' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    lib = ERR_lib_error_string(errcode);
          ^
./Modules/_hashopenssl.c:84:9: warning: incompatible integer to pointer conversion assigning to 'const char *' from 'int' [-Wint-conversion]
    lib = ERR_lib_error_string(errcode);
        ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./Modules/_hashopenssl.c:85:12: error: implicit declaration of function 'ERR_func_error_string' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    func = ERR_func_error_string(errcode);
           ^
./Modules/_hashopenssl.c:85:12: note: did you mean 'ERR_lib_error_string'?
./Modules/_hashopenssl.c:84:11: note: 'ERR_lib_error_string' declared here
    lib = ERR_lib_error_string(errcode);
          ^
./Modules/_hashopenssl.c:85:10: warning: incompatible integer to pointer conversion assigning to 'const char *' from 'int' [-Wint-conversion]
    func = ERR_func_error_string(errcode);
         ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./Modules/_hashopenssl.c:86:14: error: implicit declaration of function 'ERR_reason_error_string' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    reason = ERR_reason_error_string(errcode);
             ^
./Modules/_hashopenssl.c:86:12: warning: incompatible integer to pointer conversion assigning to 'const char *' from 'int' [-Wint-conversion]
    reason = ERR_reason_error_string(errcode);
           ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./Modules/_hashopenssl.c:120:10: error: use of undeclared identifier 'NID_sha256'
    case NID_sha256:
         ^
./Modules/_hashopenssl.c:123:10: error: use of undeclared identifier 'NID_sha384'
    case NID_sha384:
         ^
./Modules/_hashopenssl.c:126:10: error: use of undeclared identifier 'NID_sha512'
    case NID_sha512:
         ^
./Modules/_hashopenssl.c:173:16: error: implicit declaration of function 'wolfSSL_OBJ_nid2ln' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        name = OBJ_nid2ln(nid);
               ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/objects.h:47:21: note: expanded from macro 'OBJ_nid2ln'
#define OBJ_nid2ln  wolfSSL_OBJ_nid2ln
                    ^
./Modules/_hashopenssl.c:173:16: note: did you mean 'wolfSSL_BN_bin2bn'?
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/objects.h:47:21: note: expanded from macro 'OBJ_nid2ln'
#define OBJ_nid2ln  wolfSSL_OBJ_nid2ln
                    ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/bn.h:100:29: note: 'wolfSSL_BN_bin2bn' declared here
WOLFSSL_API WOLFSSL_BIGNUM* wolfSSL_BN_bin2bn(const unsigned char*, int len,
                            ^
./Modules/_hashopenssl.c:173:14: warning: incompatible integer to pointer conversion assigning to 'const char *' from 'int' [-Wint-conversion]
        name = OBJ_nid2ln(nid);
             ^ ~~~~~~~~~~~~~~~
./Modules/_hashopenssl.c:175:20: error: implicit declaration of function 'wolfSSL_OBJ_nid2sn' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
            name = OBJ_nid2sn(nid);
                   ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/objects.h:44:21: note: expanded from macro 'OBJ_nid2sn'
#define OBJ_nid2sn  wolfSSL_OBJ_nid2sn
                    ^
./Modules/_hashopenssl.c:175:20: note: did you mean 'wolfSSL_OBJ_nid2ln'?
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/objects.h:44:21: note: expanded from macro 'OBJ_nid2sn'
#define OBJ_nid2sn  wolfSSL_OBJ_nid2sn
                    ^
./Modules/_hashopenssl.c:173:16: note: 'wolfSSL_OBJ_nid2ln' declared here
        name = OBJ_nid2ln(nid);
               ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/objects.h:47:21: note: expanded from macro 'OBJ_nid2ln'
#define OBJ_nid2ln  wolfSSL_OBJ_nid2ln
                    ^
./Modules/_hashopenssl.c:175:18: warning: incompatible integer to pointer conversion assigning to 'const char *' from 'int' [-Wint-conversion]
            name = OBJ_nid2sn(nid);
                 ^ ~~~~~~~~~~~~~~~
./Modules/_hashopenssl.c:1115:5: error: implicit declaration of function 'wolfSSL_ERR_load_crypto_strings' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    ERR_load_crypto_strings();
    ^
/home/pg/mix/store/89c4c0cdfefcc34cc148a47edf181546-lib-wolfssl/include/wolfssl/openssl/err.h:29:42: note: expanded from macro 'ERR_load_crypto_strings'
#define ERR_load_crypto_strings          wolfSSL_ERR_load_crypto_strings
                                         ^
8 warnings and 12 errors generated.
make: *** [Makefile:2054: Modules/_hashopenssl.o] Error 1
`

kareem-wolfssl, 2021-11-08:
Hi @pg83 ,

How are you building Python + wolfSSL?  You will need to patch Python and use the right config for wolfSSL, you can find full instructions here: https://github.com/JacobBarthelmeh/osp/tree/Python/Python
Let me know if you still see issues with those instructions.

Thanks,
Kareem

pg83, 2021-11-09:
>You will need to patch Python and use the right config for wolfSSL, you can find full instructions here: wolfSSL/osp#81

Oh, thank you, this is just I need :)

closed 2021-11-09 by pg83