wolfSSL/wolfssl #4547: Build Error

janoschreinking, 2021-11-08:
Version: wolfSSL Release 5.0.0 (Nov 01, 2021)

[wolfssl/examples/configs/user_settings_min_ecc.h
](https://github.com/wolfSSL/wolfssl/blob/master/examples/configs/user_settings_min_ecc.h) contains a cli command:

```
./configure \
    --enable-cryptonly --enable-ecc --enable-sp \
    --disable-rsa --disable-dh --disable-sha3 --disable-sha224 --disable-md5 \
    --disable-sha --disable-pkcs12 --disable-memory \
    --disable-chacha --disable-poly1305 --disable-sha512 --disable-sha384 \
    --disable-aesgcm --disable-aescbc --disable-aes --disable-rng \
    CFLAGS="-DNO_SIG_WRAPPER -DWOLFSSL_PUBLIC_MP -DECC_USER_CURVES \
    -DNO_ECC_SIGN -DNO_ECC_DHE -DNO_ECC_KEY_EXPORT"
```

which yields in this error:

```
wolfcrypt/benchmark/benchmark.c: In function ‘bench_ecc’:
wolfcrypt/benchmark/benchmark.c:5361:20: error: unused parameter ‘doAsync’ [-Werror=unused-parameter]
 void bench_ecc(int doAsync, int curveId)
                ~~~~^~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:5077: wolfcrypt/benchmark/benchmark.o] Error 1
make[2]: *** Waiting for unfinished jobs....
wolfcrypt/src/sp_c64.c:33767:13: error: ‘sp_384_mont_inv_order_7’ defined but not used [-Werror=unused-function]
 static void sp_384_mont_inv_order_7(sp_digit* r, const sp_digit* a,
             ^~~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/sp_c64.c:26519:13: error: ‘sp_256_mont_inv_order_5’ defined but not used [-Werror=unused-function]
 static void sp_256_mont_inv_order_5(sp_digit* r, const sp_digit* a,
             ^~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

kareem-wolfssl, 2021-11-08:
Hi @janoschreinking ,

Thanks for the report, I am able to reproduce it on both 5.0 and master.
For the doAsync warning, please try pulling this PR: https://github.com/wolfSSL/wolfssl/pull/4550  You will also need this PR to fix a compilation issue in test.c: https://github.com/wolfSSL/wolfssl/pull/4535
@SparkiDev please take a look at these SP math warnings.  I can confirm both of these functions are unused in this configuration, looks like they need to only be included if HAVE_ECC_SIGN or WOLFSSL_SP_SMALL are defined.

Thanks,
Kareem

SparkiDev, 2021-11-09:
Hi @janoschreinking

A pull request has been put up to fix this: https://github.com/wolfSSL/wolfssl/pull/4553
Let us know if you have problems with this.

Thanks.
Sean

closed 2021-11-23 by kareem-wolfssl