wolfSSL/wolfssl #4555: Server can force use of signature algorithm not matching ciphersuite

argold97, 2021-11-09:
In src/internal.c `DoServerKeyExchange()` (lines 24221-24234), the client decodes the signature algorithm and hash specified by the ServerKeyExchange message. However, when calling `DecodeSigAlg()` the value of `args->sigAlgo` is overwritten. `args->sigAlgo` is later used to determine which signature verification algorithm to actually perform. This overwriting occurs without checking that the decoded signature algorithm is compatible with the current ciphersuite. So for example the server can set the ciphersuite to `TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA256`, but then send ECDSA Certificate and ServerKeyExchange messages and have the client successfully verify the key exchange, even if the client did not specify `TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA256` in its ClientHello.

```C
                    if (IsAtLeastTLSv1_2(ssl)) {
                        if ((args->idx - args->begin) + ENUM_LEN + ENUM_LEN >
                                                                        size) {
                            ERROR_OUT(BUFFER_ERROR, exit_dske);
                        }

                        DecodeSigAlg(&input[args->idx], &args->hashAlgo,
                                     &args->sigAlgo);
                        args->idx += 2;
                        hashType = HashAlgoToType(args->hashAlgo);
                        if (hashType == WC_HASH_TYPE_NONE) {
                            ERROR_OUT(ALGO_ID_E, exit_dske);
                        }
                    } else {
```
Note that this bug cannot be tested with unmodified TLS server implementations like openSSL or wolfSSL itself. A modified implementation had to be used to purposefully send mismatching ServerHello and ServerKeyExchange messages.

embhorn, 2021-11-10:
Hi @argold97,

Thanks for sharing this. We are reviewing internally.

kaleb-himes, 2021-11-10:
Hi @argold97 can you please send an email to support [at] wolfssl [dot] com referencing github issue 4555 either in the subject or body of the email so we can coordinate a fix with you in a non-public venue?

closed 2021-11-10 by kaleb-himes