wolfSSL/wolfssl #4568: gcc compilation error: fallthrough keyword

muvarov, 2021-11-15:
Bunch of projects using wolfssl issue compilation error like that:

|/usr/include/optee/export-user_ta/include/compiler.h:259:21: error: expected ')' before 'attribute'
| 259 | #define fallthrough attribute((fallthrough))
| | ^~~~~~~~~~~~~
| ./lib/wolf/wolf_symlink/wolfssl/wolfcrypt/types.h:184:50: note: in expansion of macro 'fallthrough'
| 184 | #define FALL_THROUGH attribute ((fallthrough));
| | ^~~~~~~~~~~
| lib/wolf/wolf_symlink/wolfcrypt/src/ecc.c:2980:13: note: in expansion of macro 'FALL_THROUGH'
| 2980 | FALL_THROUGH;
| | ^~~~~~~~~~~~
| In file included from ./lib/wolf/wolf_symlink/wolfssl/wolfcrypt/ecc.h:26,
| from lib/wolf/wolf_symlink/wolfcrypt/src/ecc.c:106:
| ./lib/wolf/wolf_symlink/wolfssl/wolfcrypt/types.h:184:62: error: expected identifier or '(' before ')' token
| 184 | #define FALL_THROUGH attribute ((fallthrough));
| | ^
| lib/wolf/wolf_symlink/wolfcrypt/src/ecc.c:2980:13: note: in expansion of macro 'FALL_THROUGH'
| 2980 | FALL_THROUGH;

Actually it's wrong define for fallthrough keywork. FIx is here https://github.com/wolfSSL/wolfssl/pull/4563

BR,
Maxim.

kareem-wolfssl, 2021-11-15:
Hi @muvarov ,

What compiler are you using?  `fallthrough` is the correct keyword for GCC: https://gcc.gnu.org/onlinedocs/gcc/Statement-Attributes.html

Thanks,
Kareem

kareem-wolfssl, 2021-11-15:
I misunderstood the issue @muvarov .  You have a conflicting definition of `fallthrough`.  I will work on getting this corrected.

kareem-wolfssl, 2021-11-17:
Hi @muvarov ,

Please try rebuilding with https://github.com/wolfSSL/wolfssl/pull/4580 and let me know if it helps.

Thanks,
Kareem

muvarov, 2021-11-19:
@kareem-wolfssl  I spend some time understaning this changes. And it's looks like this issue was in 74ebf510a3d73e98767eac26082eabdc84e19d31 and it's fixed in 74ebf510a3d73e98767eac26082eabdc84e19d31.

I sent PR to update submodule:
https://github.com/microsoft/MSRSec/pull/35

So I think if you can drop this change for now if you can not reproduce it locally.

closed 2021-11-23 by kareem-wolfssl