wolfSSL/wolfssl #4576: WOLFSSL_CRYPTOCELL defined build error

eliasing, 2021-11-17:
If WOLFSSL_CRYPTOCELL is defined ecc_mulmod still uses ecc_projective_add_point_safe and ecc_projective_dbl_point_safe which leads to build error!
Reverting [this](https://github.com/wolfSSL/wolfssl/commit/be6bf5687b9acb4d27c49fb872d96242d490c462) commit does not help either since ECC benchmark test fails with error = -10085.
#4466

tmael, 2021-11-25:
Hi @eliasing, confirmed the ECC build error due to some missing `!defined(WOLFSSL_CRYPTOCELL` guards. I'll submit a PR with a fix after the thanksgiving break.

eliasing, 2021-11-26:
Thanks @tmael, your PR fixes the issue.

tmael, 2021-11-29:
Thanks for confirming @eliasing, Closing since the fix has been merged into mainline.

closed 2021-11-29 by tmael