wolfSSL/wolfssl #4592: 5.0.0 OpenSSL compatibility between SSL_CTX_set_verify() callback with X509_VERIFY_PARAM_set1_host() - leaves excess DOMAIN_NAME_MISMATCH in X509_STORE_CTX.

mandree, 2021-11-20:
Greetings,

I am trying to make fetchmail compile with wolfSSL's OpenSSL compatibility layer, and it mostly works, however there are run-time issues in the verify call-back (the fetchmail code is OpenSSL centric and uses OpenSSL-style verify call backs) where wolfSSL deviates from OpenSSL behavior in the X509 context passed to the verify callback.

Error presented by my code (snippet below):

`fetchmail: SSL verify callback depth 3: preverify_ok == 1, err = -322, peer subject name mismatch
`

Hostname in use `unimail.tu-dortmund.de` or `imap.gmx.net` to recreate the situation. TLS v1.2 and v1.3 both affected on the former.

This error is passed in the callback installed with `SSL_CTX_set_verify(ssl, SSL_VERIFY_PEER, callback);` - the callback receives a verify_ok, but the X509_STORE_CTX *ctx it is passed contains this error when queried with `X509_STORE_CTX_get_error(ctx)`:

The code that reports the error above roughly looks like this:

```
static int SSL_verify_callback(int verify_ok, X509_STORE_CTX *ctx)
{
    int err = X509_STORE_CTX_get_error(ctx);
    // ...
    report(stdout, GT_("SSL verify callback depth %d: verify_ok == %d, err = %d, %s\n"),
                                         depth, verify_ok, err, X509_verify_cert_error_string(err));
}
```

To trigger, I am setting the host verify name to the expected common name like this:

```
              int r;
              X509_VERIFY_PARAM *param = SSL_get0_param(_ssl_context[sock]);

              X509_VERIFY_PARAM_set_hostflags(param, X509_CHECK_FLAG_NO_PARTIAL_WILDCARDS);
              if (0 == (r = X509_VERIFY_PARAM_set1_host(param, servercname, strlen(servercname)))) {
                  report(stderr, GT_("Warning: X509_VERIFY_PARAM_set1_host(%p, \"%s\") failed (code %#x), trying to continue.\n"),
                          (void *)_ssl_context[sock], servercname, r);
                  ERR_print_errors_fp(stderr);
              }
```

This code works fine with OpenSSL, and the callback receives no error in the ctx.

Commenting either of the set_host* flags, I find that the culprit is the X509_VERIFY_PARAM_set1_host(). It produces no output in wolfSSL's debug output.

Arguably it might not be too useful to look at the X509_STORE_CTX context's error if the preverify_ok passed to the callback is 1, but OpenSSL proper (1.0.2, 1.1.1, 3.0.0) does not leave an error in this context, so this is a deviation from OpenSSL behavior in wolfSSL's code:

`fetchmail: SSL verify callback depth 3: verify_ok == 1, err = 0, ok`

kareem-wolfssl, 2021-11-22:
Thanks for the bug reports, I am reviewing them with the team.

julek-wolfssl, 2021-11-23:
Hi @mandree,

I'm trying to reproduce your issue and I think that I've got quite close. I modified our `examples/client/client.c` client in the following way:

```
$ git diff
diff --git a/examples/client/client.c b/examples/client/client.c
index 0014855d5..efafb6d1c 100644
--- a/examples/client/client.c
+++ b/examples/client/client.c
@@ -2980,7 +2980,7 @@ THREAD_RETURN WOLFSSL_THREAD client_test(void* args)
     #endif
     }
 
-    if (!usePsk && !useAnon && !useVerifyCb && myVerifyAction != VERIFY_FORCE_FAIL) {
+    if (!usePsk && !useAnon && myVerifyAction != VERIFY_FORCE_FAIL) {
     #if defined(OPENSSL_ALL) && defined(WOLFSSL_CERT_GEN) && \
     (defined(WOLFSSL_CERT_REQ) || defined(WOLFSSL_CERT_EXT)) && \
     !defined(NO_FILESYSTEM) && !defined(NO_WOLFSSL_DIR)
@@ -3476,6 +3476,20 @@ THREAD_RETURN WOLFSSL_THREAD client_test(void* args)
         SetupAtomicUser(ctx, ssl);
 #endif
 
+
+#if 1
+    {
+        X509_VERIFY_PARAM *param = SSL_get0_param(ssl);
+
+        X509_VERIFY_PARAM_set_hostflags(param, X509_CHECK_FLAG_NO_PARTIAL_WILDCARDS);
+        if (0 == X509_VERIFY_PARAM_set1_host(param,
+                "webmail.tu-dortmund.de", strlen("webmail.tu-dortmund.de"))) {
+            err_sys("X509_VERIFY_PARAM_set1_host error");
+        }
+    }
+#endif
+
+
     if (matchName && doPeerCheck)
         wolfSSL_check_domain_name(ssl, domain);
 #ifndef WOLFSSL_CALLBACKS
diff --git a/wolfssl/test.h b/wolfssl/test.h
index 3b726afbb..2c3d95425 100644
--- a/wolfssl/test.h
+++ b/wolfssl/test.h
@@ -2636,8 +2636,12 @@ static WC_INLINE int myVerify(int preverify, WOLFSSL_X509_STORE_CTX* store)
     int i = 0;
 #endif
 #endif
+    int err = X509_STORE_CTX_get_error(store);
     (void)preverify;
 
+    printf("SSL verify callback depth %d: verify_ok == %d, err = %d, %s\n",
+            store->depth, preverify, err, X509_verify_cert_error_string(err));
+
     /* Verify Callback Arguments:
      * preverify:           1=Verify Okay, 0=Failure
      * store->error:        Failure error code (0 indicates no failure)
```

This installs a verify callback and sets the domain parameter to check for. Running this example against your webmail server I get the following output:

```
$ ./examples/client/client -h webmail.tu-dortmund.de -p 443 -g -A webmail-tu-dortmund-de-chain.pem -C -j | grep 'SSL verify callback'
SSL verify callback depth 4: verify_ok == 1, err = -322, peer subject name mismatch
SSL verify callback depth 3: verify_ok == 1, err = -322, peer subject name mismatch
SSL verify callback depth 2: verify_ok == 1, err = -322, peer subject name mismatch
SSL verify callback depth 1: verify_ok == 1, err = 0, unknown error number
```

This is expected as only the leaf certificate should pass the domain check. When compiling against OpenSSL, does your callback not get called with the full certificate chain? Does OpenSSL only perform the domain check against the leaf certificate?

mandree, 2021-12-03:
@julek-wolfssl  OpenSSL calls the callback four times, working through the entire chain, root first - but I am scratching my head whether we look at an interpretation issue. I am proposing here that checking anything in the chain but the server's certificate (depth == 0) for its Subject CommonName is meaningless - only the server certificate's Subject CommonName would contain a DNS hostname, but no other certificates in the chain are supposed to do so.

While we're here, wolfSSL might want to add a mapping that err == 0 is no error (rather than an unknown error).

My workaround currently is to call wolfSSL_check_domain_name() instead if my code is using wolfSSL, and avoiding X509_VERIFY_PARAM_set1_host() - but that is sort of a hack as it mixes the OpenSSL and wolfSSL APIs.

This is with OpenSSL 1.1.1l (l as in Lima) as packaged for Fedora 34:
```
fetchmail: --check mode enabled, not fetching mail
fetchmail: 6.4.25.rc4 querying unimail.tu-dortmund.de (protocol IMAP) at Fr 03 Dez 2021 20:56:42 CET: poll started
Trying to connect to 129.217.128.51/993...connected.
fetchmail: SSL verify callback depth 3: verify_ok == 1, err = 0, ok
fetchmail: Certificate chain, from root to peer, starting at depth 3:
fetchmail: Issuer Organization: T-Systems Enterprise Services GmbH
fetchmail: Issuer CommonName: T-TeleSec GlobalRoot Class 2
fetchmail: Subject CommonName: T-TeleSec GlobalRoot Class 2
fetchmail: SSL verify callback depth 2: verify_ok == 1, err = 0, ok
fetchmail: Certificate at depth 2:
fetchmail: Issuer Organization: T-Systems Enterprise Services GmbH
fetchmail: Issuer CommonName: T-TeleSec GlobalRoot Class 2
fetchmail: Subject CommonName: DFN-Verein Certification Authority 2
fetchmail: SSL verify callback depth 1: verify_ok == 1, err = 0, ok
fetchmail: Certificate at depth 1:
fetchmail: Issuer Organization: Verein zur Foerderung eines Deutschen Forschungsnetzes e. V.
fetchmail: Issuer CommonName: DFN-Verein Certification Authority 2
fetchmail: Subject CommonName: DFN-Verein Global Issuing CA
fetchmail: SSL verify callback depth 0: verify_ok == 1, err = 0, ok
fetchmail: Server certificate:
fetchmail: Issuer Organization: Verein zur Foerderung eines Deutschen Forschungsnetzes e. V.
fetchmail: Issuer CommonName: DFN-Verein Global Issuing CA
fetchmail: Subject CommonName: unimail.tu-dortmund.de
fetchmail: Subject Alternative Name: unimail.tu-dortmund.de
fetchmail: Subject Alternative Name: unimail.uni-dortmund.de
fetchmail: unimail.tu-dortmund.de key fingerprint: 3C:8E:BC:A8:90:83:47:86:AC:4F:63:E9:25:C6:CC:37
fetchmail: SSL/TLS: using protocol TLSv1.3, cipher TLS_AES_256_GCM_SHA384, 256/256 secret/processed bits
fetchmail: IMAP< * OK [CAPABILITY IMAP4rev1 LITERAL+ ID ENABLE AUTH=LOGIN AUTH=PLAIN SASL-IR] unimail Cyrus IMAP 3.4.2 server ready
``` 

And this is the same codebase on the same computer against wolfSSL 5.0.0:
```
fetchmail: --check mode enabled, not fetching mail
fetchmail: 6.4.25.rc4 querying unimail.tu-dortmund.de (protocol IMAP) at Fr 03 Dez 2021 21:01:54 CET: poll started
Trying to connect to 129.217.128.51/993...connected.
fetchmail: SSL verify callback depth 3: verify_ok == 1, err = -322, peer subject name mismatch
fetchmail: Certificate chain, from root to peer, starting at depth 3:
fetchmail: Issuer Organization: T-Systems Enterprise Services GmbH
fetchmail: Issuer CommonName: T-TeleSec GlobalRoot Class 2
fetchmail: Subject CommonName: T-TeleSec GlobalRoot Class 2
fetchmail: Server certificate verification error: peer subject name mismatch
fetchmail: SSL verify callback depth 2: verify_ok == 1, err = -322, peer subject name mismatch
fetchmail: Certificate at depth 2:
fetchmail: Issuer Organization: T-Systems Enterprise Services GmbH
fetchmail: Issuer CommonName: T-TeleSec GlobalRoot Class 2
fetchmail: Subject CommonName: DFN-Verein Certification Authority 2
fetchmail: SSL verify callback depth 1: verify_ok == 1, err = -322, peer subject name mismatch
fetchmail: Certificate at depth 1:
fetchmail: Issuer Organization: Verein zur Foerderung eines Deutschen Forschungsnetzes e. V.
fetchmail: Issuer CommonName: DFN-Verein Certification Authority 2
fetchmail: Subject CommonName: DFN-Verein Global Issuing CA
fetchmail: SSL verify callback depth 0: verify_ok == 1, err = 0, unknown error number
fetchmail: Server certificate:
fetchmail: Issuer Organization: Verein zur Foerderung eines Deutschen Forschungsnetzes e. V.
fetchmail: Issuer CommonName: DFN-Verein Global Issuing CA
fetchmail: Subject CommonName: unimail.tu-dortmund.de
fetchmail: Subject Alternative Name: unimail.uni-dortmund.de
fetchmail: Subject Alternative Name: unimail.tu-dortmund.de
fetchmail: unimail.tu-dortmund.de key fingerprint: 3C:8E:BC:A8:90:83:47:86:AC:4F:63:E9:25:C6:CC:37
fetchmail: SSL/TLS: using protocol TLSv1.3, cipher TLS_AES_128_GCM_SHA256, 0/0 secret/processed bits
```

julek-wolfssl, 2021-12-21:
@mandree Thank you for the additional details. We will discuss internally on what the desired behavior should be and take appropriate action.

julek-wolfssl, 2022-01-10:
A fix for this check to be performed only on leaf certs is open here: https://github.com/wolfSSL/wolfssl/pull/4716

julek-wolfssl, 2022-01-19:
Hi @mandree, the pull request fixing this issue has been merged. Thanks for the report! Closing issue.

closed 2022-01-19 by julek-wolfssl

mandree, 2022-01-19:
Thanks, this is mostly solved, with just the error string translation wrong in a different place:
`fetchmail: SSL verify callback depth 1: verify_ok == 1, err = 0, unknown error number`
^ this was wolfSSL. v now follows OpenSSL:
fetchmail: SSL verify callback depth 1: verify_ok == 1, err = 0, ok

julek-wolfssl, 2022-01-19:
Thanks. I'll make a note to add a case in the error translation to treat `0` as no error.