wolfSSL/wolfssl #4596: Incorrect secp256r1 ecc point multification result

CindyZhouYH, 2021-11-22:
I was doing elliptic curve point multiplication using ECC_SECP256R1. And I found wolfcrypt's result different from results of relic and botan.
```
Curve : ECC_SECP256R1
P.x : 56515219790691171413109057904011688695424810155802929973526481321309856242040
P.y : 112415057366643990503473735497916120931997467895538765628421173214231600270382
m : 52838058721772159177448907333242226474676273722824546318056365201184268980770050127260547419911003155460331597669032975718351095515936750613608507339331776175023257016677603122334856843636331424934221702706434483795689452927254977366615861404912014153171265016178675912153501046651190150158175216157592434074959319350283151051720611396029746176714002726368104372467558168233383940353698361967551818352819655853644037240032773557290413186403950050146280000004568223203417558998005563506178922228538616483233197658941088135584621960807585123307716639270625934232481738068077377244284065730390232364954918056665

res = m * P
```
wolfcrypt's result:
```
res.x : 76150005149208156032418643466035150058174798366135606055630004474078174003428
res.y : 75276267740638892996642367544073862415886926002257544268661282169016904967930
```
relic's result:
```
res.x : 1899476376521990906816330754666973622310274005078588400318066451818247648965
res.y : 113260418129733158120708921987807196554168163823094871581741216056595501887315
```
botan's result:
```
res.x : 1899476376521990906816330754666973622310274005078588400318066451818247648965
res.y : 113260418129733158120708921987807196554168163823094871581741216056595501887315
```

It seems that the output is correct when m's length is short. Once m's length is longer than a certain value, the result would be incorrect.

kareem-wolfssl, 2021-11-22:
Hi @CindyZhouYH ,

Which math implementation are you using?  Can you post your build settings?

Thanks,
Kareem

CindyZhouYH, 2021-11-23:
Thank you for your reply.
The testfile I use is as follows:
```
#define ASSERT_EQUAL(res, exp) if ( (res) != (exp) ) { printf("assert err\n"); abort(); }
#define ASSERT_NOT_EQUAL(res, exp) if ( (res) == (exp) ) { printf("assert err\n"); abort();}

string ax = "56515219790691171413109057904011688695424810155802929973526481321309856242040";
string ay = "112415057366643990503473735497916120931997467895538765628421173214231600270382";
string strb = "52838058721772159177448907333242226474676273722824546318056365201184268980770050127260547419911003155460331597669032975718351095515936750613608507339331776175023257016677603122334856843636331424934221702706434483795689452927254977366615861404912014153171265016178675912153501046651190150158175216157592434074959319350283151051720611396029746176714002726368104372467558168233383940353698361967551818352819655853644037240032773557290413186403950050146280000004568223203417558998005563506178922228538616483233197658941088135584621960807585123307716639270625934232481738068077377244284065730390232364954918056665";
static int HexCharToDec(const char c) {
    if ( c >= '0' && c <= '9' ) {
        return c - '0';
    } else if ( c >= 'a' && c <= 'f' ) {
        return c - 'a' + 10;
    } else if ( c >= 'A' && c <= 'F' ) {
        return c - 'A' + 10;
    } else {
        assert(0);
    }
}

std::string HexToDec(std::string s) {
    std::string ret;
    if ( s.size() >= 2 && s[0] == '0' && s[1] == 'x' ) {
        s = s.substr(2);
    }
    boost::multiprecision::cpp_int total;
    for (long i = s.size() - 1; i >= 0; i--) {
        total += boost::multiprecision::cpp_int(HexCharToDec(s[i])) << ((s.size()-i-1)*4);
    }
    std::stringstream ss;
    ss << total;
    if ( ss.str().empty() ) {
        return "0";
    } else {
        return ss.str();
    }
}

void wolfCrypt_ECC_point_mul(){
    int curveID = ECC_SECP256R1;
    int curveIdx;
    const ecc_set_type* curve = nullptr;
    ASSERT_NOT_EQUAL(curveIdx = wc_ecc_get_curve_idx(curveID), ECC_CURVE_INVALID);
    ASSERT_NOT_EQUAL(curve = wc_ecc_get_curve_params(curveIdx), nullptr);
    ecc_point* res = wc_ecc_new_point_h(nullptr);
    ecc_point* a = wc_ecc_new_point_h(nullptr);
    mp_int* b = (mp_int*)::malloc(sizeof(mp_int));
    mp_init(b);
    mp_int* A = (mp_int*)::malloc(sizeof(mp_int));
    mp_init(A);
    mp_int* modulus = (mp_int*)::malloc(sizeof(mp_int));
    mp_init(modulus);
    mp_int* x = (mp_int*)::malloc(sizeof(mp_int));
    mp_init(x);
    mp_int* y = (mp_int*)::malloc(sizeof(mp_int));
    mp_init(y);
    mp_int* z = (mp_int*)::malloc(sizeof(mp_int));
    mp_init(z);
    string az = "1";
    // construct point a
    ASSERT_EQUAL(mp_read_radix(a->x, ax.c_str(), 10), 0);
    ASSERT_EQUAL(mp_read_radix(a->y, ay.c_str(), 10), 0);
    ASSERT_EQUAL(mp_read_radix(a->z, az.c_str(), 10), 0);
    // check validation
    ASSERT_EQUAL(wc_ecc_point_is_on_curve(a, curveIdx), 0);
    // preperation
    ASSERT_EQUAL(mp_read_radix(b, strb.c_str(), 10), 0);
    ASSERT_EQUAL(mp_read_radix(A, HexToDec(curve->Af).c_str(), 10), 0);
    ASSERT_EQUAL(mp_read_radix(modulus, HexToDec(curve->prime).c_str(), 10), 0);
    // multiplication
    ASSERT_EQUAL(wc_ecc_mulmod_ex(b, a, res, A, modulus, 1, nullptr), 0);
    // read result
    int size;
    ASSERT_EQUAL(mp_radix_size(res->x, 10, &size), MP_OKAY);
    char * wolf_x_c = (char*)::malloc(size);
    ASSERT_EQUAL(mp_toradix(res->x, wolf_x_c, 10), MP_OKAY);

    ASSERT_EQUAL(mp_radix_size(res->y, 10, &size), MP_OKAY);
    char * wolf_y_c = (char*)::malloc(size);
    ASSERT_EQUAL(mp_toradix(res->y, wolf_y_c, 10), MP_OKAY);

    printf("%s\n%s\n", wolf_x_c, wolf_y_c);
    
    wc_ecc_del_point(a);
    wc_ecc_del_point(res);
    free(b);
    free(A);
    free(modulus);
    free(x);
    free(y);
    free(z);
    free(wolf_x_c);
    free(wolf_y_c);
}
```
I cloned the source code and built it into a static library:
```
autoreconf -ivf
./configure --enable-static --enable-md2 --enable-md4 --enable-ripemd --enable-blake2 --enable-blake2s --enable-pwdbased --enable-scrypt --enable-hkdf --enable-cmac --enable-arc4 --enable-camellia --enable-rabbit --enable-aesccm --enable-aesctr --enable-hc128 --enable-xts --enable-des3 --enable-idea --enable-x963kdf --enable-harden --enable-aescfb --enable-aesofb --enable-aeskeywrap --enable-keygen --enable-curve25519 --enable-curve448 --enable-shake256 --disable-crypttests --disable-examples --enable-compkey --enable-ed448 --enable-ed25519 --enable-ecccustcurves --enable-xchacha --enable-cryptocb --enable-eccencrypt --enable-smallstack --enable-ed25519-stream --enable-ed448-stream
make -j$(nproc)
export WOLFCRYPT_LIBWOLFSSL_A_PATH=`realpath src/.libs/libwolfssl.a`
export WOLFCRYPT_INCLUDE_PATH=`realpath .`
```
and compiled the testfile by:
```
clang++ -fsanitize=address,undefined -D_GLIBCXX_DEBUG -g -DFUZZING_HEADERS_NO_IMPL -o testwolf wolf_ecc_mul.cpp -I $(WOLFCRYPT_INCLUDE_PATH) -fPIC -no-pie $(WOLFCRYPT_LIBWOLFSSL_A_PATH)
```

After testing, wolfCrypt's result is consistent with botan and relic when the length of `strb` is less than 79.

SparkiDev, 2021-12-02:
Hi @CindyZhouYH

I noticed that the scalar you are multiplying by is much larger than the order of the curve.
When I compiled the above code, after some rework, against the latest wolfSSL, the wc_ecc_mulmod_ex() call returns error -217: ECC_OUT_OF_RANGE_E.
This is because the number of bits in the modulus is less than the number of bits in the scalar.

This function will not work if you pass a scalar that is larger than the order.

May I ask how this test case came up?

CindyZhouYH, 2022-01-27:
Sorry for the delayed reply.
I tried again using the latest version. The result is just as you said.
Maybe there are some problems with the version I used before.
Thanks again:)

closed 2022-01-27 by CindyZhouYH