wolfSSL/wolfssl #4598: Compilation error for CPU=mips32 and --enable-sp-math-all=huge

e2iplayer, 2021-11-22:
Hello,

there is compilation errors when want to compile WolfSSL for  CPU=mips32.

**How to reproduce:**
```
wget https://github.com/wolfSSL/wolfssl/archive/refs/tags/v5.0.0-stable.tar.gz
wget https://github.com/Broadcom/stbgcc-4.8/releases/download/stbgcc-4.8-1.7/stbgcc-4.8-1.7.tar.bz2
tar -xvf stbgcc-4.8-1.7.tar.bz2
tar -xvf v5.0.0-stable.tar.gz

cd wolfssl-5.0.0-stable/

export PATH=$PATH:$PWD:/mnt/raw/investigations/2021-11-22_wolfssl/stbgcc-4.8-1.7/bin/
libtoolize
./autogen.sh
./configure \
    --enable-all \
    --enable-sp-math-all=huge \
    --host=mipsel-linux-gnu \
    --build=i686-linux
make
```

compilation errors:
```
wolfcrypt/src/sp_int.c:6548:13: note: in expansion of macro 'SP_ASM_MUL_ADD'
             SP_ASM_MUL_ADD(l, h, o, a->dp[14], b->dp[6]);
             ^
wolfcrypt/src/sp_int.c:1415:5: error: unknown register name '$hi' in 'asm'
     __asm__ __volatile__ (                               \
     ^
wolfcrypt/src/sp_int.c:6549:13: note: in expansion of macro 'SP_ASM_MUL_ADD'
             SP_ASM_MUL_ADD(l, h, o, a->dp[15], b->dp[5]);
             ^
wolfcrypt/src/sp_int.c:1415:5: error: unknown register name '$lo' in 'asm'
     __asm__ __volatile__ (                               \
     ^
...
```

Could you please check this?

kareem-wolfssl, 2021-11-22:
Confirmed.  Do you know if your target has these HI/LO registers?
@SparkiDev please take a look.  Not clear if we support little endian MIPS (mipsel).

e2iplayer, 2021-11-22:
I think all MIPS have these registers. 
This could be problem with assembler 
maybe
```
%hi
%lo
```
should be used.
Instead of
```
$hi
$lo
```
but I do not know asm, sorry.

SparkiDev, 2021-11-30:
Hi @e2iplayer,

I've looked into this and came to the conclusion that older gcc compilers don't support $hi and $lo.
Newer compilers support %hi and %lo so I've put up a pull request that uses these names instead: https://github.com/wolfSSL/wolfssl/pull/4616

Let me know if there are other problems!

closed 2021-12-01 by e2iplayer