wolfSSL/wolfssl #4601: WOLFSSL_CRYPTOCELL defined build error 2

eliasing, 2021-11-24:
If WOLFSSL_CRYPTOCELL is defined  **wc_RsaSSL_Verify** shall be used, but **wc_RsaSSL_VerifyInline** is used in [RsaSign and VerifyRsaSign](https://raw.githubusercontent.com/wolfSSL/wolfssl/master/src/internal.c) and [ConfirmSignature ](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L13446) which is not defined.

tmael, 2021-11-25:
Hi @eliasing, I've confirmed that there's a build error due to some missing `!defined(WOLFSSL_CRYPTOCELL` guards in ecc.c and test.c. I will put up a PR when I get a moment. It appears that you are having RSA-related issues. Can you please share the build error?

eliasing, 2021-11-26:
Hi @tmael, #4604 does not fix this issue.
I am trying to enable TLS1.3, which using RSA keys requires RSA PSS support and then the above issue raises.
The build error is:
undefined reference to `wc_RsaSSL_VerifyInline'

tmael, 2021-11-29:
Hi @eliasing, Closing. TLS1.3 support with Cryptocell is unavailable at the moment and you can't expect it to work out of the box. We have been tracking a TLS/DTLS port feature request internally. If you need to expedite the request, please contact support@wolfssl.com.

closed 2021-11-29 by tmael