wolfSSL/wolfssl #4605: IO_URING and wolfSSL

kosmas-valianos, 2021-11-25:
Follow-up of [Z13175](https://wolfssl.zendesk.com/hc/en-us/requests/13175)

We are trying to migrate our applications from the normal Linux `IO` to the mighty `IO_URING`. Unfortunately despite the wolfSSL's ability to set callbacks when reading/writing it's at the moment impossible to integrate it. The reason is that with `IO_URING` you submit a read/write operation and the execution is happening in the future. WolfSSL expects the result of the read/write directly.

Are there any plans to expose some API with that in mind? It shall be for example: `SSL_add_read()`/`SSL_add_write` which will add the submit read/write request and respectively `SSL_handle_read()`/`SSL_handle_write()` for usage when the execution has happened and the result has been fetched from the CQ.

Obviously this is Linux specific while wolfSSL tries to be as agnostic as possible but given the massive boost one can get with `IO_URING` maybe something to consider

embhorn, 2021-11-29:
Hi @kosmas-valianos

Thank you so much for officially registering your feature request with wolfSSL. I've noted your interest in our internal feature tracker. We try to work on requested features whenever we get free engineering cycles unless a customer decides to back the effort for a finite implementation timeline. If at any time a feature becomes a progress blocker for a project it can be accelerated. 

Please let me know if there is anything else we can assist with at this time or if you have any follow-up questions to how we manage feature requests.

closed 2021-11-29 by embhorn

dgarske, 2021-11-29:
Hi @kosmas-valianos ,

We do have the ability to override the allocation of the IO send/recv buffers that would allow you to provide custom buffer pointers from IO_URING.

To do this you would define `XMALLOC_USER` and supply your own allocator functions `XMALLOC` and `XFREE`. You can intercept the `type` argument for `DYNAMIC_TYPE_IN_BUFFER` and `DYNAMIC_TYPE_OUT_BUFFER`. You might refer to the `HAVE_IO_POOL` section in memory.c for an example of this.

https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/memory.c#L969

```
void* XMALLOC(size_t n, void* heap, int type);
void XFREE(void *p, void* heap, int type);
```

Thanks,
David Garske, wolfSSL

kosmas-valianos, 2021-11-29:
Thanks @dgarske , that sounds like a viable solution to what we are after

sfjhkjdsh, 2023-05-30:
@kosmas-valianos did you manage to find a working example?