wolfSSL/wolfssl #4623: EmbedSend() handling of EPIPE

argold97, 2021-12-02:
I recently updated from wolfSSL 4.5.0 to 5.0.0 and noticed a functional change/bug that I'm not sure was intended.

Prior to https://github.com/wolfSSL/wolfssl/commit/c88afdef87f5e6178087b592208ca403e685fb1d, `EmbedSend()` would translate an `EPIPE` errno into `WOLFSSL_CBIO_ERR_CONN_CLOSE`. Later on this error code would inform some other code that the peer had closed the connection, prompting some retry logic.

After https://github.com/wolfSSL/wolfssl/commit/c88afdef87f5e6178087b592208ca403e685fb1d, the errno to error code translation was moved to `TranslateIoError()`, and this function doesn't have a case for `EPIPE` so the same condition results in `WOLFSSL_CBIO_ERR_GENERAL`. Returning this generic error code instead of `WOLFSSL_CBIO_ERR_CONN_CLOSE` caused some of my retry logic to break.

My question is whether this functional change was intentional or an oversight.

julek-wolfssl, 2021-12-03:
Thank you for your submission @argold97. I don't think that this omission was intentional and was an oversight during a refactor. I have proposed a fix to your issue in https://github.com/wolfSSL/wolfssl/pull/4628. Could you please run your tests against this pull request and see if it resolves the `EPIPE` handling?

argold97, 2021-12-03:
Yes, that works fine.

julek-wolfssl, 2021-12-21:
Pull request is merged. Closing issue. Thank you for your report.

closed 2021-12-21 by julek-wolfssl