wolfSSL/wolfssl #4659: Can't run make for 16-bit architecture

cbjornram, 2021-12-14:
Hi,

I have tried and failed multiple times now with compiling WolfSSL for 16-bit architecture. I am not sure which parts I should disable and want to ask for help about it. My configuration looks like this:

`./configure --prefix=$HOME/wolfssl --enable-16bit --disable-benchmark --disable-crypttests --disable-examples --disable-kcapi-hmac --disable-hkdf --disable-truncatedhmac --disable-enc-then-mac`

I probably don't want to disable all those parts, but as part of troubleshooting it is like this for now and I will change it when I have managed to get a successful first build. However when running `make` after this configuration this is the output I get:

```
/Library/Developer/CommandLineTools/usr/bin/make -j5  all-recursive
make[2]: warning: -jN forced in submake: disabling jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-kdf.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
wolfcrypt/src/random.c:371:9: error: implicit conversion loses integer precision: 'unsigned long' to 'int'
      [-Werror,-Wshorten-64-to-32]
        + ((outSz % OUTPUT_BLOCK_LEN) ? 1 : 0);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/random.c:468:21: error: implicit conversion loses integer precision: 'unsigned long' to 'int'
      [-Werror,-Wshorten-64-to-32]
    for (i = dataSz - 1; i >= 0; i--)
           ~ ~~~~~~~^~~
wolfcrypt/src/random.c:501:38: error: implicit conversion loses integer precision: 'unsigned long' to 'int'
      [-Werror,-Wshorten-64-to-32]
    len = (outSz / OUTPUT_BLOCK_LEN) + ((outSz % OUTPUT_BLOCK_LEN) ? 1 : 0);
        ~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/random.c:572:21: error: implicit conversion loses integer precision: 'unsigned long' to 'int'
      [-Werror,-Wshorten-64-to-32]
        dIdx = dLen - 1;
             ~ ~~~~~^~~
wolfcrypt/src/random.c:573:26: error: implicit conversion loses integer precision: 'unsigned long' to 'int'
      [-Werror,-Wshorten-64-to-32]
        for (sIdx = sLen - 1; sIdx >= 0; sIdx--) {
                  ~ ~~~~~^~~
wolfcrypt/src/random.c:725:29: error: implicit conversion loses integer precision: 'word32' (aka 'unsigned long') to
      'int' [-Werror,-Wshorten-64-to-32]
                            scratchSz) == 0) {
                            ^~~~~~~~~
6 errors generated.
wolfcrypt/src/hmac.c:1238:43: error: implicit conversion loses integer precision: 'word32' (aka 'unsigned long') to
      'int' [-Werror,-Wshorten-64-to-32]
            int    tmpSz = (n == 1) ? 0 : hashSz;
                   ~~~~~                  ^~~~~~
1 error generated.
make[2]: *** [wolfcrypt/src/src_libwolfssl_la-random.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [wolfcrypt/src/src_libwolfssl_la-hmac.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

As you can see I still get errors warning for integer overflow. I understand that this is expected for large parts of the library, and it is those parts I wish to disable. However I can not figure out what to disable to get rid of these errors.

Could anyone help me and guide me in some direction? Which part of WolfSSL is `/src/random.c` part of and how do I disable it?

Thank you for you help.

kareem-wolfssl, 2021-12-14:
Hi @cbjornram ,

Our 16-bit support is currently incomplete.  It will require some engineering time to get it fully working again.
May I ask what kind of project you are building this for, personal or commercial?

Thanks,
Kareem

cbjornram, 2021-12-15:
Hi @kareem-wolfssl,

Thank you for your reply. I understand, and in that case I suspect it might be better for me to look for alternative solutions.

I am working on a personal project just for fun where I'm trying to get a ATmega2560 to send sensor values (arriving through hardware serial communication) with MQTTv5 and TLS 1.2 encryption, over a SIM800H modem. I've gotten the MQTTv5 part working with your WolfMQTT library (with minor modifications for the 16-bit architecture), and was initially hoping to let the SIM800H take care of the TLS encryption, however I realised too late it only supports TLS 1.0. So looking for other software solutions I started looking into WolfSSL. I could go with other hardware (either another microprocessor or modem) but I thought it was worth giving this a try first.

Thank you,
Christoffer

kareem-wolfssl, 2021-12-15:
No problem @cbjornram , I wish I had a better answer.
You may be able to get it working by overriding our types (word32 etc) in wc_port.h to the appropriate size.
We do support a variety of hardware, really just about any 32-bit or higher architecture should work.

Thanks,
Kareem

cbjornram, 2021-12-17:
I will try that that. Thank you for your help.

Kind regards,
Christoffer

closed 2021-12-17 by cbjornram