wolfSSL/wolfssl #4683: Doesn't compile on VS2019 x64

namazso, 2021-12-21:
repro:

```
copy examples/configs/user_settings_all.h IDE/WIN/user_settings.h
msbuild.exe /t:Build /p:WindowsTargetPlatformVersion=10.0 /p:PlatformToolset=v142 /p:Configuration=Release /p:Platform=x64 wolfssl64.sln
```

output:
```
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
Build started 2021-12-21 16:32:10.
Project "C:\wolfssl-5.0.0-gplv3-fips-ready\wolfssl64.sln" on node 1 (Build target(s)).
ValidateSolutionConfiguration:
  Building solution configuration "Release|x64".
Project "C:\wolfssl-5.0.0-gplv3-fips-ready\wolfssl64.sln" (1) is building "C:\wolfssl-5.0.0-gplv3-fips-ready\wolfssl.vc
xproj" (2) on node 1 (default targets).
PrepareForBuild:
  Creating directory "x64\Release\obj\".
  Creating directory "x64\Release\obj\wolfssl.tlog\".
InitializeBuildStatus:
  Creating "x64\Release\obj\wolfssl.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
CustomBuild:
  Performing Custom Build Tools
  Microsoft (R) Macro Assembler (x64) Version 14.29.30138.0
  Copyright (C) Microsoft Corporation.  All rights reserved.

   Assembling: wolfcrypt\src\aes_asm.asm
  Performing Custom Build Tools
  Microsoft (R) Macro Assembler (x64) Version 14.29.30138.0
  Copyright (C) Microsoft Corporation.  All rights reserved.

   Assembling: wolfcrypt\src\sp_x86_64_asm.asm
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\CL.exe /c /I
  ./ /I./IDE/WIN /Zi /nologo /W3 /WX- /diagnostics:column /O2 /Oi /GL /D WOLFSSL_LIB /D WOLFSSL_USER_SETTINGS /D CYASSL
  _USER_SETTINGS /D _UNICODE /D UNICODE /Gm- /EHsc /MD /GS /Gy /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"x64\
  Release\obj\\" /Fd"C:\wolfssl-5.0.0-gplv3-fips-ready\x64\Release\wolfssl.pdb" /external:W3 /Gd /TC /FC /errorReport:q
  ueue src\crl.c src\internal.c src\wolfio.c src\keys.c src\ocsp.c src\ssl.c src\tls.c src\tls13.c wolfcrypt\src\aes.c
  wolfcrypt\src\arc4.c wolfcrypt\src\asn.c wolfcrypt\src\blake2b.c wolfcrypt\src\blake2s.c wolfcrypt\src\camellia.c wol
  fcrypt\src\chacha.c wolfcrypt\src\chacha20_poly1305.c wolfcrypt\src\cmac.c wolfcrypt\src\coding.c wolfcrypt\src\curve
  25519.c wolfcrypt\src\curve448.c wolfcrypt\src\cpuid.c wolfcrypt\src\des3.c wolfcrypt\src\dh.c wolfcrypt\src\dsa.c wo
  lfcrypt\src\ecc.c wolfcrypt\src\ed25519.c wolfcrypt\src\ed448.c wolfcrypt\src\error.c wolfcrypt\src\fe_448.c wolfcryp
  t\src\fe_operations.c wolfcrypt\src\ge_448.c wolfcrypt\src\ge_low_mem.c wolfcrypt\src\ge_operations.c wolfcrypt\src\h
  ash.c wolfcrypt\src\hc128.c wolfcrypt\src\hmac.c wolfcrypt\src\idea.c wolfcrypt\src\integer.c wolfcrypt\src\kdf.c wol
  fcrypt\src\logging.c wolfcrypt\src\md2.c wolfcrypt\src\md4.c wolfcrypt\src\md5.c wolfcrypt\src\memory.c wolfcrypt\src
  \pkcs7.c wolfcrypt\src\pkcs12.c wolfcrypt\src\poly1305.c wolfcrypt\src\pwdbased.c wolfcrypt\src\rabbit.c wolfcrypt\sr
  c\random.c wolfcrypt\src\rc2.c wolfcrypt\src\ripemd.c wolfcrypt\src\rsa.c wolfcrypt\src\sha.c wolfcrypt\src\sha256.c
  wolfcrypt\src\sha3.c wolfcrypt\src\sha512.c wolfcrypt\src\signature.c wolfcrypt\src\sp_c32.c wolfcrypt\src\sp_c64.c w
  olfcrypt\src\sp_int.c wolfcrypt\src\sp_x86_64.c wolfcrypt\src\srp.c wolfcrypt\src\tfm.c wolfcrypt\src\wc_encrypt.c wo
  lfcrypt\src\wc_pkcs11.c wolfcrypt\src\wc_port.c wolfcrypt\src\wolfmath.c wolfcrypt\src\wolfevent.c
  crl.c
  internal.c
  wolfio.c
C:\wolfssl-5.0.0-gplv3-fips-ready\src\wolfio.c(1029,47): warning C4244: 'return': conversion from 'SOCKET' to 'int', po
ssible loss of data [C:\wolfssl-5.0.0-gplv3-fips-ready\wolfssl.vcxproj]
  keys.c
  ocsp.c
  ssl.c
  tls.c
  tls13.c
  aes.c
  arc4.c
  asn.c
  blake2b.c
  blake2s.c
  camellia.c
  chacha.c
[... 827 lines cut ...]
dy\examples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol fe_load [C:\wolfssl-5.0.0-gplv3-fips-ready\ex
amples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol lm_sub [C:\wolfssl-5.0.0-gplv3-fips-ready\exa
mples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol lm_copy [C:\wolfssl-5.0.0-gplv3-fips-ready\ex
amples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol fprime_mul [C:\wolfssl-5.0.0-gplv3-fips-ready
\examples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol lm_neg [C:\wolfssl-5.0.0-gplv3-fips-ready\exa
mples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol fe_mul__distinct [C:\wolfssl-5.0.0-gplv3-fips
-ready\examples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol fprime_add [C:\wolfssl-5.0.0-gplv3-fips-ready
\examples\server\server.vcxproj]
  wolfssl.lib(ge_low_mem.obj) : error LNK2001: unresolved external symbol fe_sqrt [C:\wolfssl-5.0.0-gplv3-fips-ready\ex
amples\server\server.vcxproj]
  C:\wolfssl-5.0.0-gplv3-fips-ready\x64\Release\server.exe : fatal error LNK1120: 14 unresolved externals [C:\wolfssl-5
.0.0-gplv3-fips-ready\examples\server\server.vcxproj]

    53 Warning(s)
    75 Error(s)

Time Elapsed 00:00:15.35
```

dgarske, 2021-12-21:
Hi @namazso,

Thank you for the report. I am able to reproduce these fe_448 math warnings with Visual Studio. We will work on adding explicit casting for these. The VS compiler is very strict on the casting. I am not concerned with these warnings causing any runtime issues.

Thanks,
David Garske, wolfSSL

namazso, 2021-12-21:
I'm not concerned about the warnings either, a bit more about the linking errors as I don't get an usable output.

SparkiDev, 2021-12-22:
Hi @namazso

The project is missing the file: fe_low_mem.c

Please add this file and let us know if you link.

namazso, 2021-12-22:
I can confirm it works after adding that file to the wolfssl project

SparkiDev, 2022-01-17:
The file fe_low_mem.c has been added to wolfssl.vcxproj in the wolfSSL 5.1.1 release.

closed 2022-01-17 by SparkiDev