wolfSSL/wolfssl #4684: unresolved external symbol InetPton

namazso, 2021-12-21:
```
C:\wolfssl-5.0.0-gplv3-fips-ready\src\ssl.c(48843,9): warning C4013: 'InetPton' undefined; assuming extern returning in
t [C:\wolfssl-5.0.0-gplv3-fips-ready\wolfssl.vcxproj]
```

... later ...

```
"C:\wolfssl-5.0.0-gplv3-fips-ready\examples\client\client.vcxproj" (default target) (7) ->
  wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol InetPton [C:\wolfssl-5.0.0-gplv3-fips-ready\examples
\client\client.vcxproj]
  C:\wolfssl-5.0.0-gplv3-fips-ready\x64\Release\client.exe : fatal error LNK1120: 1 unresolved externals [C:\wolfssl-5.
0.0-gplv3-fips-ready\examples\client\client.vcxproj]
```

if windows headers were included, InetPton would be defined as InetPtonW and not error out. Repro:

download wolfssl-5.0.0-gplv3-fips-ready.zip and extract somewhere, then in visual studio 2019 x64 powershell:

```
 echo "#ifndef _WIN_USER_SETTINGS_H_EXTRA_`n#define _WIN_USER_SETTINGS_H_EXTRA_`n#define OPENSSL_EXTRA`n#define OPENSSL_ALL`n#define WOLFSSL_ASIO`n#define ASIO_USE_WOLFSSL`n#define WOLFSSL_KEY_GEN`n#define BOOST_ASIO_USE_WOLFSSL`n#define HAVE_EX_DATA`n#define SSL_TXT_TLSV1_2`n#define SSL_TXT_TLSV1_1`n#define OPENSSL_NO_SSL2`n#define OPENSSL_NO_SSL3`n#endif`n" >> IDE\WIN\user_settings.h
 msbuild.exe /t:Build /p:WindowsTargetPlatformVersion=10 /p:PlatformToolset=v142 /p:Configuration=Release /p:Platform=x64 wolfssl64.sln
 ```

embhorn, 2021-12-21:
Hi @namazso 

Are you building with VS? Which VS configuration are you building?

Thanks,
Eric - wolfSSL Support

namazso, 2021-12-21:
yes, latest version of VS 2019.  Everything else is in the opening post, but I'll decipher the msbuild line for you:
`/p:WindowsTargetPlatformVersion=10 /p:PlatformToolset=v142`
retargeted for vs2019 toolset and latest win10 sdk (10.22000 on my system aka win11 sdk)
`/p:Configuration=Release`
Release config
`/p:Platform=x64`
On x64
`wolfssl64.sln`
all targets

namazso, 2021-12-21:
for me the graphical IDE also reproduced the problems when doing the config changes (aka the echo line)

embhorn, 2021-12-21:
Please try adding an include for `ws2tcpip.h`  in `wolfssl/wolfcrypt/wc_port.h`

```
diff --git a/wolfssl/wolfcrypt/wc_port.h b/wolfssl/wolfcrypt/wc_port.h
index eb4a4e0a4..e1d1c8be9 100644
--- a/wolfssl/wolfcrypt/wc_port.h
+++ b/wolfssl/wolfcrypt/wc_port.h
@@ -606,6 +606,7 @@

 /* THREADING/MUTEX SECTION */
 #ifdef USE_WINDOWS_API
+    #include <ws2tcpip.h>
     #ifdef WOLFSSL_GAME_BUILD
         #include "system/xtl.h"
     #else
```

embhorn, 2021-12-21:
I can also confirm build errors with `OPENSSL_ALL` defined.

duk-37, 2021-12-22:
relatively simple fix/workaround for this that works independent of include order: (`Windows.h` needs to be included before `ws2tcpip.h`)

(just needs to be above the two XINET_NTOP/XINET_PTON definitions)
```
#ifdef USE_WINDOWS_API
	#define _WINSOCKAPI_
	#include <Windows.h>
	#include <winsock2.h>
	#include <ws2tcpip.h>
#endif
```

namazso, 2021-12-22:
@embhorn  i can confirm the patch works

closed 2021-12-24 by namazso