wolfSSL/wolfssl #4727: ocsp-stapling2.test fails

ryandesign, 2022-01-06:
With wolfssl 5.1.1 on macOS 10.15.7, ocsp-stapling2.test fails for me.

I'm not sure if I'm reading the test-suite.log correctly but it looks like the problem might be:

```
SSL_accept error -132, Buffer error, output too small or input too big
wolfSSL error: SSL_accept failed
```

[test-suite.log](https://github.com/wolfSSL/wolfssl/files/7819605/test-suite.log)

embhorn, 2022-01-06:
Hi @ryandesign 

Is this a persistent failure? Could you please share the wolfSSL configuration used to reproduce this?

ryandesign, 2022-01-06:
It happened all three times I tried the tests (once with 5.0.0, twice with 5.1.1), so yes, so far it is persistent.

`./configure` was run with `--prefix=/opt/local --enable-distro --disable-jobserver --disable-silent-rules`. The environment variables that were set were:

```
CC='ccache /usr/bin/clang'
CC_PRINT_OPTIONS='YES'
CC_PRINT_OPTIONS_FILE='/opt/local/var/macports/build/_Users_rschmidt_macports_macports-ports-ryandesign-fork_devel_wolfssl/wolfssl/work/.CC_PRINT_OPTIONS'
CFLAGS='-pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64'
CPATH='/opt/local/include'
CPPFLAGS='-I/opt/local/include -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk'
CXX='ccache /usr/bin/clang++'
CXXFLAGS='-pipe -Os -stdlib=libc++ -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64'
DEVELOPER_DIR='/Library/Developer/CommandLineTools'
F90FLAGS='-pipe -Os -m64'
FCFLAGS='-pipe -Os -m64'
FFLAGS='-pipe -Os -m64'
INSTALL='/usr/bin/install -c'
LDFLAGS='-L/opt/local/lib -Wl,-headerpad_max_install_names -Wl,-syslibroot,/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64'
LIBRARY_PATH='/opt/local/lib'
MACOSX_DEPLOYMENT_TARGET='10.15'
OBJC='ccache /usr/bin/clang'
OBJCFLAGS='-pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64'
OBJCXX='ccache /usr/bin/clang++'
OBJCXXFLAGS='-pipe -Os -stdlib=libc++ -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64'
SDKROOT='/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk'
```

The summary output from the configure script was:

```
Configuration summary for wolfssl version 5.1.1

   * Installation prefix:        /opt/local
   * System type:                apple-darwin19.6.0
   * Host CPU:                   x86_64
   * C Compiler:                 ccache /usr/bin/clang
   * C Flags:                    -pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64   -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wshorten-64-to-32 -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:               ccache /usr/bin/clang++
   * C++ Flags:                  -pipe -Os -stdlib=libc++ -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64
   * CPP Flags:                  -I/opt/local/include -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk
   * CCAS Flags:                 -pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -arch x86_64
   * LIB Flags:
   * Debug enabled:              no
   * Coverage enabled:
   * Warnings as failure:        no
   * make -j:                    no
   * VCS checkout:               no

   Features
   * FIPS:                       no
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              yes
   * OpenSSL Extra API:          yes
   * OpenSSL Coexist:            no
   * Old Names:                  no
   * Max Strength Build:         no
   * Distro Build:               yes
   * Reproducible Build:         yes
   * fastmath:                   yes
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       yes
   * AES:                        yes
   * AES-NI:                     no
   * AES-CBC:                    yes
   * AES-CBC length checks:      yes
   * AES-GCM:                    yes
[... 135 lines cut ...]
   * Fast RSA:                   no
   * Single Precision:           no
   * SP math implementation:     no
   * Async Crypto:               no
   * PKCS#8:                     yes
   * PKCS#11:                    no
   * PKCS#12:                    yes
   * Cavium Nitrox:              no
   * Cavium Octeon (Sync):       no
   * Intel Quick Assist:         no
   * ARM ASM:                    no
   * ARM ASM SHA512 Crypto
   * AES Key Wrap:               yes
   * Write duplicate:            no
   * Xilinx Hardware Acc.:       no
   * Inline Code:                yes
   * Linux AF_ALG:               no
   * Linux KCAPI:                no
   * Linux devcrypto:            no
   * Crypto callbacks:           yes
   * i.MX6 CAAM:                 no
   * IoT-Safe:                   no
   * IoT-Safe HWRNG:             no
   * NXP SE050:                  no
```

ryandesign, 2022-01-06:
I was doing with [MacPorts](https://ports.macports.org/port/wolfssl) by running `sudo port test wolfssl`.

dgarske, 2022-01-07:
Hi @ryandesign ,

Looks like an issue with synchronization due to the ready file using the same name.

In my logs here it shows:

```
------------- TEST CASE 7 LOAD CERT IN SSL -------------------

removing ready file: /Users/davidgarske/GitHub/wolfssl/workspace.pid91533/wolf_ocsp_s2_readyF591533
waiting for ready file...
Also load cert/key into wolfSSL object
found ready file, starting client...
CONNECTED(00000003)
```

In yours the logs immediately show `found ready file, starting client...` and the openssl client tries to connect before the server is ready.

I am working on a fix and will point you to it shortly.

Thanks,
David Garske, wolfSSL

dgarske, 2022-01-07:
Hi @ryandesign ,

Actually it is removing the file before, but the server thinks it still exists. Is there something MacPorts is doing with the file system that could cause this sync behavior? If you add a call to `sync` or even `sleep 0.1` does the behavior resolve?

I am not familiar with MacPorts use to reproduce. Perhaps you can share some instructions? I am on `macOS 11.6.1`.

Thanks,
David Garske, wolfSSL

ryandesign, 2022-01-07:
I am using the APFS filesystem, which does have different characteristics than the old HFS+ filesystem, but all users on macOS 10.14 and later are using APFS since the OS upgrade auto-converts HFS+ disks to APFS.

Based on your comments I thought it might be due to running the tests in parallel with `make check -j8`. Normally in MacPorts we don't run tests in parallel but because wolfssl's build system is unusual and enables parallel building and testing by default, I figured it would be ok. But the test failure remains if I test serially with `make check -j1` so that's not it.

To rule out any problems with MacPorts, I tried building 5.1.1 outside of MacPorts using just `./configure && make check`. It only ran 7 tests which all passed, which didn't include ocsp-stapling2.test. Then I tried `./configure --enable-distro && make check` which ran 14 tests, 13 of which passed; ocsp-stapling2.test still failed, so that rules out any MacPorts involvement. I also tried `./configure --disable-jobserver --enable-distro && make -j8 && make check` with the same result.

I tried adding `sync` or `sleep 1` at the end of `remove_single_rF` in ocsp-stapling2.test; no change.

If I run this test by itself outside of `make` by running `./scripts/ocsp-stapling2.test; echo $?` it still fails with exit status 1.

dgarske, 2022-01-07:
Hi @ryandesign,

This test relies on the openssl command line to setup a OCSP responder. What does your `openssl version` report? I do not see that issue here on my Mac. Perhaps you can try building wolfSSL with `--enable-debug` also and provide the log running `./scripts/ocsp-stapling2.test`?

Thanks,
David Garske, wolfSSL

ryandesign, 2022-01-07:
MacPorts recently updated to openssl 3...
```
$ openssl version
OpenSSL 3.0.1 14 Dec 2021 (Library: OpenSSL 3.0.1 14 Dec 2021)
```

ryandesign, 2022-01-07:
Going back to openssl 1.1 fixes the test.
```
$ /opt/local/libexec/openssl11/bin/openssl version
OpenSSL 1.1.1l  24 Aug 2021
```

dgarske, 2022-01-07:
Hi @ryandesign ,

I can reproduce with `openssl version` `OpenSSL 3.0.1 14 Dec 2021 (Library: OpenSSL 3.0.1 14 Dec 2021)`.

```
SSL_accept error -132, Buffer error, output too small or input too big
wolfSSL error: SSL_accept failed
```

@embhorn will you take a look or assign to someone?

Thanks,
David Garske, wolfSSL

ryandesign, 2022-01-07:
> ```
> SSL_accept error -132, Buffer error, output too small or input too big
> wolfSSL error: SSL_accept failed
> ```

This error does also appear in many of the other tests which purportedly completed successfully.

elms, 2022-01-07:
@ryandesign I'm reproducing the error and digging into it. Can you provide a list of any other specific occurrences?

ryandesign, 2022-01-07:
Here's what I see:
```
$ grep 'output too small or input too big' -r scripts
scripts/ocsp-stapling.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log.standalone:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log.standalone:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log.standalone:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log.standalone:SSL_accept error -132, Buffer error, output too small or input too big
scripts/ocsp-stapling2.log.standalone:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/unit.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/crl-revoked.log:SSL_accept error -132, Buffer error, output too small or input too big
scripts/crl-revoked.log:SSL_accept error -132, Buffer error, output too small or input too big
```

elms, 2022-01-07:
@ryandesign After digging into the error I was able to pass the test by using `./configure --enable-secure-renegotiation --enable-ocspstapling2`. openssl3 appears to have changed a warning to an error for use of legacy renegotiation.

Instead of requiring secure renegotiation, I've changed the test in https://github.com/wolfSSL/wolfssl/pull/4735 to address the test with openSSL 3. I'll look for the other -132 errors you reported now.

elms, 2022-01-13:
@ryandesign I'm closing as this is fixed with #4742. Thanks for the report and let us know if you run into anything else.

closed 2022-01-13 by elms