wolfSSL/wolfssl #4737: --enable-all --enable-sessionexport

douzzer, 2022-01-08:
unit.test fails thusly on an `--enable-all --enable-sessionexport` build:
```
ssl->version.major=254 exp[idx-2]=254 ssl->version.minor=253 exp[idx-1]=255
src/internal.c L1390 ImportOptions VERSION_ERROR

ERROR - tests/api.c line 4299 failed with:
    expected: wolfSSL_dtls_import(ssl, buf, sz) >= 0
    result:   -326 < 0
```

ejohnstown, 2022-12-29:
The exported version is {0xFE,0xFF}(DTLSv1.0). The import is expecting a version of {0xFE,0xFD}(DTLSv1.2). wolfSSL_dtls_import() returns VERSION_ERROR. The test case is expecting no error.

The test case is test_wc_CryptoCb_TLS(). It is setting up the callbacks data for the server with the correct SSL method function. The server thread is run with the function test_export() and it always uses wolfDTLSv1_2_server_method(). The function wolfSSL_send_session() calls the callback function, and passes NULL as the userCtx.

The actual export test only tests with DTLSv1.2.

ejohnstown, 2022-12-30:
Fixed by #5943. Thanks @JacobBarthelmeh.

closed 2022-12-30 by ejohnstown