wolfSSL/wolfssl #4744: XSTAT used as a structure and function definition makes it hard to use properly

CookiePLMonster, 2022-01-11:
If one wants to make WolfSSL use custom IO functions, they may define their own macroes for functions in `user_settings.h`. However, in the case of `XSTAT` it's needlessly hard to do since WolfSSL uses that macro both for the structure definition and the name of the function call.

For one example, see this PR: https://github.com/RPCS3/rpcs3/pull/11359 where `XFOPEN` is defined to point towards a custom wrapper around `_wfopen`, allowing WolfSSL to use UTF-8 paths transparently on Windows. Technically, for `XSTAT` a similar function could (and should) have been defined, with a theoretical declaration of:
```c
int wolfSSL_stat_utf8(
   const char *path,
   struct _stat *buffer
);
```

However, this currently seems impossible as using
```c
#define XSTAT       wolfSSL_stat_utf8
```

will result in compilation errors when looking for `struct wolfSSL_stat_utf8`. Additionally, since `_stat` is a macro itself, it's tricky to get right without potentially locally mirroring the definition of `struct _stat`.

It would be much better and easier to use if WolfSSL separated `_stat` and `struct _stat` as two independent macros.

kareem-wolfssl, 2022-01-14:
Hi @CookiePLMonster,

Good point, I've created a PR to separate out `struct _stat`'s definition: https://github.com/wolfSSL/wolfssl/pull/4757/files
Let me know if it fixes the issue for you.

Thanks,
Kareem

CookiePLMonster, 2022-01-15:
Thank you! I can confirm this fixes the usage issue, I have created a proof-of-concept PR utilizing this feature and showing it is functional now: https://github.com/RPCS3/rpcs3/pull/11378

kareem-wolfssl, 2022-03-07:
@CookiePLMonster The PR was merged with a slight change, we are now using `XSTAT_TYPE` to override the `struct XSTAT` definition to allow for more flexibility.  So you will need to `#define XSTAT_TYPE struct _stat`.

CookiePLMonster, 2022-03-19:
Thank you! I have now updated the dependent PR https://github.com/RPCS3/rpcs3/pull/11378 and I can confirm that these changes work fine.

closed 2022-03-19 by CookiePLMonster