wolfSSL/wolfssl #4763: wolfSSL_EVP_get_cipherbyname - missing uppercase cipher names

t0x1c, 2022-01-16:
wolfSSL_EVP_get_cipherbyname does not recognize uppercase AES128/AES256 names (trying to port OpenSSL code).
Internal lookup table contains upper and lower case equivalents for cipher names, except for AES128 and AES256 (and maybe newer ones with GCM). Not a big deal. Just a consistency matter.

SparkiDev, 2022-01-17:
Hi @t0x1c,

A pull request has been put up to support any case alternative versions: https://github.com/wolfSSL/wolfssl/pull/4764

Let me know if this covers your use cases!

Thanks,
Sean :-)

t0x1c, 2022-01-17:
Hi Sean,
I just renamed my variables to lowercase to work around this for the time being. Since you changed it to case-insensitive comparison it should cover all possible cases. Thanks.

closed 2022-01-17 by t0x1c