wolfSSL/wolfssl #4766: recursive calls in sp_int.c and ssl.c

douzzer, 2022-01-17:
`clang-tidy` analyzer `misc-no-recursion` uncovered 4 recursive codepoints, all 4 of which are direct recursion, not indirect via a larger cycle:
```
/home/douzzer/com/wolfssl/src/wolfssl/wolfcrypt/src/sp_int.c:8119:5: warning: function 'sp_invmod' is within a recursive call chain [misc-no-recursion]
int sp_invmod(sp_int* a, sp_int* m, sp_int* r)
    ^
/home/douzzer/com/wolfssl/src/wolfssl/src/ssl.c:32874:5: warning: function 'wolfSSL_ASN1_item_i2d' is within a recursive call chain [misc-no-recursion]
int wolfSSL_ASN1_item_i2d(const void *src, byte **dest,
    ^
/home/douzzer/com/wolfssl/src/wolfssl/src/ssl.c:45519:9: warning: function 'wolfSSL_OBJ_obj2txt' is within a recursive call chain [misc-no-recursion]
    int wolfSSL_OBJ_obj2txt(char *buf, int bufLen, const WOLFSSL_ASN1_OBJECT *a,
        ^
/home/douzzer/com/wolfssl/src/wolfssl/src/ssl.c:59549:5: warning: function 'wolfSSL_BIO_dump' is within a recursive call chain [misc-no-recursion]
int wolfSSL_BIO_dump(WOLFSSL_BIO *bio, const char *buf, int length)
    ^
```

If these recursions are formally bounded and deemed acceptable, we can add point suppressions for them, but it would be neater to refactor them away entirely.

SparkiDev, 2022-01-17:
sp_invmod calls itself when the modulus is even. It swaps the modulus and value to invert and does fixups after. (Bails earlier if both modulus and value are even.)
I took this pattern from the tfm.c code.
I'll take a look at reworking it.

douzzer, 2022-02-01:
This is resolved for `sp_invmod()` (#4768) and `wolfSSL_ASN1_item_i2d()` (#4772).

`wolfSSL_OBJ_obj2txt()` and `wolfSSL_BIO_dump()` are still marked `NOLINT(misc-no-recursion)`.

`wolfSSL_OBJ_obj2txt()` recursion is formally bounded at max depth 2 and doesn't need refactoring.

`wolfSSL_BIO_dump()` does still need refactoring.

PR #4814 (2022-02-02): wolfSSL_BIO_dump: fix output format and make iterative
# Description

wolfSSL_BIO_dump(): fix output format and make iterative
wolfSSL_OBJ_obj2txt(): make iterative, test and rework.

Fixes #4766

# Testing

Added test into api.c.

# Checklist

 - [x] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-02-02 by dgarske