wolfSSL/wolfssl #4786: [Bug]: possible integer overflow in sp_384_mont_reduce_order_15()

rod-chapman, 2022-01-21:
### Contact Details

rod@proteancode.com

### Version

5.1.1

### Description

Building with HAVE_ECC, HAVE_ECC_SIGN, with curve P384 and SHA384.

In sp_384_mont_reduce_order_15() in sp_c32.c, the variable mu is calculated as follows:
        mu = (a[i] * mp) & 0x3ffffff;
on line 28350 and similarly on line 28354.

With mp = 384_mp_order = 0x8fdc45

it looks like that "*" is for signed 32-bit integer, so suffers from a potential for undefined behaviour on overflow before the "&" operator is applied.

Is there some unstated precondition on the values of a[i] that I have missed?

### Reproduction steps

None.

dgarske, 2022-01-21:
Hi @rod-chapman , thank you for the detailed report. @SparkiDev will look into this when he is back in the office Monday.

SparkiDev, 2022-01-23:
Hi @rod-chapman,

All words in 'a' will contain no more than 27-bits and will be a positive number.
mu is fixed for the order and is 27-bits.
Only 27-bits of the multiplication result are required.

Does this answer your concerns?

rod-chapman, 2022-01-24:
Not really... if all the elements of a are limited to 27 bits, and mp (not mu) is 27 bits, then you need at least 54 bits to store the intermediate result before the "&" operation is applied. It doesn't matter that a is positive - the underlying type is signed, therefore this is an undefined behaviour. (It might happen to work OK for all current targets and compilers, but that is not guaranteed by the C standard.)

I think you should cast both argument to 64-bits, do the multiply and & operations in 64-bit before casting the result back to 32-bit to be both portable and well-defined.

I discovered this by re-writing the code in Rust. It "panics" every time it is run.

SparkiDev, 2022-01-27:
Hi @rod-chapman

While it is undefined, I haven't had a problem with this as of yet.
But in order to obviate a problem in the future, and to stop Rust complaining, I've put a pull request up with the change to make the operands the unsigned type.

Let me know if there are more issues like this in the code or Rust still 'panics'.

rod-chapman, 2022-01-29:
Will do. I think it's worth fixing to future-proof your code. It's also a blatant MISRA violation, of course.
I will keep you posted if my Rust code raises any more issues.

SparkiDev, 2022-01-31:
The pull request is merged.
Can this issue be closed?

rod-chapman, 2022-02-01:
Looks good to me.

SparkiDev, 2022-02-01:
Thanks @rod-chapman!

closed 2022-02-01 by SparkiDev