wolfSSL/wolfssl #4788: [Bug]: wc_ecc_encrypt Not Found When Linking Against Static Linux SGX Library

xanderdunn, 2022-01-24:
### Version

v5.1.1-stable

### Description

I am using the SGX static library [here](https://github.com/wolfSSL/wolfssl/tree/master/IDE/LINUX-SGX). I am successfully building and running the wolfSSL SGX example provided [here](https://github.com/wolfSSL/wolfssl-examples/tree/master/SGX_Linux).

The [ECC API Reference](https://www.wolfssl.com/doxygen/group__ECC.html#gafc55ff3a455341d143786e8a27c8bbbe) shows function `wc_ecc_encrypt` for encrypting plaintext with an ECC key. Additionally, the function is in [ecc.h](https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/ecc.h#L870). However, the linker is unable to find this symbol when I call `wc_ecc_encrypt` from Wolfssl_Enclave.c when linking to the WolfSSL static library.

### Reproduction steps

- Build wolfSSL static library for SGX: `make -f sgx_t_static.mk HAVE_WOLFSSL_SP=1`
- Add a new function to the Wolfssl_Enclave.c:
```c
void ecall_create_ecc_key() {
     ecc_key key;
     WC_RNG rng;
     int is_valid;
     wc_ecc_init(&key);
     wc_InitRng(&rng);
     int curveId = ECC_SECP521R1;
     int keySize = wc_ecc_get_curve_size_from_id(curveId);
     /*ecall_printf("Creating WolfSSL ecc key...\n");*/
     int success = wc_ecc_make_key(&rng, keySize, &key);
     if (success != 0) {
         /*ecall_printf("Failed to create WolfSSL ECC key, error code %d!\n", success);*/
     }

     byte msg[] = { "Test123" };
     byte out[sizeof(msg)];
     word32 outSz = sizeof(out);

     success = wc_ecc_encrypt(&key, &key, msg, sizeof(msg), out, &outSz, NULL);
     if (success != 0) {
         /*ecall_printf("Failed to encrypt using ECC key, error code %d!\n", success);*/
     }
     //wc_ecc_decrypt();
 }
```
- Build the wolfSSL SGX example: `make SGX_MODE=HW SGX_PRERELEASE=1 SGX_WOLFSSL_LIB=/home/xander/dev/wolfssl/IDE/LINUX-SGX/ WOLFSSL_ROOT=/home/xander/dev/wolfssl SGX_DEBUG=0`

### Relevant log output

```shell
- The build fails:

trusted/Wolfssl_Enclave.o: In function `ecall_create_ecc_key':
Wolfssl_Enclave.c:(.text+0x721): undefined reference to `wc_ecc_encrypt'
collect2: error: ld returned 1 exit status
```
- Indeed, the symbol is not in the the static library: `nm ~/dev/wolfssl/IDE/LINUX-SGX/libwolfssl.sgx.static.lib.a | grep wc_ecc_encrypt` finds nothing, although some of the other symbols such as `wc_ecc_make_key` are there.
```

xanderdunn, 2022-01-24:
Thanks Eric!

Running `./config.status` before build is described in the [SGX-specific build instructions](https://github.com/wolfSSL/wolfssl/tree/master/IDE/LINUX-SGX#overview-and-build).

I reconfigured and then ran `make` without calling `./config.status` and can confirm that all of my desired options remain in wolfssl/options.h before I do the static build, but it still has no effect on the symbols built into the static library. It doesn't appear that the [static build makefile](https://github.com/wolfSSL/wolfssl/blob/master/IDE/LINUX-SGX/sgx_t_static.mk) is making use of the options chosen by `./configure`.

However, when I manually add the options I want to that makefile:
```makefile
 # TLS 1.3
Wolfssl_C_Extra_Flags += -DWOLFSSL_TLS13 -DHAVE_TLS_EXTENSIONS -DWOLFSSL_ECIES_ISO18033 -DHAVE_FFDHE_2048 -DWC_RSA_PSS -DWOLFSSL_ECIES  _ISO18033 -DNO_OLD_TLS -DWOLFSSL_NO_TLS12

# ECC
Wolfssl_C_Extra_Flags += -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DHAVE_ECC_ENCRYPT -DHAVE_ECC384
```
It does build in the symbols I want:
```
$ nm libwolfssl.sgx.static.lib.a | grep wolfTLSv1_3_client_method
00000000000033a0 T wolfTLSv1_3_client_method
00000000000033d0 T wolfTLSv1_3_client_method_ex
$ nm libwolfssl.sgx.static.lib.a | grep wc_ecc_encrypt
0000000000006860 T wc_ecc_encrypt
```

This will work for me for the time being, but let me know if there's a more elegant way of making use of `configure`'s options

xanderdunn, 2022-01-24:
With the above changes to the Makefile, I'm having difficulty building and running with TLS 1.3 rather than 1.2. I see these warnings during compile:
```
trusted/Wolfssl_Enclave.c: In function ‘GetMethod’:
trusted/Wolfssl_Enclave.c:190:40: warning: returning ‘int’ from a function with return type ‘WOLFSSL_METHOD *’ makes pointer from integer without a cast [-Wint-conversion]
  190 |         case WOLFTLSv13_CLIENT: return wolfTLSv1_3_client_method();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
trusted/Wolfssl_Enclave.c:191:40: warning: returning ‘int’ from a function with return type ‘WOLFSSL_METHOD *’ makes pointer from integer without a cast [-Wint-conversion]
  191 |         case WOLFTLSv13_SERVER: return wolfTLSv1_3_server_method();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```
Apparently it thinks `wolfTLSv1_3_client_method()` and `wolfTLSv1_3_server_method()` are returning `int` when they should be returning `WOLFSSL_METHOD *`. 

And sure enough it segfaults at runtime when the code calls `wolfSSL_CTX_new()` on the return value of `wolfTLSv1_3_server_method`. I'm not sure where it's getting the idea that `wolfTLSv1_3_server_method()` returns an int. Double checked that I'm on 5.1.1 and the signatures appear correct. In wolfssl/ssl.h:
```
WOLFSSL_ABI WOLFSSL_API WOLFSSL_METHOD *wolfTLSv1_3_server_method(void);
 ```
In src/tls.c:
```
WOLFSSL_METHOD* wolfTLSv1_3_server_method(void)
```

Any ideas where it would be getting the `int` return value from / what I might be doing wrong to build in TLS 1.3 support?

JacobBarthelmeh, 2022-01-26:
Hi @xanderdunn, for now I would suggest adding the macro's desired to the wolfssl/wolfcrypt/settings.h file (https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/settings.h#L1761). Or by setting up a user_settings.h file and using the macro WOLFSSL_USER_SETTINGS when compiling both the wolfSSL library and the applications.

My guess is that since the declaration of wolfTLSv1_3_server_method is guarded by the macro WOLFSSL_TLS13 that when the header files are being included it is not be declared, and defaulting to the 'int' return value of an undeclared function. Defining the extra macros in wolfssl/wolfcrypt/settings.h should resolve this.

The ./config.status helps by clearing out wolfssl/options.h of previous build debris.

Warm Regards,
Jacob

xanderdunn, 2022-01-30:
Thanks @JacobBarthelmeh. Adding this to wolfssl/wolfcrypt/settings.h:
```
// ECC
#define TFM_ECC256
#define ECC_SHAMIR
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC384
#define HAVE_HKDF
#define WOLFSSL_X86_64_BUILD

// TLS 1.3
#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define WOLFSSL_ECIES_ISO18033
#define HAVE_FFDHE_2048
#define WC_RSA_PSS
#define WOLFSSL_ECIES_ISO18033
#define NO_OLD_TLS
```

Now both the headers and the library symbols are found. Build successful 👍 

When setting the wolfssl-exmaples/SGX_Linux to use TLS 1.3 rather than 1.2, the client is now failing to connect, but that's likely a difference in usage that's outside the scope of this issue.

closed 2022-01-30 by xanderdunn