wolfSSL/wolfssl #4791: SHA1 client certificate validation fails if NO_OLD_TLS is specified

t0x1c, 2022-01-24:
### Version

5.1.1

### Description

wolfSSL server is configured to validate client certificate. Client presents a certificate signed with RSA/SHA1.
NO_OLD_TLS is specified, which cancels out SHA1. But it's brought back with WOLFSSL_ALLOW_TLS_SHA1.

DoCertificateVerify fails: after call to wc_EncodeSignature following XMEMCMP fails with non-zero result.

Looks like some conditional parts of the code check for both flags (like !NO_OLD_TLS || WOLFSSL_ALLOW_TLS_SHA1), but some other SHA-related parts only check for NO_OLD_TLS, so some parts of the code don't get executed (example: HashRaw in internal.c)

embhorn, 2022-01-26:
@t0x1c 

I've created #4793 to fix this issue. Please let me know if this addresses your concerns in this issue.

t0x1c, 2022-02-01:
Sorry for late response. I tried to apply your changes to 5.1.1 release code, and it looks like it's not enough - server still assumes certificate invalid. I searched for NO_OLD_TLS usage and there are many more conditional places involving SHA computations, though I'm not sure which of them are used for certificate validation: InitHandshakeHashes, FreeHandshakeHashes, BuildFinished, SendCertificateVerify, SendServerKeyExchange, maybe more..

t0x1c, 2022-06-02:
thanks! my hands are full, cant verify it right now, but assuming the changes it should work

embhorn, 2024-05-15:
Fixed by #7070

closed 2024-05-15 by embhorn