wolfSSL/wolfssl #4796: curve25519_shared_secret doesn't match RFC7748 example in section 6.1

petertorelli, 2022-01-26:
### Version

5.1.0

### Description

In [RFC7748 Section 6.1](https://datatracker.ietf.org/doc/html/rfc7748#section-6.1) there are some example points used to compute a shared secret.

First I init the keys, load the private, check the public, then load the public (abbreviated a little so it's easier to read):
```c
int ret;
curve25519_key pri, pub;
wc_curve25519_init(&pri);
wc_curve25519_init(&pub);
ret = wc_curve25519_import_private(p_private, prilen, &pri);
// check ret...
```
Here I notice that the private key is clamped according to Section 5 during the import, but the public key is not scaled, so I do that otherwise the check fails...
```c
// RFC7748 Section 5
p_public[0] &= 127;
ret = wc_curve25519_check_public(p_public, publen, EC25519_BIG_ENDIAN);
// check ret...
ret = wc_curve25519_import_public(p_public, publen, &pub);
// check ret...
```

Now with both keys loaded I compute a shared secret:

```c
ret = wc_curve25519_shared_secret(&pri, &pub, p_secret, &slen);
```

When I print the secret in big-endian, I don't get the expected results, from the RFC:

```
   Alice's private key, a:
     77076d0a7318a57d3c16c17251b26645df4c2f87ebc0992ab177fba51db92c2a
: 
: 
   Bob's public key, X25519(b, 9):
     de9edb7d7b7dc1b4d35b61c2ece435373f8343c85b78674dadfc7e146f882b4f
   Their shared secret, K:
     4a5d9d5ba4ce2de1728e3bf480350f25e07e21c947d19e3376f09b3c1e161742
```

The shared secret I get doesn't match... (debug printfs)...

```
pri: 0x77076d0a7318a57d3c16c17251b26645df4c2f87ebc0992ab177fba51db92c2a
pub: 0xde9edb7d7b7dc1b4d35b61c2ece435373f8343c85b78674dadfc7e146f882b4f
sec: 0x586dec12a4df5464289affb7368691c7aad7c35440f3c064b868646decfc44d1
```

...aaaand a sanity check:

```python
% python3
Python 3.9.9 (main, Nov 21 2021, 03:23:44) 
[Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from binascii import hexlify
>>> import x25519
>>> pri = bytes.fromhex('77076d0a7318a57d3c16c17251b26645df4c2f87ebc0992ab177fba51db92c2a')
>>> pub = bytes.fromhex('de9edb7d7b7dc1b4d35b61c2ece435373f8343c85b78674dadfc7e146f882b4f')
>>> sec = x25519.scalar_mult(pri, pub);
>>> print(hexlify(sec))
b'4a5d9d5ba4ce2de1728e3bf480350f25e07e21c947d19e3376f09b3c1e161742'
```

All the imports default to big-endian which is how the bytes are stored in the C buffers, and the keys are initialized. Not sure what else I'm forgetting to do.

embhorn, 2022-01-26:
Hello @petertorelli 

Could you please share the wolfSSL configuration you are using?

petertorelli, 2022-01-26:
Hi @embhorn,

First, I'm using 5.1.1-stable, not 5.1.0. Here's my invocation:

```
$ ./configure 'CFLAGS=-DWOLFSSL_AES_DIRECT -DHAVE_AES_ECB -DWOLFSSL_ECDSA_DETERMINISTIC_K' --enable-ecc --enable-keygen --enable-aesccm --enable-sp --enable-sp-asm --enable-eccencrypt --enable-curve25519 --enable-ed25519
```

dgarske, 2022-01-26:
Hi @petertorelli ,

Took me a bit to write up that test case. I can reproduce both issues.
1) The public check if not clamped `bob_pub[0] &= 127;` -> `Error -217:  ECC Qx or Qy out of range error`
2) Different shared secret then test vector. 
```
Got this: 586dec12a4df5464289affb7368691c7aad7c35440f3c064b868646decfc44d1
Expected this: 4a5d9d5ba4ce2de1728e3bf480350f25e07e21c947d19e3376f09b3c1e161742
```

I am looking into this and suspect it has to do with the clamping when importing keys manually.

The test case has been pushed here for now: https://github.com/dgarske/wolfssl-examples/blob/curve25519_test/pk/curve25519/curve25519_test.c

Thanks,
David Garske, wolfSSL

petertorelli, 2022-01-26:
Thanks David. I have a test case in GitHub, but it is part of a larger project I was working on with Jacob Barthelmeh at EEMBC for [SecureMark](https://github.com/eembc/securemark-tls/blob/v2/examples/selfhosted/main.c) and wasn't sure if you had a simpler testbench.

SparkiDev, 2022-01-27:
I modified hex2bin() in curve25519_test.c to place the bytes in reverse order and the test passes.
Please confirm this works for you too.

Thanks,
Sean

SparkiDev, 2022-01-27:
Also, importing and exporting with EC25519_LITTLE_ENDIAN worked for me also.

dgarske, 2022-01-27:
Hi All,

Confirmed it is an endianness issue with the order of the byes in the test vectors. I've push an update to the example and put up a pull request here: https://github.com/wolfSSL/wolfssl-examples/pull/285

@petertorelli let me know if you see any other issues.

Thanks,
David Garske, wolfSSL

petertorelli, 2022-01-27:
Thank you for confirming this: when I swapped endianness during my first test I had mistakenly masked the wrong byte which gave me a -217 (`ECC_OUT_OF_RANGE`).

The import public key function does not mask the HOB against 127 (unlike import private which clamps). Just to confirm, does this mean it still must by done prior to import?

closed 2022-01-27 by petertorelli

dgarske, 2022-01-27:
Hi @petertorelli ,

I cannot find any requirement from the RFC to mask the signed bit on the public key import. However I do see a note about the final result being masked to avoid the signed bit being set. With the endianness setup correctly the MSB is not set on the public key portion and the generated public key computed from the private key will also never have it set.

@SparkiDev do you have any feedback for @petertorelli  on his question `The import public key function appears to not mask the HOB against 127 (unlike import private which clamps). Just to confirm, does this mean it still must by done prior to import`?

Thanks,
David Garske, wolfSSL

petertorelli, 2022-01-27:
@dgarske Thanks again!

SparkiDev, 2022-01-27:
Hi @petertorelli

The clamping happens at time of private key generation.
An exported private key will have been clamped and so importing doesn't need to clamp.
We expect that any externally generated private key to have been clamped.

Sean