wolfSSL/wolfssl #4835: [Bug]: SP math 32 bit build fails

guidovranken, 2022-02-07:
### Contact Details

guidovranken@gmail.com

### Version

Git repository checkout

### Description

SP math build fails on 32 bit

This is currently blocking an OSS-Fuzz update: https://github.com/google/oss-fuzz/pull/7226

### Reproduction steps

```sh
export CFLAGS="$CFLAGS -DHAVE_AES_ECB -DWOLFSSL_DES_ECB -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DWOLFSSL_ECDSA_SET_K -DWOLFSSL_ECDSA_SET_K_ONE_LOOP -DWOLFSSL_PUBLIC_ECC_ADD_DBL -m32"
git clone --depth 1 https://github.com/wolfSSL/wolfssl.git
cd wolfssl/
autoreconf -ivf
./configure --enable-static --enable-md2 --enable-md4 --enable-ripemd --enable-blake2 --enable-blake2s --enable-pwdbased --enable-scrypt --enable-hkdf --enable-cmac --enable-arc4 --enable-camellia --enable-aesccm --enable-aesctr --enable-xts --enable-des3 --enable-x963kdf --enable-harden --enable-aescfb --enable-aesofb --enable-aeskeywrap --enable-aessiv --enable-keygen --enable-curve25519 --enable-curve448 --enable-shake256 --disable-crypttests --disable-examples --enable-compkey --enable-ed448 --enable-ed25519 --enable-xchacha --enable-cryptocb --enable-eccencrypt --enable-smallstack --enable-ed25519-stream --enable-ed448-stream --enable-sp --enable-sp-math
make
```

### Relevant log output

```shell
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-rsa.o: in function `wc_RsaFunction':
rsa.c:(.text+0x1a8f): undefined reference to `sp_RsaPrivate_4096'
/usr/bin/ld: rsa.c:(.text+0x1b1d): undefined reference to `sp_RsaPublic_2048'
/usr/bin/ld: rsa.c:(.text+0x1be5): undefined reference to `sp_RsaPublic_4096'
/usr/bin/ld: rsa.c:(.text+0x1c9a): undefined reference to `sp_RsaPrivate_3072'
/usr/bin/ld: rsa.c:(.text+0x1ce1): undefined reference to `sp_RsaPublic_3072'
/usr/bin/ld: rsa.c:(.text+0x1ddd): undefined reference to `sp_RsaPrivate_2048'
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-sp_int.o: in function `sp_exptmod_ex':
sp_int.c:(.text+0x3386): undefined reference to `sp_ModExp_2048'
/usr/bin/ld: sp_int.c:(.text+0x341c): undefined reference to `sp_ModExp_1536'
/usr/bin/ld: sp_int.c:(.text+0x3468): undefined reference to `sp_ModExp_3072'
/usr/bin/ld: sp_int.c:(.text+0x34b0): undefined reference to `sp_ModExp_1024'
/usr/bin/ld: sp_int.c:(.text+0x3504): undefined reference to `sp_ModExp_4096'
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-dh.o: in function `_ffc_validate_public_key':
dh.c:(.text+0x268): undefined reference to `sp_ModExp_2048'
/usr/bin/ld: dh.c:(.text+0x338): undefined reference to `sp_ModExp_3072'
/usr/bin/ld: dh.c:(.text+0x370): undefined reference to `sp_ModExp_4096'
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-dh.o: in function `wc_DhCheckKeyPair':
dh.c:(.text+0xbab): undefined reference to `sp_ModExp_4096'
/usr/bin/ld: dh.c:(.text+0xbe1): undefined reference to `sp_ModExp_3072'
/usr/bin/ld: dh.c:(.text+0xc21): undefined reference to `sp_ModExp_2048'
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-dh.o: in function `wc_DhAgree':
dh.c:(.text+0x14cc): undefined reference to `sp_DhExp_3072'
/usr/bin/ld: dh.c:(.text+0x1524): undefined reference to `sp_DhExp_2048'
/usr/bin/ld: dh.c:(.text+0x154e): undefined reference to `sp_DhExp_4096'
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-dh.o: in function `wc_DhGenerateKeyPair':
dh.c:(.text+0xec4): undefined reference to `sp_DhExp_4096'
/usr/bin/ld: dh.c:(.text+0x11c2): undefined reference to `sp_DhExp_2048'
/usr/bin/ld: dh.c:(.text+0x1212): undefined reference to `sp_DhExp_3072'
/usr/bin/ld: src/.libs/libwolfssl.so.30.1.2: hidden symbol `sp_ModExp_1024' isn't defined
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
```
```

SparkiDev, 2022-02-08:
Hi @guidovranken

I've put up the pull request: https://github.com/wolfSSL/wolfssl/pull/4837
I believe your configuration will work now and not break others.

SparkiDev, 2022-02-14:
Hi @guidovranken

The pull request has been merged.
Is this all working for you now?

guidovranken, 2022-02-14:
Yes the issue is fixed, thanks.

closed 2022-02-14 by guidovranken