wolfSSL/wolfssl #4850: Handle DTLS Hello packets without creating state.

mkm85, 2022-02-10:
### Version

v5.1.1

### Description

I have a use case where wolfSSL is used in a server process (built on top of asio). This process handles DTLS 1.2 connections.

I would like not to maintain a WOLFSSL object for each unverified Client Hello packet received ie. Client Hello packets without a cookie. The reason is that an attack using spoofed source addresses would make it trivial to exhaust the servers resources.

```
Flow diagram explaining the desired behavior
client ---ClientHello(no cookie)-------------> server
client <---HelloVerificationRequest(cookie)--- server (no state/connection is maintained in the server yet)
client ---ClientHello(with valid cookie)---->  server (The server creates and maintains state/connection)
.... rest of the handshake
```

I have read the documentation and the source code, there does not seem to be a way to obtain the desired behavior. 

I have made an experiment where the WOLFSSL obejct is freed if a HelloVerificationRequest is sent. But in that case the message_seq in the handshake is incompatible with what e.g. a MbedTLS client expects. The DTLS standard does not explicitly state what the message_seq in the handshake packet should be in that scenario, but 1 is a good guess. Currently WOLFSSL sends 0 if the first packet handled by a WOLFSSL object is a ClientHello with a valid cookie.

ejohnstown, 2022-02-10:
Is it possible to share a test application that illustrates the issue?

mkm85, 2022-02-11:
I do unfortunately not have a suitable test application which shows my issue.

I can try to clarify what the functionality I'm looking for is like:

The openssl equivalent for the functionality is called DTLSv1_listen https://www.openssl.org/docs/man1.1.1/man3/DTLSv1_listen.html

In the DTLS 1.2 RFC it is described in section 4.2.1 https://datatracker.ietf.org/doc/html/rfc6347#section-4.2.1
```
... without retaining any per-client state on the server. ....
```

mkm85, 2022-02-14:
The idea you are proposing is close to what I tried to do. The problem with that idea is that wolfSSL sends the message sequence 0 instead of 1 in the second ServerHello handshake packet. This is probably because it is the first packet sent in that handshake context, but from the clients perspective it is the second packet received from the server.

closed 2025-12-16 by ejohnstown