wolfSSL/wolfssl #4854: [Bug]: wolfSSL_get_error() docs could clarify on SSL_ERROR_NONE

bagder, 2022-02-11:
### Version

I'm browsing the docs online: https://www.wolfssl.com/doxygen/group__Debug.html#gae30b3ae133f07c6b9d2b567367489b02

### Description

The documentation for `wolfSSL_get_error` explains its return code as;

> code On successful completion, this function will return the unique error code describing why the previous API function failed. 
> SSL_ERROR_NONE will be returned if ret > 0.

(I'll ignore the weird-looking first word "code")

It is (clearly) possible for this function to return `SSL_ERROR_NONE` also when `ret == 0` and not only when ret is larger than zero. The docs above does explicitly say that it can't, but the wording mislead me into believing it did and I think a little clarification could reduce the risk others repeating my mistake.

Ref: https://github.com/curl/curl/pull/8431

julek-wolfssl, 2022-02-11:
Doc updated in https://github.com/wolfSSL/wolfssl/pull/4855. Please review if it makes sense now.

julek-wolfssl, 2022-02-22:
Closing as docs have been updated.

closed 2022-02-22 by julek-wolfssl