wolfSSL/wolfssl #4879: [Bug]: SSL_get_verify_result regression with LE certificates introduced in 5.1.1-stable

ynezz, 2022-02-22:
### Contact Details

ynezz@true.cz

### Version

5.1.1-stable

### Description

In `v4.8.1-stable`  `wolfSSL_get_verify_result()` used to return `X509_V_OK` when used against `https://letsencrypt.org` site. With update to `v5.1.1-stable` it now returns unknown error value `39`.

4.8.1 example/client behaviour:

```
Alternate cert chain used
 issuer : /C=US/O=Let's Encrypt/CN=R3
...snip...
Peer verify result = 0
```

5.1.1 example/client behaviour:

```
Alternate cert chain used
 issuer : /C=US/O=Let's Encrypt/CN=R3
...snip...
Peer verify result = 39
```

### Reproduction steps

Working test case with 4.8.1:

```shell
git checkout v4.8.1-stable && \
git clean -xfd && \
 ./autogen.sh && \
C_EXTRA_FLAGS="-DSHOW_CERTS -DWOLFSSL_ALT_CERT_CHAINS" ./configure --enable-sni --enable-opensslextra --enable-altcertchains --enable-opensslall --enable-sni --prefix=/opt/devel/openwrt/testing && \
make && \
./examples/client/client -g -h letsencrypt.org -p 443 -S letsencrypt.org -A /etc/ssl/certs/ISRG_Root_X1.pem
...snip...
Peer verify result = 0
```

Failing test case with 5.1.1:

```
git checkout v5.1.1-stable && \
git clean -xfd && \
 ./autogen.sh && \
C_EXTRA_FLAGS="-DSHOW_CERTS -DWOLFSSL_ALT_CERT_CHAINS" ./configure --enable-sni --enable-opensslextra --enable-altcertchains --enable-opensslall --enable-sni --prefix=/opt/devel/openwrt/testing && \
make && \
./examples/client/client -g -h letsencrypt.org -p 443 -S letsencrypt.org -A /etc/ssl/certs/ISRG_Root_X1.pem
...snip...
Peer verify result = 39
```

### Relevant log output

```shell
Alternate cert chain used
 issuer : /C=US/O=Let's Encrypt/CN=R3
 subject: /CN=lencr.org
 altname = lencr.org
 altname = letsencrypt.com
 altname = letsencrypt.org
 altname = www.lencr.org
 altname = www.letsencrypt.com
 altname = www.letsencrypt.org
 serial number:03:4e:29:5a:d6:74:ae:fd:51:cd:0d:61:11:f9:e3:e3:bd:88 
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            03:4e:29:5a:d6:74:ae:fd:51:cd:0d:61:11:f9:e3:e3:bd:88
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: /C=US/O=Let's Encrypt/CN=R3
        Validity
            Not Before: Feb  7 03:00:32 2022 GMT
            Not After : May  8 03:00:31 2022 GMT
        Subject: /CN=lencr.org
        Subject Public Key Info:
            Public Key Algorithm: EC
                 Public-Key: (256 bit)
                 pub:
                     04:4f:a9:f0:1e:70:ba:16:48:b2:ed:b8:56:b3:94:
                     16:93:cb:37:d0:d8:21:7b:50:63:a7:5f:27:c7:62:
                     14:04:53:3c:44:91:a1:c8:f2:cd:af:62:d7:04:63:
                     72:53:18:65:6a:50:e7:7c:12:c8:10:2c:26:17:6d:
                     2f:43:b7:ee:74
                ASN1 OID: SECP256R1
        X509v3 extensions:
            X509v3 Subject Key Identifier: 
                FD:72:0E:4A:A9:65:33:54:85:66:AE:4E:47:4A:06:D3:00:4E:0F:99
            X509v3 Authority Key Identifier: 
                keyid:14:2E:B3:17:B7:58:56:CB:AE:50:09:40:E6:1F:AF:9D:8B:14:C2:C6

            X509v3 Basic Constraints: 
                CA:FALSE
    Signature Algorithm: sha256WithRSAEncryption
        b9:b0:a7:4a:a7:d4:32:9a:ef:1b:cd:14:27:83:87:b3:bb:bc:
        62:9e:5b:0b:f3:1b:0d:2b:fd:85:9f:61:e8:18:82:14:c7:11:
        97:27:e0:24:ad:f5:82:08:90:23:ee:42:f6:02:a5:29:c6:d9:
        00:60:19:ff:54:38:73:78:e5:83:d0:0f:30:40:ba:37:5b:0f:
        50:4d:14:e8:8c:98:a4:1b:10:1b:e6:59:27:2a:a9:aa:e2:04:
        7c:12:d3:85:8c:b3:1c:a1:28:61:33:af:77:02:b0:2b:f2:c4:
        a7:d3:96:c3:5c:d3:c2:56:79:a1:ed:0d:ce:67:f2:2e:67:99:
        2e:81:2d:41:25:0c:61:20:5a:12:f6:33:c4:21:da:0d:03:5f:
        1a:b1:65:ad:04:70:46:88:01:55:a7:02:47:7a:7e:5f:70:73:
        d3:8b:ef:b3:f9:f3:59:54:48:65:06:38:e4:18:dc:8b:a2:49:
        98:bf:ae:e2:d5:7d:d0:85:0b:48:b8:af:09:18:8e:f7:f5:b1:
        63:ed:94:c2:2c:74:26:55:5e:98:3d:98:96:3d:29:23:28:b6:
        8c:dc:69:e1:a2:aa:d4:42:66:66:10:41:c5:30:c5:f1:12:27:
        05:90:30:47:11:f5:e6:46:6a:b5:e0:93:70:9b:7a:b0:6d:31:
        df:ff:c8:8e
our cert info:
 issuer : /C=US/ST=Montana/L=Bozeman/O=wolfSSL_2048/OU=Programming-2048/CN=www.wolfssl.com/emailAddress=info@wolfssl.com
 subject: /C=US/ST=Montana/L=Bozeman/O=wolfSSL_2048/OU=Programming-2048/CN=www.wolfssl.com/emailAddress=info@wolfssl.com
 altname = example.com
 altname = 127.0.0.1
 serial number:53:16:7c:a0:56:50:46:27:82:ed:60:b4:da:33:d8:6a:c0:ea:dc:31 
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            53:16:7c:a0:56:50:46:27:82:ed:60:b4:da:33:d8:6a:c0:ea:dc:31
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: /C=US/ST=Montana/L=Bozeman/O=wolfSSL_2048/OU=Programming-2048/CN=www.wolfssl.com/emailAddress=info@wolfssl.com
        Validity
            Not Before: Dec 20 23:07:24 2021 GMT
            Not After : Sep 15 23:07:24 2024 GMT
        Subject: /C=US/ST=Montana/L=Bozeman/O=wolfSSL_2048/OU=Programming-2048/CN=www.wolfssl.com/emailAddress=info@wolfssl.com
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                Public-Key: (2048 bit)
                Modulus:
                    00:c3:03:d1:2b:fe:39:a4:32:45:3b:53:c8:84:2b:
                    2a:7c:74:9a:bd:aa:2a:52:07:47:d6:a6:36:b2:07:
                    32:8e:d0:ba:69:7b:c6:c3:44:9e:d4:81:48:fd:2d:
                    68:a2:8b:67:bb:a1:75:c8:36:2c:4a:d2:1b:f7:8b:
                    ba:cf:0d:f9:ef:ec:f1:81:1e:7b:9b:03:47:9a:bf:
                    65:cc:7f:65:24:69:a6:e8:14:89:5b:e4:34:f7:c5:
                    b0:14:93:f5:67:7b:3a:7a:78:e1:01:56:56:91:a6:
                    13:42:8d:d2:3c:40:9c:4c:ef:d1:86:df:37:51:1b:
                    0c:a1:3b:f5:f1:a3:4a:35:e4:e1:ce:96:df:1b:7e:
                    bf:4e:97:d0:10:e8:a8:08:30:81:af:20:0b:43:14:
                    c5:74:67:b4:32:82:6f:8d:86:c2:88:40:99:36:83:
                    ba:1e:40:72:22:17:d7:52:65:24:73:b0:ce:ef:19:
                    cd:ae:ff:78:6c:7b:c0:12:03:d4:4e:72:0d:50:6d:
                    3b:a3:3b:a3:99:5e:9d:c8:d9:0c:85:b3:d9:8a:d9:
                    54:26:db:6d:fa:ac:bb:ff:25:4c:c4:d1:79:f4:71:
                    d3:86:40:18:13:b0:63:b5:72:4e:30:c4:97:84:86:
                    2d:56:2f:d7:15:f7:7f:c0:ae:f5:fc:5b:e5:fb:a1:
                    ba:d3
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Subject Key Identifier: 
                33:D8:45:66:D7:68:87:18:7E:54:0D:70:27:91:C7:26:D7:85:65:C0
            X509v3 Authority Key Identifier: 
                keyid:33:D8:45:66:D7:68:87:18:7E:54:0D:70:27:91:C7:26:D7:85:65:C0

            X509v3 Basic Constraints: 
                CA:TRUE
    Signature Algorithm: sha256WithRSAEncryption
        b8:e8:e3:2a:48:6c:04:8b:f8:81:14:1a:ce:14:ed:c7:f0:d3:
        cb:9a:91:d9:2c:1d:6e:73:36:8f:a3:61:c4:1f:da:d1:4b:b6:
        40:d0:6a:c4:2b:43:c8:2f:fb:ee:5a:c9:41:9d:2b:6f:f3:39:
        67:20:ec:7c:d6:a0:7f:06:79:cd:52:2c:c9:3c:5b:bf:e5:01:
        47:90:f0:82:88:f1:3d:45:25:f4:d1:4b:ec:ac:3f:1b:ce:a1:
        0e:61:a0:29:41:f6:21:0e:9f:73:b3:39:34:c4:1e:55:5f:9f:
        e7:42:ca:ab:8f:3c:62:86:26:94:b5:b7:8b:7c:65:4c:3e:b7:
        ac:f5:51:0d:a5:14:0f:6f:2b:fe:62:95:26:1e:10:52:ae:44:
        58:95:dc:b4:c4:76:2f:14:28:64:45:aa:94:61:da:1a:d0:cf:
        b3:3a:83:c8:66:fb:e8:58:dc:d4:91:4a:9a:e7:c8:b6:ea:f9:
        52:19:b2:3d:5f:95:29:ac:8b:cf:9b:5c:d6:dd:cd:6b:f2:71:
        fd:b6:4d:18:98:08:5b:8a:e7:2b:cb:bd:68:97:1c:02:aa:41:
        59:0d:f8:0e:50:d7:48:6f:81:c4:00:70:56:67:64:1a:b3:56:
        fc:23:f4:84:49:36:f7:7f:38:94:38:da:40:81:c0:b9:b0:ad:
        ea:ce:38:f2
Peer verify result = 39
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
SSL curve name is SECP256R1
Alternate cert chain used
Session timeout set to 500 seconds
Client Random : 87D7802F9A762E39526FA108DAF25E5EC9FCA31D6E11B76C458CEC604132306E
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
    Session-ID: FE067DC34E2BD8D47E67988EB9BE8FED7330DF4F6EF60A225702D193085E690F
    Session-ID-ctx: 
    Master-Key: B6D33BE3B85390C86EEC01331551ABB2FF0C0BB8A103D44D41E6D72EA3FBD7F29BE05BF46B0C9650C1EC9938C98B38AB
    TLS session ticket: NONE
    Start Time: 1645524868
    Timeout   : 500 (sec)
    Extended master secret: yes
SSL connect ok, sending GET...
HTTP/1.0 400 Bad Request
content-length: 19
date: Tue, 22 Feb 2022 10:14:28 GMT
```

ynezz, 2022-02-22:
We've upgraded WolfSSL library being used in OpenWrt from 4.8.1 to 5.1.1 and got following bug report https://github.com/openwrt/openwrt/issues/9283, [affected code](https://gitlab.com/openwrt/project/ustream-ssl/-/blob/master/ustream-openssl.c#L271):

```c
        res = SSL_get_verify_result(ssl);
	if (res != X509_V_OK) {
		if (us->notify_verify_error)
			us->notify_verify_error(us, res, X509_verify_cert_error_string(res));
		return;
	}
```

ynezz, 2022-02-22:
BTW I've already tried to git bisect the issue, but it seems like the Git tree is unusable for that.

ynezz, 2022-02-22:
Seems like commit 1b6b16c2c39796e73262befe89c6abc3d6cf770f or 1acf906612883716e417e25bc098be64c311d448 by @julek-wolfssl touched the codepaths around `peerVerifyRet` and `X509_V_ERR_INVALID_CA`.

dgarske, 2022-02-22:
Hi @ynezz ,

I believe this issue was fixed with https://github.com/wolfSSL/wolfssl/pull/4857

Let us know if you still see an issue after applying that patch.

Thanks,
David Garske, wolfSSL

ynezz, 2022-02-22:
100% tests passed, thanks!

closed 2022-02-22 by ynezz