wolfSSL/wolfssl #4887: [Bug]: wolfssl_Accept craches on Solaris/SPARC in TLS 1.3

t0x1c, 2022-02-24:
### Version

5.2.0

### Description

OS: Solaris 11.3 running on SPARC T4 CPU.
wolfssl_Accept crashes with SIGBUS. This type of exception occurs on SPARC when code tries to read/write unaligned data. Only happens when client connects over TLS 1.3. Works fine with TLS 1.2 client.

I tried to debug it by directly including wolfssl source files into the project. It compiles but when ran fails early on SSL_CTX_new because RNG init routines do not work for some reason - I couldn't debug them that deep (this is my configuration problem, since RNG in static library built with 'configure' works fine).

### Reproduction steps

../wolfssl-5.2.0/configure --enable-static --disable-shared --enable-tlsv12 --enable-tls13 --enable-sni --enable-supportedcurves --enable-extended-master --enable-alpn --enable-secure-renegotiation --enable-session-ticket --disable-enc-then-mac --enable-nullcipher --enable-dsa --enable-ecc --enable-aesccm --enable-aesgcm --enable-aesgcm-stream --enable-camellia --enable-idea --enable-ed448 --enable-curve448 --enable-ed25519 --enable-curve25519 --enable-des3 --enable-arc4 --enable-pkcs7 --enable-pkcs12 --enable-rc2 --enable-sha512 --enable-opensslextra --enable-base64encode CFLAGS="-m64 -DWOLFSSL_SOLARIS -g0 -O3 -fPIC -DSMALL_SESSION_CACHE -DHAVE_NULL_CIPHER -DWOLFSSL_SHA384 -DWOLFSSL_SHA512 -DHAVE_AESCCM -DHAVE_AESGCM -DWOLFSSL_AESGCM_STREAM -DWOLFSSL_AES_XTS -DHAVE_CAMELLIA -DHAVE_CHACHA -DHAVE_POLY1305 -DHAVE_ECC -DWOLFSSL_DH_CONST -DWOLFSSL_DER_LOAD -DWOLFSSL_KEY_GEN -DWC_RC2 -DHAVE_AES_DECRYPT -DNO_WOLFSSL_STUB -DWOLFSSL_STATIC_EPHEMERAL -DWOLFSSL_SHAKE256 -DWOLFSSL_DH_EXTRA -DHAVE_FFDHE_2048 -DWOLFSSL_EVP_INCLUDED -DHAVE_PKCS12 -DWOLFSSL_KEY_GEN"

dgarske, 2022-02-24:
@SparkiDev any ideas?

t0x1c, 2022-02-24:
Update: I could do more debugging but I'm stuck with RNG issues in debug build.

random.c, wc_RNG_HealthTestLocal() fails:

**line 1314:**  ret = wc_RNG_HealthTest(0, seedB, sizeof(seedB_data), ....  // this returns 0 (success)
seedB buffer contents:
```
0x0000000100201978  a6 5a d0 f3 45 db 4e 0e ff e8 75 c3 a2 e7 1f 42  ¦ZРуEЫN.яиuГўз.B
0x0000000100201988  c7 12 9d 62 0f f5 c1 19 a9 ef 55 f0 51 85 e0 fb  З.ќb.хБ.©пUрQ.аы
0x0000000100201998  85 81 f9 31 75 17 27 6e 06 e9 60 7d db cb cc 2e  .Ѓщ1u.'n.й`}ЫЛМ.
```

**line 1318:**  ConstantCompare(check, outputB, ... // this fails

Contents of those 128 byte buffers passed to ConstantCompare:
a:
```
0xFFFFFFFF7DDFA560  21 39 bb da 75 60 56 dd a3 11 2a d7 a2 fd 71 77  !9»Ъu`VЭЈ.*Чўэqw
0xFFFFFFFF7DDFA570  c4 20 b3 4e 5d 95 c6 3d 73 8f 6e 31 8b 3c f6 02  Д іN].Ж=sЏn1.<ц.
0xFFFFFFFF7DDFA580  5e 3d 5c 82 9b dd 28 4d e4 e7 7b bc a4 b7 e3 5c  ^=\..Э(Mдз{ј¤·г\
0xFFFFFFFF7DDFA590  fd b9 3b 1a e7 48 78 c9 ac 97 5a 40 a9 76 b1 13  э№;.зHxЙ¬—Z@©v±.
0xFFFFFFFF7DDFA5A0  13 7c a7 1c ab f0 6b 0b 34 84 a3 1a 11 a3 37 c5  .|§.«рk.4.Ј..Ј7Е
0xFFFFFFFF7DDFA5B0  fd 60 32 7f 85 09 7b 1f 8c e0 de fb cc 2a ca 98  э`2...{.ЊаЮыМ*К.
0xFFFFFFFF7DDFA5C0  17 a6 8e 46 0e e8 58 d8 10 eb 21 4e 4c 7d 66 5e  .¦ЋF.иXШ.л!NL}f^
0xFFFFFFFF7DDFA5D0  82 ee 59 a0 0d 19 c9 92 b6 33 b8 4f b1 16 d3 5d  .оY ..Й’¶3ёO±.У]
```
b:
```
0x00000001002019A8  d3 e1 60 c3 5b 99 f3 40 b2 62 82 64 d1 75 10 60  Уб`Г[™у@Іb.dСu.`
0x00000001002019B8  e0 04 5d a3 83 ff 57 a5 7d 73 a6 73 d2 b8 d8 0d  а.]ЈѓяWҐ}s¦sТёШ.
0x00000001002019C8  aa f6 a6 c3 5a 91 bb 45 79 d7 3f d0 c8 fe d1 11  Єц¦ГZ‘»EyЧ?РИюС.
0x00000001002019D8  b0 39 13 06 82 8a df ed 52 8f 01 81 21 b3 fe bd  °9...ЉЯнRЏ.Ѓ!іюЅ
0x00000001002019E8  c3 43 e7 97 b8 7d bb 63 db 13 33 de d9 d1 ec e1  ГCз—ё}»cЫ.3ЮЩСмб
0x00000001002019F8  77 cf a6 b7 1f e8 ab 1d a4 66 24 ed 64 15 e5 1c  wП¦·.и«.¤f$нd.е.
0x0000000100201A08  cd e2 c7 ca 86 e2 83 99 0e ea eb 91 12 04 15 52  НвЗК.вѓ™.кл‘...R
0x0000000100201A18  8b 22 95 91 02 81 b0 2d d4 31 f4 c9 f7 04 27 df  .".‘.Ѓ°-Ф1фЙч.'Я
```

t0x1c, 2022-02-24:
Ok, found RNG problem - looks like machine endianness is configured during config time, so if I include source code directly, endianness is not detected properly in compile time, so it falls back to little endian, which is wrong for SPARC.

dgarske, 2022-02-24:
Hi @t0x1c ,

That is great you discovered it was an endianness issue. Do you have a solution to it?

The endianness build option `BIG_ENDIAN_ORDER` or `WORDS_BIGENDIAN` should get set in the `wolfssl/options.h`, which your application should include before any other wolfSSL headers.

Thanks,
David Garske, wolfSSL

t0x1c, 2022-02-24:
Yeah, I've discovered this variable and applied it. With endianness fixed, SHA and RNG work fine and context gets created, however it fails to establish SSL connection, having various socket-related issues. I need to think on other methods of creating a debug build.

Here's how I detect endianness in my code. Not sure where and how to use it in your case, since I'm not familiar with wolfssl architecture.

```
#if defined(__x86_64) || defined(__amd64) || defined(__x86_64__) || defined(__amd64__) || defined(_M_X64) || defined(_M_AMD64)
	#define XX_ARCH_AMD64
	#define XX_BYTE_ORDER XX_LITTLE_ENDIAN
#elif defined(__i386) || defined(__i386__) || defined(_M_IX86) || defined(__IA32__)
	#define XX_ARCH_X86
	#define XX_BYTE_ORDER XX_LITTLE_ENDIAN
#elif defined(__powerpc64__) || defined(__ppc64__) || defined (__PPC64__) || defined(_ARCH_PPC64) || defined(_M_PPC)
	#if (defined(_LITTLE_ENDIAN) && (_LITTLE_ENDIAN == 1)) || (defined(__LITTLE_ENDIAN__) && (__LITTLE_ENDIAN__ == 1)) || (__BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__)
		#define XX_ARCH_PPC64LE
		#define XX_BYTE_ORDER XX_LITTLE_ENDIAN
	#else
		#define XX_ARCH_PPC64
		#define XX_BYTE_ORDER XX_BIG_ENDIAN
	#endif
#elif defined(__sparc__) || defined(__sparc) || defined (__sparc_v9__) || defined(__sparcv9)
	#define XX_ARCH_SPARC64
	#define XX_BYTE_ORDER XX_BIG_ENDIAN
#elif defined(__ia64__) || defined(__IA64__) || defined (_IA64) || defined(__ia64) || defined(_M_IA64) || defined(__itanium__)
	#define XX_ARCH_IA64
	#define XX_BYTE_ORDER XX_BIG_ENDIAN
#else
	#error Unsupported architecture
#endif
```

t0x1c, 2022-03-01:
Ok, I made a debug build, but it works fine in debug mode. So this error only occurs in release which I can't debug. So this is something related to optimizations.

t0x1c, 2022-03-01:
Here's full info on the compiler used:
```
~:root> gcc -dumpspecs
*asm:
--traditional-format %{v:-V} %{Qy:} %{!Qn:-Qy} %{Ym,*} -s %(asm_cpu) %(asm_arch) %{fpie|fPIE|fpic|fPIC:-K PIC}

*asm_debug:
%{%:debug-level-gt(0):%{gstabs*:--gstabs}%{!gstabs*:%{g*:--gdwarf2}}} %{fdebug-prefix-map=*:--debug-prefix-map %*}

*asm_final:
%{gsplit-dwarf:
       objcopy --extract-dwo     %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O}      %{c:%{o*:%:replace-extension(%{o*:%*} .dwo)}%{!o*:%b.dwo}}%{!c:%b.dwo}
       objcopy --strip-dwo       %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O}     }

*asm_options:
%{-target-help:%:print-asm-header()} %{v} %{w:-W} %{I*}  %{gz|gz=zlib:--compress-debug-sections=zlib} %{gz=none:--compress-debug-sections=none} %{gz=zlib-gnu:--compress-debug-sections=zlib-gnu} %a %Y %{c:%W{o*}%{!o*:-o %w%b%O}}%{!c:-o %d%w%u%O}

*invoke_as:
%{!fwpa*:   %{fcompare-debug=*|fdump-final-insns=*:%:compare-debug-dump-opt()}   %{!S:-o %|.s |
 as %(asm_options) %m.s %A }  }

*cpp:
%(cpp_cpu) %(cpp_arch) %(cpp_endian) %(cpp_other) %(cpp_subtarget)

*cpp_options:
%(cpp_unique_options) %1 %{m*} %{std*&ansi&trigraphs} %{W*&pedantic*} %{w} %{f*} %{g*:%{%:debug-level-gt(0):%{g*} %{!fno-working-directory:-fworking-directory}}} %{O*} %{undef} %{save-temps*:-fpch-preprocess}

*cpp_debug_options:
%{d*}

*cpp_unique_options:
%{!Q:-quiet} %{nostdinc*} %{C} %{CC} %{v} %@{I*&F*} %{P} %I %{MD:-MD %{!o:%b.d}%{o*:%.d%*}} %{MMD:-MMD %{!o:%b.d}%{o*:%.d%*}} %{M} %{MM} %{MF*} %{MG} %{MP} %{MQ*} %{MT*} %{!E:%{!M:%{!MM:%{!MT:%{!MQ:%{MD|MMD:%{o*:-MQ %*}}}}}}} %{remap} %{g3|ggdb3|gstabs3|gxcoff3|gvms3:-dD} %{!iplugindir*:%{fplugin*:%:find-plugindir()}} %{H} %C %{D*&U*&A*} %{i*} %Z %i %{E|M|MM:%W{o*}}

*trad_capable_cpp:
cc1 -E %{traditional|traditional-cpp:-traditional-cpp}

*cc1:
%{m32:%{m64:%emay not use both -m32 and -m64}} %{m32:-mptr32 -mno-stack-bias   %{!mcpu*:%{!mv8plus:-mcpu=v9}}} %{mv8plus:-m32 -mptr32 -mno-stack-bias   %{!mcpu*:-mcpu=v9}} %{%:sanitize(address):-fasynchronous-unwind-tables}

*cc1_options:
%{pg:%{fomit-frame-pointer:%e-pg and -fomit-frame-pointer are incompatible}} %{!iplugindir*:%{fplugin*:%:find-plugindir()}} %1 %{!Q:-quiet} %{!dumpbase:-dumpbase %B} %{d*} %{m*} %{aux-info*} %{fcompare-debug-second:%:compare-debug-auxbase-opt(%b)}  %{!fcompare-debug-second:%{c|S:%{o*:-auxbase-strip %*}%{!o*:-auxbase %b}}}%{!c:%{!S:-auxbase %b}}  %{g*} %{O*} %{W*&pedantic*} %{w} %{std*&ansi&trigraphs} %{v:-version} %{pg:-p} %{p} %{f*} %{undef} %{Qn:-fno-ident} %{Qy:} %{-help:--help} %{-target-help:--target-help} %{-version:--version} %{-help=*:--help=%*} %{!fsyntax-only:%{S:%W{o*}%{!o*:-o %b.s}}} %{fsyntax-only:-o %j} %{-param*} %{coverage:-fprofile-arcs -ftest-coverage} %{fprofile-arcs|fprofile-generate*|coverage:   %{!fprofile-update=single:     %{pthread:-fprofile-update=prefer-atomic}}}

*cc1plus:

*link_gcc_c_sequence:
%G %{!nolibc:%L} %G %{!nolibc:%L}

*link_ssp:
%{fstack-protector|fstack-protector-all|fstack-protector-strong|fstack-protector-explicit:}

*endfile:
%{Ofast|ffast-math|funsafe-math-optimizations:crtfastmath.o%s}    %(endfile_arch) %(endfile_vtv) %(endfile_crtend) crtn.o%s

*link:
%{h*} %{v:-V}    %{!shared:%{!static:%{rdynamic: }}}    %{static:-dn -Bstatic}    %{shared:-G -dy %{!mimpure-text:-z text}}  %{mclear-hwcap|fopenmp*:-M %sclearcap.map}    %{symbolic:-Bsymbolic -G -dy -z text}    %(link_arch)    %{Qy:} %{!Qn:-Qy}

*lib:
%{!symbolic:     %{pthreads|pthread:-lpthread}      %{p|pg:-ldl} -lc}

*link_gomp:

*libgcc:
-lgcc

*startfile:
%{!shared:%{!symbolic:                    %{p:mcrt1.o%s;                             pg:gcrt1.o%s gmon.o%s;                               :crt1.o%s}}}                  crti.o%s %(startfile_arch) %(startfile_crtbegin)                        %(startfile_vtv)

*cross_compile:
0

[... 103 lines cut ...]
*startfile_vtv:

*link_arch32:
%{G:-G}    %{YP,*}    %{R*}    %{!YP,*:%{p|pg:-Y P,%R/usr/lib/libp%R/lib:%R/usr/lib}       %{!p:%{!pg:-Y P,%R/lib:%R/usr/lib}}}

*link_arch64:
%{mcmodel=medlow:-M /usr/lib/ld/sparcv9/map.below4G} %{G:-G}    %{YP,*}    %{R*}    %{!YP,*:%{p|pg:-Y P,%R/usr/lib/libp/sparcv9:%R/lib/sparcv9:%R/usr/lib/sparcv9}                 %{!p:%{!pg:-Y P,%R/lib/sparcv9:%R/usr/lib/sparcv9}}}

*link_arch_default:
%{mcmodel=medlow:-M /usr/lib/ld/sparcv9/map.below4G} %{G:-G}    %{YP,*}    %{R*}    %{!YP,*:%{p|pg:-Y P,%R/usr/lib/libp/sparcv9:%R/lib/sparcv9:%R/usr/lib/sparcv9}                 %{!p:%{!pg:-Y P,%R/lib/sparcv9:%R/usr/lib/sparcv9}}}

*link_arch:
 %{m32:%(link_arch32)} %{m64:%(link_arch64)} %{!m32:%{!m64:%(link_arch_default)}}

*endfile_arch:

*endfile_crtend:
crtend.o%s

*endfile_vtv:

*link_command:
%{!fsyntax-only:%{!c:%{!M:%{!MM:%{!E:%{!S:    %(linker) %{fuse-linker-plugin:    %e-fuse-linker-plugin is not supported in this configuration}%{flto|flto=*:%<fcompare-debug*}     %{flto} %{fno-lto} %{flto=*} %l %{no-pie:} %{pie:%e-pie is not supported in this configuration} %{fuse-ld=*:-fuse-ld=%*}  %{gz|gz=zlib:-z compress-sections=zlib} %{gz=none:-z compress-sections=none} %{gz=zlib-gnu:-z compress-sections=zlib-gnu} %X %{o*} %{e*} %{N} %{n} %{r}    %{s} %{t} %{u*} %{z} %{Z} %{!nostdlib:%{!r:%{!nostartfiles:%S}}}     %{static|no-pie|static-pie:} %@{L*} %(mfwrap) %(link_libgcc) %{fvtable-verify=none:} %{fvtable-verify=std:   %e-fvtable-verify=std is not supported in this configuration} %{fvtable-verify=preinit:   %e-fvtable-verify=preinit is not supported in this configuration} %{!nostdlib:%{!r:%{!nodefaultlibs:%{%:sanitize(address):%{!m32:%e-fsanitize=address is not supported i [... cut ...] anitizer.spec)%(link_libtsan)}    %{static:%ecannot specify -static with -fsanitize=thread}}    %{%:sanitize(undefined):%{static-libubsan:-Bstatic} -lubsan %{static-libubsan:-Bdynamic} %{static-libubsan|static:%:include(libsanitizer.spec)%(link_libubsan)}}    %{%:sanitize(leak): %{static-liblsan|static:%:include(libsanitizer.spec)%(link_liblsan)}}}}}     %{!nostdlib:%{!r:%{!nodefaultlibs:%(link_ssp) %(link_gcc_c_sequence)}}}    %{!nostdlib:%{!r:%{!nostartfiles:%E}}} %{T*}
%(post_link) }}}}}}
```

t0x1c, 2022-03-01:
@dgarske , @SparkiDev 
I've found one of the places, but there may be others. It's just as I thought - unaligned access. 
compare_InternalTickets in CreateTicket() casts byte array to InternalTicket structure. This array is on 2-byte boundary (2 bytes alignment), so accessing any variable that is larger than 2 bytes will cause a SIGBUS crash. Screenshot attached.

[image]

t0x1c, 2022-03-01:
To progress past this problem, I made a quick workaround by wrapping InternalTicket struct definition in **#pragma pack(push, 1)** / **#pragma pack(pop)**, so the compiler could generate guard code for accessing members. It works and I didn't notice other issues in my quick test, but that's not a guarantee this is the only place with unaligned access.

Though this workaround adds performance penalty, so the better approach would be to add padding to arrays in ExternalTicket struct or reorder arrays or struct members to make physical layout naturally aligned.

dgarske, 2022-03-01:
Hi @t0x1c ,

Excellent find with the alignment. Would it make sense to have 4 byte alignment set on the structure like `#pragma pack(push, 4)`? I'll work on putting a PR up to fix this. I may try and leverage our `WOLFSSL_GENERAL_ALIGNMENT` macro.

Thanks, David Garske, wolfSSL

t0x1c, 2022-03-01:
This struct has packing of 4 already because it contains word32 members. Marking it as pack-1 tells compiler it can by unaligned. Anyway, I used this approach for testing only since it introduces a performance penalty on any operation involving such struct.
Here's more info on performance (scroll down to disassembly examples)
https://devblogs.microsoft.com/oldnewthing/20200103-00/?p=103290

dgarske, 2022-03-01:
Hi @t0x1c ,

I've put up a fix here:
https://github.com/wolfSSL/wolfssl/pull/4908

Let me know if that works for you. If not let me know if you see padding issue on the internal ticket itself and I can investigate further.

Thanks,
David Garske, wolfSSL

t0x1c, 2022-03-02:
David,
I tried to compile wolfssl from your tick_pad - problem seems to be resolved.
Though I was not able install it from the source code package obtained on github (some files seem to be missing compared to the release package offered on wolfssl.com). So I just took 5.2.0 release package and replaced "src", "wolfssl", and "wolfcrypt" folders with the ones in your branch. make(compile) worked well, but make install didn't, so I installed it manually.

dgarske, 2022-03-02:
Hi @t0x1c ,

Glad to hear PR 4908 fixed the issue! I'll assign it for merge.

If you clone from GitHub you need to run `./autogen.sh` first to build the configure files. I suspect that is the reason for the difference.

Thanks,
David Garske, wolfSSL

t0x1c, 2022-03-02:
Oh, my bad. Thanks for info!

closed 2022-12-29 by ejohnstown