wolfSSL/wolfssl #4910: Stack Smashing Detected or SegFault on wc_InitRsaKey

petertorelli, 2022-03-01:
### Version

5.1.1

### Description

I'm seeing failures on Ubuntu 20.04 using gcc 9.3.0 for a trivial example:

Source code:
```C
#include <wolfssl/wolfcrypt/rsa.h>

int
main(int argc, char *argv)
{
    RsaKey pri;
    int ret;

    ret = wc_InitRsaKey(&pri, NULL);
    if (ret) {
        printf("Failed to init private key: %d\n", ret);
        return -1;
    }
    wc_FreeRsaKey(&pri);
    return 0;
}
```

Configure command, no options at all:

```Bash
% ./autogen.sh
% ./configure
% make
% sudo make install
```
All tests pass on `make test`.

Error messages:

```bash
$ gcc -g wolftest.c -lwolfssl
In file included from /usr/local/include/wolfssl/wolfcrypt/types.h:34,
                 from /usr/local/include/wolfssl/wolfcrypt/rsa.h:36,
                 from wolftest.c:1:
/usr/local/include/wolfssl/wolfcrypt/settings.h:2356:14: warning: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Wcpp]
 2356 |             #warning "For timing resistance / side-channel attack prevention consider using harden options"
      |              ^~~~~~~
$ ./a.out 
Segmentation fault (core dumped)
$ ./a.out 
*** stack smashing detected ***: terminated
Aborted (core dumped)
$ ./a.out 
*** stack smashing detected ***: terminated
Aborted (core dumped)
$ ./a.out 
*** stack smashing detected ***: terminated
Aborted (core dumped)
$ ./a.out 
Segmentation fault (core dumped)

```

GDB log:

```gdb
(gdb) run
Starting program: /home/ptorelli/a.out 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
__memset_avx2_erms () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:151
151     ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S: No such file or directory.
(gdb) bt
#0  __memset_avx2_erms () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:151
#1  0x00007ffff7f5565e in wc_InitRsaKey_ex () from /usr/local/lib/libwolfssl.so.30
#2  0x00005555555551e8 in main (argc=1, argv=0x7fffffffe288 "") at wolftest.c:9
(gdb) 
```

dgarske, 2022-03-01:
Hi @petertorelli ,

Can you share your wolftest.c? I suspect you are not including `wolfssl/options.h` and the size of the RsaKey struct does not match what wolfSSL expects and your application is defining.

Thanks,
David Garske, wolfSSL

petertorelli, 2022-03-01:
Hi, the code I listed is wolftest.c. Let me add options.h.

kaleb-himes, 2022-03-01:
Peter,

Confirmed, if you include options.h it should resolve the undefined behavior you're observing. Here is a proposed patch to your app:

```
--- original.c	2022-03-01 14:59:17.000000000 -0700
+++ sanity.c	2022-03-01 14:58:59.000000000 -0700
@@ -1,7 +1,9 @@
+#include <stdio.h>
+#include <wolfssl/options.h>
 #include <wolfssl/wolfcrypt/rsa.h>
 
 int
-main(int argc, char *argv)
+main(int argc, char **argv)
 {
     RsaKey pri;
     int ret;
```

petertorelli, 2022-03-01:
Resolved. Thank you. Is `options.h` generally required for all WolfSSL-based applications or just RSA?

closed 2022-03-01 by petertorelli

kaleb-himes, 2022-03-01:
@petertorelli it is for all (with a slight caveat)

Either <wolfssl/options.h> (when using ./configure to build) or <wolfssl/wolfcrypt/settings.h> (for all other cases)

You may include one or both but they should ALWAYS be the first header before any other wolfssl header in any application to avoid undefined behavior. 

Cheers,

- KH

petertorelli, 2022-03-01:
Hi @kaleb-himes,

So if I'm building Wolfssl I need to use /options.h, but if I downloaded it or some other scenario I use /wolfcrypt/settings.h?

That's a little confusing, can you direct me to the manual so I can read up on that? It leads me to believe that my code wouldn't be portable unless the recipient was aware of how I had built/installed wolfssl?

Thanks,
Peter

kaleb-himes, 2022-03-01:
Peter,

Gladly! Here is a blog post on it: https://www.wolfssl.com/how-do-i-manage-the-build-configuration-of-wolfssl/
(For any other build stuff please see [Chapter 2 of the wolfSSL Manual](https://www.wolfssl.com/docs/wolfssl-manual/ch2/)

In general if you are building on unix/linux for unix/linux or cross-copiling for an embedded target using a toolchain and autoconf/libtool with `./configure` you are good to use options.h

If you are building with an IDE (Windows MSVS or CCS or Android Studio etc) that is when you are probably not going to have an options.h being auto-generated and should depend on <wolfssl/wolfcrypt/settings.h> instead. In this case one can globally define WOLFSSL_USER_SETTINGS at the project level and create a custom "user_settings.h" header that you can send to your recipients to ensure you both use the same build settings. Whenever WOLFSSL_USER_SETTINGS is defined globally <wolfssl/wolfcrypt/settings.h> will search the include path for a file named "user_settings.h".

- KH