wolfSSL/wolfssl #4935: [Bug]:

ilikdoge, 2022-03-08:
### Version

master

### Description

./configure --enable-fast-rsa --enable-intelasm --enable-sp --enable-sp-math --enable-sp-asm --enable-sp-math-all --enable-fastmath --enable-tls13 --enable-sni --enable-aesni --enable-alpn --enable-aesgcm-stream --prefix=/usr

linux 5.13 x86_64

wc_AesGcmEncryptUpdate segmentation fault executing aligned move on unaligned stack pointer
[image]

culprit (line 14314) rsp is not aligned to a 16 byte boundary after the instruction
```asm
subq	$0x98, %rsp```

### Reproduction steps

run wc_AesGcmEncryptUpdate with plaintext length greater than 128 bytes

kareem-wolfssl, 2022-03-08:
Thanks for the report.  Working on reproducing it here.

kareem-wolfssl, 2022-03-08:
Reproduced.

Instructions:

```
CFLAGS='-DBENCHMARK_AESGCM_STREAM' ./configure --enable-fast-rsa --enable-intelasm --enable-sp --enable-sp-math --enable-sp-asm --enable-sp-math-all --enable-fastmath --enable-tls13 --enable-sni --enable-aesni --enable-alpn --enable-aesgcm-stream --prefix=/usr
make clean && make
./wolfcrypt/benchmark/benchmark -aes-gcm
```

```
Program received signal SIGSEGV, Segmentation fault.
AES_GCM_encrypt_update_avx2 () at wolfcrypt/src/aes_gcm_asm.S:14331
14331           vmovdqa %xmm4, 128(%rsp)
(gdb) bt
#0  AES_GCM_encrypt_update_avx2 () at wolfcrypt/src/aes_gcm_asm.S:14331
#1  0x7f730444a928ce71 in ?? ()
#2  0x38d93649e356e643 in ?? ()
#3  0x1c553075e3477b2d in ?? ()
#4  0xcb4f6ac6cb4f6ac6 in ?? ()
#5  0x823f97e2823f97e2 in ?? ()
#6  0x000000000040bd00 in ?? ()
#7  0x00007ffff7fc42a0 in ?? ()
#8  0x00007ffff7e16c43 in ASSISTS () at wolfcrypt/src/aes_asm.S:1142
#9  0x00007ffff7de04a6 in AES_set_encrypt_key (userKey=0x7fffffffdca0 "\001#Eg\211\253\315\357\376\336\272\230vT2\020 ", bits=32767, aes=0x7ffff790b010) at wolfcrypt/src/aes.c:789
#10 0x00007ffff7de9add in wc_AesGcmEncryptUpdate (aes=0x7fffffffdca0, out=0x7ffff780a010 "", in=0x7ffff790b010 "", sz=1048576, authIn=0x7fffffffd8db "", authInSz=13) at wolfcrypt/src/aes.c:9309
#11 0x000000000040562b in bench_aesgcm_internal (doAsync=0,
    key=0x410070 <bench_key_buf> "\001#Eg\211\253\315\357\376\336\272\230vT2\020\211\253\315\357\001#Eg\001#Eg\211\253\315\357\360\361\362\363\364\365\366\367\370\371\372\373\374\375\376\377", keySz=16,
    iv=0x4100b0 <bench_iv_buf> "\022\064Vx\220\253\315\357\001\001\001\001\001\001\001\001\021!1AQaq\201MB", ivSz=12, encLabel=0x410478 "AES-128-GCM-enc", decLabel=0x410488 "AES-128-GCM-dec")
    at wolfcrypt/benchmark/benchmark.c:2825
#12 0x0000000000405a7c in bench_aesgcm (doAsync=0) at wolfcrypt/benchmark/benchmark.c:2925
#13 0x00000000004046a3 in benchmarks_do (args=0x0) at wolfcrypt/benchmark/benchmark.c:1810
#14 0x0000000000404d4c in benchmark_test (args=0x0) at wolfcrypt/benchmark/benchmark.c:2581
#15 0x000000000040bcee in main (argc=1, argv=0x7fffffffe250) at wolfcrypt/benchmark/benchmark.c:7699
(gdb) info registers
rax            0x7fffffffddd0      140737488346576
rbx            0x40bd00            4242688
rcx            0x7ffff790b010      140737346842640
rdx            0x7ffff780a010      140737345789968
rsi            0xa                 10
rdi            0x7fffffffdca0      140737488346272
rbp            0x7fffffffd830      0x7fffffffd830
rsp            0x7fffffffd728      0x7fffffffd728
r8             0x100000            1048576
r9             0x7fffffffe010      140737488347152
r10            0x7ffff780a010      140737345789968
r11            0x7ffff790b010      140737346842640
r12            0x7fffffffe000      140737488347136
r13            0x100000            1048576
r14            0x0                 0
r15            0x0                 0
rip            0x7ffff7e2791c      0x7ffff7e2791c <AES_GCM_encrypt_update_avx2+108>
eflags         0x10206             [ PF IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0
k0             0x0                 0
k1             0x0                 0
k2             0x0                 0
k3             0x0                 0
k4             0x0                 0
k5             0x0                 0
k6             0x0                 0
k7             0x0                 0
```
Environment: Fedora Linux on i7-1165G7 (WSL2)

@SparkiDev please take a look.

SparkiDev, 2022-03-08:
Hi @ilikdoge 

I've put up a pull request that fixes this issue: https://github.com/wolfSSL/wolfssl/pull/4939
Let me know how it goes for you!

Thanks,
Sean :-)

ilikdoge, 2022-03-09:
The changes in the pull request work, thanks for the quick response. I'm still rather new to github and git so if there's something I need to do for this issue, please let me know.

I've taken a look at your changes which was also what I did to temporarily fix my problem on my end, however I did not encounter any issues with aligned vector moves with other registers, so it may be possible to align the stack pointer and gain some speed assuming the other inputs are always aligned.

The following code runs about 2x faster without the subtraction on line 8
(I haven't touched assembly or vector instructions in a while so I might be doing something wrong here)
[image]
[image]
[image]
CPU: AMD 5800X
Env: Pop OS 21.04 Linux 5.13

ilikdoge, 2022-03-09:
In my testing it seems that aligned pointer only gives <1% speed boost unfortunately

SparkiDev, 2022-03-09:
Hi @ilikdoge 
From looking at the instruction timing document I have, Risen processors have the same performance characteristics for VMOVDQA and VMOVDQU.
The Piledriver CPU, though, had a slower VMOVEDQU when loading (if I read the table right).
In your last message did you try with and without aligned moves and saw little difference?

ilikdoge, 2022-03-09:
Yes, the instruction doesn't matter as much as whether the pointer is actually aligned. When I tested AES GCM with aligned pointers it did not give a measurable performance increase, so it's not worth it to keep the aligned instructions.

ilikdoge, 2022-03-09:
I am going to close this issue as it is resolved. Hopefully the pull request be merged soon. Thanks!

closed 2022-03-09 by ilikdoge

SparkiDev, 2022-03-09:
Great, thanks for the confirmation.
I'll let you know when the PR has been merged!

SparkiDev, 2022-03-09:
Thanks for the confirmation.

PR #4939 (2022-03-15): AES-GCM: stack alignment issues
# Description

AES-GCM: stack alignment issues
Don't expect stack to be aligned.
vmovdqu is no longer slower than vmovdqa.

Fixes #4935

# Testing

Tested on Linux x86_64 with AVX2.
No performance difference.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation