wolfSSL/wolfssl #4962: Question: Why doesn't `wc_RsaSSL_Sign` perform PKCS1_v15 encoding?

petertorelli, 2022-03-16:
### Version

5.1.1

### Description

This is a general design question, not a bug.

I'm comparing various crypto implementations and observed that in wolfCrypt's `wc_RsaSSL_Sign` function, it does not perform the DER encoding per the PKCS1_v15 standard, but it does however offer PKCS1_v15 padding option. The user must call `wc_EncodeSignature` before signing. It is a little confusing of RsaSSL is supposed to be PKCS1_v15 compliant or not. Compare to something like mbedTLS where the function `mbedtls_rsa_rsassa_pkcs1_v15_sign` performs the encoding.

Is this because `wc_RsaSSL_Sign` isn't intended to be PKCS1_v15 compliant out-of-the-box, or some other reason. This is more of a research question to understand the implementation strategies better.

Thanks,
Peter

SparkiDev, 2022-03-17:
Hi @petertorelli
The wc_RsaSSL_Sign function is for the SSL/TLS handshake.
Performing the DER encoding is not required for signature in early SSL/TLS.
We do have a generic wc_Signature API that will do the encoding.

petertorelli, 2022-03-17:
Thanks, was not aware of wc_Signature.

closed 2022-03-17 by petertorelli