wolfSSL/wolfssl #4970: [Bug]: Selection between status_request and status_request_v2 inconsistent

jensguballa, 2022-03-20:
### Version

5.2.0

### Description

./configure --enable-ocspstapling --enable-ocspstapling2
Target platform: Linux kirk 5.13.0-35-generic #40-Ubuntu SMP Mon Mar 7 08:03:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Starting the server: 
./examples/server/server -C 5000 -x -p 44330 -d 

### Testcase1:
Sending a ClientHello (TLSv1.2) with the extensions in this sequence:
  status_request, status_request_v2

Server behavior: ServerHello contains extension status_request, but no status_request_v2 .

### Testcase2:
Sending a ClientHello (TLSv1.2) with the extensions in this sequence:
  status_request_v2, status_request

Server behavior: ServerHello contains extension status_request_v2 but no status_request.

So it looks like the selection between these two extensions is based on the sequence (first extension found gets processed), but I would expect in both cases to prioritize status_request_v2.

### Reproduction steps

If desired, I can share the script I have used to setup and reproduce the tests.

julek-wolfssl, 2022-03-22:
Hi @jensguballa,
thank you for your report. Please share your script for reproducing this issue. This will aid in solving this bug quickly.
Thanks,
Juliusz

jensguballa, 2022-03-22:
I have used the tool tlsmate (https://gitlab.com/guballa/tlsmate) for the test. 

Installation (I recommend using virtual environments):
```
pip install tlsmate
```
Starting the TLS server (from the `examples/server` directory of the wolfssl build):
```
./server -C 5000 -x -p 44330 -d
```

This is the test script:[tlsmate_wolfssl_ocsp.py.zip](https://github.com/wolfSSL/wolfssl/files/8327122/tlsmate_wolfssl_ocsp.py.zip)

Unzip the script and execute from the directory where it resides (bash shell assumed):
```
PYTHONPATH=$PYTHONPATH:. tlsmate --plugin=tlsmate_wolfssl_ocsp wolfssl localhost --port=44330
```
The script executes both testcases and prints the extensions received in the ServerHello. Let me know if you need more information.

julek-wolfssl, 2022-03-23:
@jensguballa Thank you for providing an easy way to reproduce this bug. I never heard of tlsmate but from what I'm seeing I may use it in the future. The test case is very clean and make reproducing easy. I have posted a fix here: https://github.com/wolfSSL/wolfssl/pull/4978.

julek-wolfssl, 2022-03-25:
@jensguballa https://github.com/wolfSSL/wolfssl/pull/4978 has been merged to wolfSSL master. Thanks again.

closed 2022-03-25 by julek-wolfssl

jensguballa, 2022-03-25:
Fix works. Great!

In case you have any comments or questions regarding tlsmate let me know, it is developed by me.