wolfSSL/wolfssl #5038: fp_montgomery_reduce_ex and fp_montgomery_setup not working on a 32-bit RISC-V IBEX based SOC

sarahSherifAli, 2022-04-09:
### Version

5.1.1

### Description

I am trying to run fp_montgomery_reduce_ex and fp_montgomery_setup on Chameleon SOC using RISC-V toolchain but it seems that there is a problem in the definitions as when I defined:
typedef unsigned int       fp_digit;
#define SIZEOF_FP_DIGIT 4
typedef ulong64            fp_word;

 fp_montgomery_reduce_ex works fine but fp_montgomery_setup doesn't work however according to the comments and documentation this is the correct definition

SparkiDev, 2022-04-11:
Hi @sarahSherifAli 

I can't see anything in fp_montfomery_setup that would not work with SIZEOF_DP_DIGIT at 4.
The code will calculate the inverse for the bottom word of the prime.
The code is written to support fp_digit being 32-bit and 64-bit.

Sean

dgarske, 2022-04-18:
Hi @sarahSherifAli,

Have you tried building with the latest release or even latest master? What are the build errors you are seeing?

We have tested fast math on the RISC-V (HiFive Unleashed and HiFive1 Rev B) boards. See https://github.com/wolfSSL/wolfssl/tree/master/IDE/RISCV for details.

You might also try with `--enable-sp`, which should use the sp_c32.c math. This is our own math library (vs fast math tfm.c based on Tom's fast math).

Thanks,
David Garske, wolfSSL

SparkiDev, 2022-04-25:
Hi @sarahSherifAli

Could you please give us an example of when fp_montgomery_setup() doesn't work?
The code is designed to work with 32-bit and 64-bit when FP_64BIT is defined.

Thanks,
Sean Parkinson, wolfSSL

SparkiDev, 2022-05-30:
Hi @sarahSherifAli 

Some time has passed since your initial comment.
Please let us know if this issue is still active.

Thanks,
Sean Parkinson, wolfSSL

SparkiDev, 2022-06-13:
Hi @sarahSherifAli 

I'm afraid I have to close this issue due to inactivity.
If you have further questions please re-open or raise a new issue.

Sean