wolfSSL/wolfssl #5043: [Bug]: STM32L4R5ZI requires HAL_RNG_DeInit in wc_GenerateSeed or it fails on second call.

petertorelli, 2022-04-11:
### Contact Details

peter.j.torelli@gmail.com

### Version

5.2.0

### Description

I'm targeting a STM32L4R5ZI on a Nucleo-144 platform. The only deviations from the `wolfSSL.I-CUBE-wolfSSL_conf.h` are modifying the fall-through platform test by adding `#define WOLFSSL_STM32L4` and changing the `HAL_CONSOLE_UART`.

By default, the project initializes the RNG during the `main.c:MX_RNG_Init()` function. However `random.c:wc_GenerateSeed()` for CubeMX initializes it again. Then, when I try to call `wc_InitRng` it fails with error -199 (`RNG_FAILURE_E`). When I step through the code, I see that it is failing in the HAL with a seed error (the SEIS is set). This code (in `stm32l4xx_hal_rng.c`) passes with the first call to `MX_RNG_Init`, but then fails when called via `wc_GenerateSeed`:

```C
  /* verify that no seed error */
  if (__HAL_RNG_GET_IT(hrng, RNG_IT_SEI) != RESET)
  {
    hrng->State = HAL_RNG_STATE_ERROR;
    return HAL_ERROR;
  }
```

I can fix this problem by making two changes. First, I call `HAL_RNG_DeInit` in `main.c` on the default handle created by CubeMX. Second, I modify `random.c:wc_GenerateSeed` and add `HAL_RNG_DeInit` at line 2045, just before the `wolfSSL_CryptHwMutexUnLock`.

Between the CubeMX files, the wolfSSL config files, and the numerous C files, I'm not sure how to include even a small project. How else can I help?

petertorelli, 2022-04-11:
I made a git repo with everything minus the Wolf `Middlewares` folder.

https://github.com/petertorelli/wolf-issue-5043

See function test_rng() which simply does a wc_InitRng like wc_GenerateSeed and fails with -199.

kareem-wolfssl, 2022-04-14:
Hi Peter,

Thanks for the report!  Indeed I see this issue, we are initializing the RNG in our GenerateSeed function but not DeIniting it.  I am working on some STM32 fixes and will include a fix for this issue in it.

Thanks,
Kareem

PR #5307 (2022-07-01): Fix a couple of STM32 bugs, and add some missing mutex frees.
# Description

Fixes #5043, zd# 14302, 14066

# Testing

Build + test suite

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-07-01 by dgarske