wolfSSL/wolfssl #5051: Test (check) fails on wolfssl-5.2.0-gplv3-fips-ready

Subetov, 2022-04-13:
### Version

wolfssl-5.2.0-gplv3-fips-ready

### Description

./configure --enable-fips=ready
make
make check

============================================================================
Testsuite summary for wolfssl 5.2.0
============================================================================
# TOTAL: 7
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0

ready file /home/user/wolfssl-5.2.0-gplv3-fips-ready/wolfssl_resume_ready34117
-r          Resume session
-e
Starting example server for resume test...

-e
Resume test supported
./examples/server/server -r -R /home/user/wolfssl-5.2.0-gplv3-fips-ready/wolfssl_resume_ready34117 -p 0
-e waiting for ready file...
wolfSSL error: can't load server cert file, check file and run from wolfSSL home dir
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e waiting for ready file...
-e NO ready file ending test...
in cleanup
killing server
./scripts/resume.test: 45: kill: No such process

FAIL scripts/resume.test (exit status: 1)

FAIL: scripts/google
====================

PING www.google.com (172.217.10.100) 56(84) bytes of data.
64 bytes from lga34s15-in-f4.1e100.net (172.217.10.100): icmp_seq=1 ttl=113 time=18.3 ms
64 bytes from lga34s15-in-f4.1e100.net (172.217.10.100): icmp_seq=2 ttl=113 time=18.4 ms

--- www.google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1002ms
rtt min/avg/max/mdev = 18.346/18.395/18.444/0.049 ms
wolfSSL error: can't load client cert file, check file and run from wolfSSL home dir
-e

Client connection failed
FAIL scripts/google.test (exit status: 1)

FAIL: scripts/unit
==================

starting unit tests...
wolfSSL error: AES-CBC CAST failed
FAIL scripts/unit.test (exit status: 1)

embhorn, 2022-04-14:
Hello @Subetov 

The FIPS hash has to be updated before the tests can pass, 
> 4. Update the in-core memory hash.
`$ ./wolfcrypt/test/testwolfcrypt`

Here is a link to the FIPS Ready User Guide:
https://www.wolfssl.com/docs/fips-ready-user-guide/

Let us know if there are questions.

Thanks,
Eric @ wolfSSL Support

Subetov, 2022-04-14:
Hello @embhorn 
Thank you for reply.
I have already updated the hash, according to this page https://www.wolfssl.com/docs/fips-ready-user-guide .
Maybe a did something wrong.
Could you please provide correct example of updating the hash please.

embhorn, 2022-04-14:
You can also use the FIPS hash script to update the hash:
`./fips-hash.sh`

Then make again: 
`make`

And finally run make check:
`make check`

The error suggests that the path is invalid. Are you running the commands from the base wolfSSL directory?
```
eric@ubuntu:~/Downloads/wolfssl-5.2.0-gplv3-fips-ready$ make check
make -j9  check-recursive
make[1]: Entering directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[2]: Entering directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
make -j9   testsuite/testsuite.test   scripts/resume.test       scripts/tls13.test scripts/external.test scripts/google.test scripts/openssl.test  scripts/unit.test
make[3]: Entering directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[3]: warning: -j9 forced in submake: resetting jobserver mode.
make[3]: Nothing to be done for 'scripts/resume.test'.
make[3]: Nothing to be done for 'scripts/tls13.test'.
make[3]: Nothing to be done for 'scripts/external.test'.
make[3]: Nothing to be done for 'scripts/google.test'.
make[3]: Nothing to be done for 'scripts/openssl.test'.
make[3]: 'scripts/unit.test' is up to date.
make[3]: Leaving directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make -j9  check-TESTS
make[3]: Entering directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[3]: warning: -j9 forced in submake: resetting jobserver mode.
make[4]: Entering directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[4]: warning: -j9 forced in submake: resetting jobserver mode.
PASS: scripts/openssl.test
PASS: scripts/external.test
PASS: scripts/resume.test
PASS: scripts/google.test
PASS: testsuite/testsuite.test
PASS: scripts/tls13.test
PASS: scripts/unit.test
============================================================================
Testsuite summary for wolfssl 5.2.0
============================================================================
# TOTAL: 7
# PASS:  7
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
make[4]: Leaving directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[3]: Leaving directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[2]: Leaving directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
make[1]: Leaving directory '/home/eric/Downloads/wolfssl-5.2.0-gplv3-fips-ready'
```

Subetov, 2022-04-14:
Yes, from the base (root) directory.
```
➜  ~/Downloads/wolfssl-5.2.0-gplv3-fips-ready$ ./fips-hash.sh
➜  ~/Downloads/wolfssl-5.2.0-gplv3-fips-ready$ make
/Library/Developer/CommandLineTools/usr/bin/make -j9  all-recursive
make[2]: warning: -jN forced in submake: disabling jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-fips_test.lo
  CCLD     src/libwolfssl.la
  CCLD     wolfcrypt/benchmark/benchmark
  CCLD     wolfcrypt/test/testwolfcrypt
  CCLD     examples/benchmark/tls_bench
  CCLD     examples/client/client
  CCLD     examples/echoclient/echoclient
  CCLD     examples/server/server
  CCLD     examples/echoserver/echoserver
  CCLD     testsuite/testsuite.test
  CCLD     tests/unit.test
➜  ~/Downloads/wolfssl-5.2.0-gplv3-fips-ready$ make check
/Library/Developer/CommandLineTools/usr/bin/make -j9  check-recursive
make[2]: warning: -jN forced in submake: disabling jobserver mode.
/Library/Developer/CommandLineTools/usr/bin/make -j9   testsuite/testsuite.test   scripts/resume.test       scripts/tls13.test scripts/external.test scripts/google.test scripts/openssl.test  scripts/unit.test
make[3]: warning: -jN forced in submake: disabling jobserver mode.
make[3]: Nothing to be done for `scripts/resume.test'.
make[3]: Nothing to be done for `scripts/tls13.test'.
make[3]: Nothing to be done for `scripts/external.test'.
make[3]: Nothing to be done for `scripts/google.test'.
make[3]: Nothing to be done for `scripts/openssl.test'.
make[3]: `scripts/unit.test' is up to date.
/Library/Developer/CommandLineTools/usr/bin/make -j9  check-TESTS
make[3]: warning: -jN forced in submake: disabling jobserver mode.
make[4]: warning: -jN forced in submake: disabling jobserver mode.
PASS: scripts/openssl.test
FAIL: testsuite/testsuite.test
PASS: scripts/external.test
FAIL: scripts/unit.test
FAIL: scripts/google.test
FAIL: scripts/resume.test
PASS: scripts/tls13.test
============================================================================
Testsuite summary for wolfssl 5.2.0
============================================================================
# TOTAL: 7
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
make[4]: *** [test-suite.log] Error 1
make[3]: *** [check-TESTS] Error 2
make[2]: *** [check-am] Error 2
make[1]: *** [check-recursive] Error 1
make: *** [check] Error 2
============================================================================
# TOTAL: 7
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
make[4]: *** [test-suite.log] Error 1
make[3]: *** [check-TESTS] Error 2
make[2]: *** [check-am] Error 2
make[1]: *** [check-recursive] Error 1
make: *** [check] Error 2`
```

embhorn, 2022-04-14:
Could you tell us about the operating environment? Is this running from Windows WSL by chance?

Subetov, 2022-04-14:
@embhorn thank you so much!
Probably I did something wrong when I tried to update the hash manually, canceled the changes and the `./fips-hash.sh `did a trick.
I am sorry to bother you.
Thank you very much again

embhorn, 2022-04-14:
Great to hear! I'll mark this as resolved.

closed 2022-04-14 by embhorn