wolfSSL/wolfssl #5064: [Bug]: d2i_ECPrivateKey() failure

masap, 2022-04-20:
### Contact Details

masashi.honma@gmail.com

### Version

e89b4cce4c5fcb42c76b46afd945352e3ec0863f

### Description

When using ```./configure --enable-all``` trailing program works fine. But when using ```./configure --enable-md4 --enable-des3 --enable-pwdbased --enable-wpas --enable-wpas-dpp \
--enable-opensslall --enable-opensslextra --enable-dsa --enable-keying-material --disable-tls13```, ```d2i_ECPrivateKey()``` returns ```NULL```. It seems this check causes failure.
```
int wc_ecc_set_curve(ecc_key* key, int keysize, int curve_id)
...
    if (keysize > ECC_MAXSIZE) {
        return ECC_BAD_ARG_E;
    }
...
```

This is the program extracted from hostapd/wpa_supplicant auto test.
```
#include <stdio.h>
#include <string.h>

#include <wolfssl/options.h>
#include <wolfssl/openssl/ec.h>
#include <wolfssl/openssl/evp.h>

int main(void)
{
	const unsigned char data[121] = {
		0x30, 0x77, 0x02, 0x01, 0x01, 0x04, 0x20, 0x02, 0x48, 0x1a, 0x57, 0xe7, 0xb5, 0xea, 0x27, 0x0d,
		0xbe, 0xeb, 0x43, 0x23, 0xff, 0x48, 0x64, 0x23, 0xc3, 0x3a, 0xd9, 0x02, 0x77, 0xeb, 0x5c, 0x89,
		0xb8, 0xdb, 0xcd, 0x15, 0x59, 0xa8, 0xd8, 0xa0, 0x0a, 0x06, 0x08, 0x2a, 0x86, 0x48, 0xce, 0x3d,
		0x03, 0x01, 0x07, 0xa1, 0x44, 0x03, 0x42, 0x00, 0x04, 0x79, 0x19, 0x2c, 0xb4, 0xa4, 0x15, 0xf8,
		0x54, 0x80, 0x32, 0xa0, 0x5a, 0xca, 0xa1, 0x64, 0x9e, 0x7d, 0xa7, 0x01, 0x0b, 0x1d, 0xb2, 0x1f,
		0xcc, 0x85, 0xd5, 0xfa, 0xb6, 0x21, 0xce, 0x4c, 0x01, 0x62, 0x9f, 0xfb, 0x14, 0x30, 0x54, 0x4c,
		0xca, 0xca, 0x4f, 0xb4, 0x8b, 0xbb, 0x62, 0xf9, 0x4a, 0x76, 0xc7, 0xb1, 0x84, 0x8d, 0x23, 0x3d,
		0x58, 0x5a, 0x61, 0x99, 0xf5, 0x8c, 0xa5, 0xa1, 0x66,
	};

	size_t der_len = 121;
	const unsigned char *der = data;
	EC_KEY *eckey = d2i_ECPrivateKey(NULL, (const unsigned char **)&der, der_len);
	if (!eckey) {
		fprintf(stderr, "d2i_ECPrivateKey() failed: %s\n", ERR_error_string(ERR_get_error(), NULL));
		return -1;
	}
	EC_KEY_free(eckey);

	return 0;
}
```

embhorn, 2022-04-20:
Hi @masap 

Thanks for this excellent report. I was able to reproduce with your instructions.

With `--enable-all`, the macro `WOLFCRYPT_HAVE_SAKKE` changes the value of `ECC_MAXSIZE`:

https://github.com/wolfSSL/wolfssl/blob/49f510544af1438312e86782123cb9a9acce9fbf/wolfssl/wolfcrypt/ecc.h#L152
```
#ifdef WOLFCRYPT_HAVE_SAKKE
    ECC_MAXSIZE     = 128,  /* MAX Private Key size */
    ECC_MAXSIZE_GEN = 128,  /* MAX Buffer size required when generating ECC keys*/
#else
    ECC_MAXSIZE     = 66,   /* MAX Private Key size */
    ECC_MAXSIZE_GEN = 74,   /* MAX Buffer size required when generating ECC keys*/
#endif
```

@SparkiDev - should the max private key size be increased for non-SAKKE configs?

SparkiDev, 2022-04-26:
Hi @masap
I've put up a pull request (#5083) that decodes the DER instead of interpreting the data as a raw key.
This fixes the issue properly.

Let me know if this works for you.

Thanks,
Sean Parkinson, wolfSSL

masap, 2022-04-26:
Great ! Your PR works fine ! I will close this issue when the PR is merged.

And your PR also fixes #5069 as well. Is this to be expected?

SparkiDev, 2022-04-26:
Hi @masap
I did test the POC from #5069 with the fix and saw that the import is now working properly and thus the rest of the calls will work too.

I'll let you know when it is merged.

Sean :-)

PR #5083 (2022-04-27): d2i_ECPrivateKey() takes DER encoded data
# Description

d2i_ECPrivateKey() takes DER encoded data
Code was incorrectly using data as a private key instead of DER
decoding.

Please describe the scope of the fix or feature addition.

Fixes #5064

# Testing

See POC code in Issue.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2022-04-27 by dgarske