wolfSSL/wolfssl #5069: [Bug]: EVP_PKEY_assign_EC_KEY() failed to assign private only key

masap, 2022-04-22:
### Contact Details

masashi.honma@gmail.com

### Version

4a4b019e307a2cce25351270aea74d020bc5deda

### Description

In the following program `EVP_PKEY_assign_EC_KEY()` succeeds for `OpenSSL 1.1.1f` but fails for `wolfSSL`. It seems this check causes failure.
```C
int wc_ecc_export_x963(ecc_key* key, byte* out, word32* outLen)
{
...
   if (key->type == ECC_PRIVATEKEY_ONLY)
       return ECC_PRIVATEONLY_E;
...
```

This is similar to #5064 but different issue.

### Reproduction steps

Run this program with `./configure --enable-all --enable-md4 --enable-des3 --enable-pwdbased --enable-wpas --enable-wpas-dpp --enable-opensslall --enable-opensslextra --enable-dsa --enable-keying-material --disable-tls13`.

```C
#include <stdio.h>
#include <string.h>

#if defined(CONFIG_WOLFSSL)
#include <wolfssl/options.h>
#include <wolfssl/openssl/ec.h>
#include <wolfssl/openssl/evp.h>
#else /* CONFIG_WOLFSSL */
#include <openssl/ec.h>
#include <openssl/evp.h>
#include <openssl/err.h>
#endif /* CONFIG_WOLFSSL */

int main(void)
{
	const unsigned char data[121] = {
		0x30, 0x77, 0x02, 0x01, 0x01, 0x04, 0x20, 0x02, 0x48, 0x1a, 0x57, 0xe7, 0xb5, 0xea, 0x27, 0x0d,
		0xbe, 0xeb, 0x43, 0x23, 0xff, 0x48, 0x64, 0x23, 0xc3, 0x3a, 0xd9, 0x02, 0x77, 0xeb, 0x5c, 0x89,
		0xb8, 0xdb, 0xcd, 0x15, 0x59, 0xa8, 0xd8, 0xa0, 0x0a, 0x06, 0x08, 0x2a, 0x86, 0x48, 0xce, 0x3d,
		0x03, 0x01, 0x07, 0xa1, 0x44, 0x03, 0x42, 0x00, 0x04, 0x79, 0x19, 0x2c, 0xb4, 0xa4, 0x15, 0xf8,
		0x54, 0x80, 0x32, 0xa0, 0x5a, 0xca, 0xa1, 0x64, 0x9e, 0x7d, 0xa7, 0x01, 0x0b, 0x1d, 0xb2, 0x1f,
		0xcc, 0x85, 0xd5, 0xfa, 0xb6, 0x21, 0xce, 0x4c, 0x01, 0x62, 0x9f, 0xfb, 0x14, 0x30, 0x54, 0x4c,
		0xca, 0xca, 0x4f, 0xb4, 0x8b, 0xbb, 0x62, 0xf9, 0x4a, 0x76, 0xc7, 0xb1, 0x84, 0x8d, 0x23, 0x3d,
		0x58, 0x5a, 0x61, 0x99, 0xf5, 0x8c, 0xa5, 0xa1, 0x66,
	};

	size_t der_len = 121;
	const unsigned char *der = data;
	EC_KEY *eckey = d2i_ECPrivateKey(NULL, (const unsigned char **)&der, der_len);
	if (!eckey) {
		fprintf(stderr, "d2i_ECPrivateKey() failed: %s\n", ERR_error_string(ERR_get_error(), NULL));
		return -1;
	}

	EC_KEY_set_conv_form(eckey, POINT_CONVERSION_COMPRESSED);

	EVP_PKEY *pkey = EVP_PKEY_new();
	if (!pkey) {
		fprintf(stderr, "EVP_PKEY_new() failed: %s\n", ERR_error_string(ERR_get_error(), NULL));
		return -1;
	}

	if (EVP_PKEY_assign_EC_KEY(pkey, eckey) != 1) {
		fprintf(stderr, "EVP_PKEY_assign_EC_KEY() failed: %s\n", ERR_error_string(ERR_get_error(), NULL));
		return -1;
	}

	EC_KEY_free(eckey);

	return 0;
}
```

embhorn, 2022-04-22:
Hi @masap 

Thanks for sharing this report.

@JacobBarthelmeh - perhaps we could do something like this with only the private key?
https://github.com/wolfSSL/wolfssl/blob/4a4b019e307a2cce25351270aea74d020bc5deda/wolfcrypt/src/ecc.c#L7716-L7724

masap, 2022-04-27:
Could I close this issue according to comment of @SparkiDev https://github.com/wolfSSL/wolfssl/issues/5064#issuecomment-1109440353 ?.

SparkiDev, 2022-04-27:
Closing as fix for #5064 made the POC work.

closed 2022-04-27 by SparkiDev