wolfSSL/wolfssl #5108: SendCertificateVerify() creates invalid data when SendHandshakeMsg() returns WANT_WRITE

rliebscher, 2022-05-05:
### Contact Details

rene.liebscher@webfleet.com

### Version

wolfSSL 5.1.1

### Description

There will be created a wrong TLS fragment. It depends from the previous exchanged data.

The problem appears when while TLS connection establishment SendCertificateVerify() (internal.c) is called.
This calls SendHandshakeMsg(). This might return with WANT_WRITE because the lower layer is not ready to transmit the data.

At the next call of wolfSSL_connect() you go the whole way again until SendHandshakeMsg().
But there are some states already changed.
- First in SendCertificateVerify() the state is reset at begin (ssl->options.asyncState = TLS_ASYNC_BEGIN;)
- This results in a complete new calculation of the content of the message (for the next call of SendHandshakeMsg(), which would also handle fragmentation) The content is based on the hash of all previous messages.
- But SendHandshakeMsg() had already updated the hashes for the message, it could not send in the last cycle.
- And SendHandshakeMsg() had its internal state updated (ssl->fragOffset += fragSz;)

Now you have two cases:
- The new data have a length equal or less then the previous content. Then no new content is added (while (ssl->fragOffset < inputSz)) but also no call of SendBuffered() happens. (The data in buffer are going to be send with the next added data then, SendChangeCipher()) This would be OK.
- But if the new data have a larger size, SendHandshakeMsg() would try to make another fragment of the new bytes at the end. This is the problem.

So I guess either the updating of the hashes should not happen in SendHandshakeMsg() too early in case SendBuffered() returns with WANT_WRITE. (and all functions are called again)
Or the content of the message has to be stored and not recalculated until all fragments could successfully buffered into the output buffer (meaning also the state should not be reset to at the start of SendCertificateVerify())

I have no real code to show the problem as it depends from previous messages which can contain anything. I found it by analysing why SendHandshakeMsg() was called with data of different sizes.
I guess it should be possible to write unit tests for the problem.

BTW: There is another caller of SendHandshakeMsg(), SendServerKeyExchange(), which has the call of SendHandshakeMsg() in its state TLS_ASYNC_FINALIZE, not in TLS_ASYNC_END as in SendCertificateVerify(). Might be worth to check this too.

PS: WOLFSSL_ASYNC_CRYPT is not defined

embhorn, 2022-05-05:
Hello @rliebscher 

Thanks for this excellent report. We are reviewing internally and will post an update.

julek-wolfssl, 2022-05-11:
Hi @rliebscher, this is an interesting bug report and I am able to reproduce issues stemming from `WANT_WRITE`'s being returned from the network layer. We are discussing how to best tackle this problem.
Thanks
Juliusz

julek-wolfssl, 2022-05-17:
Continued in https://github.com/wolfSSL/wolfssl/pull/5149.