wolfSSL/wolfssl #5115: Stack smashing detected

Cavalletta98, 2022-05-06:
### Version

5.3.0

### Description

Hi, i'm including the library in my project and build it using the CMakeLists.txt provided. Once i build the project and run it, i get this error: *** stack smashing detected ***: terminated. I have already imported the file options.h into my project but didn't fix it
Can someone help me?

embhorn, 2022-05-06:
Hello @Cavalletta98 

What config options are you using? I can try to reproduce your issue.

Thanks,
Eric @ wolfSSL Support

Cavalletta98, 2022-05-06:
I'm using the one inside examples/config . It Is called user_settings_all.h

embhorn, 2022-05-09:
@Cavalletta98 

Did you follow these instructions?
https://github.com/wolfSSL/wolfssl/blob/59fdf05155e43c14db8c9340d1fe32490094f89b/INSTALL#L79

Could you please try using the automake build? That will help us to know if it is a problem with the Cmake method.
https://github.com/wolfSSL/wolfssl/blob/59fdf05155e43c14db8c9340d1fe32490094f89b/INSTALL#L1

embhorn, 2022-05-09:
Could you please try these instructions and let us know the result?
https://github.com/wolfSSL/wolfssl/blob/59fdf05155e43c14db8c9340d1fe32490094f89b/INSTALL#L94-L111

embhorn, 2022-05-18:
@Cavalletta98 - were you able to build successfully? Can I mark this issue as resolved?

Cavalletta98, 2022-05-18:
Sorry for not answer but i'm trying to build in a different way as you suggest. You suggest to only build wolfssl but i'm trying to build wolfssl inside my project by using my own CMakeLists.txt

haydenroche5, 2022-05-19:
Hi. If you want to configure wolfSSL with a user_settings.h file, you need to use this CMake option:

```
add_option("WOLFSSL_USER_SETTINGS"
    "Use your own user_settings.h and do not add Makefile CFLAGS (default: disabled)"
    "no" "yes;no")
```

(i.e. your project's CMakeLists.txt needs to set `WOLFSSL_USER_SETTINGS` to 1). You do not need options.h if you're building with user_settings.h. They are essentially mutually exclusive.

However, this isn't the approach I would recommend. I would go the options.h route and turn on whatever features you need with the options documented in wolfSSL's CMakeLists.txt. You can start by just turning nothing explicitly on and taking the defaults, which will populate options.h with the relevant macros.

If you're still not able to get things working, I'll need your project's CMakeLists.txt, CMakeCache.txt, and the `cmake` command you're running to tell what's wrong.

Cavalletta98, 2022-06-06:
Thank you for your help. We used your suggestions to improve our build system and making smaller application to test the issue. We finally found the problem: wrong buffer size in out application code (no relation to wolfssl library). Thank you again for your support