wolfSSL/wolfssl #5141: Hope to support RFC8998 (SM2, SM3, SM4), Thanks.

shove70, 2022-05-13:


dgarske, 2022-05-13:
Hi @shove70 ,

We do support SM2 verify, but it is not in the public wolfSSL repository. Please emails facts@wolfssl.com if you are interested in evaluating it.

We will eventually add the other SM Chinese algorithms, but we will continue to keep them in a separate repository and release bundle.

Thanks,
David Garske, wolfSSL

shove70, 2022-05-16:
> We do support SM2 verify, but it is not in the public wolfSSL repository. Please emails [facts@wolfssl.com](mailto:facts@wolfssl.com) if you are interested in evaluating it.
> 
> We will eventually add the other SM Chinese algorithms, but we will continue to keep them in a separate repository and release bundle.

Thank you for your guidance.

closed 2022-05-16 by shove70